contiki-ng/contiki-ng #1213: cooja need serial separate from console log

alexrayne, 2020-03-27:
Hallow!
Try to rut board-router-cooja-tsch test as example. and found that ping6 does not sence any response. 
Reason, imho, that serial of board-router is need SLIP connection with host, but thete is a lot of log messages are gone to that serial too. 
At least, it difficult to see - what actual data are go there, at most - all serial interconnect is broken by log.
Can anybody explain me how is cooja implements serial interface? is there possibility to deploy logging printf console from serial interconnect? myabe somebody have serial interface alredy implemented?

atiselsts, 2020-03-30:
Hi Alex,

There are two kinds of traffic on SLIP interface:
1) debug lines created by `printf()` and `puts()` calls;
2) network traffic payload.

Debug lines are printed, network traffic payload is forwarded to the TUN interface. If something is ending with '\n', then it's likely to be a debug line. The problem is that network traffic can also have '\n' inside its payload. The rule to distinguish between debug lines and network traffic is to look if the data has non-ascii characters besides '\n'. This is done in the is_sensible_string() function.

This is not the most elegant or even correct of solutions, but is seems to work most of the time.

alexrayne, 2020-03-30:
Thank you for answer! Somewhat about it i supposed. It looks pretty if all work ok. 
But now, my BR does not any reaction on ping requests. Therefore i starts rework of serial/log mote interfaces.
At present i separate serial fromlog, and improve serial UI with dump/hex view

alexrayne, 2020-03-30:
fix by PR #1218

closed 2020-03-30 by alexrayne