contiki-ng/contiki-ng #1216: Creating MQTT Simulation on Cooja

xtrega, 2020-03-29:
Hello there. I'm new to github and I want to ask a question about MQTT simulation on Cooja.

So, for my thesis, I'm trying to use cooja simulator and doing a MQTT simulation on it. I setting up my Contiki-NG with docker image. So here's my question:

1. besides mqtt-client mote, is rpl-border-router mote is also needed to create the simulation? 
2. Are there another types of mote that I can use for mqtt-client besides cooja mote? I tried Z1 and sky and it's not working.
3. I'm already tried it once, and try to connect the border-router. What's the next step? and How do I connect the mqtt-client on cooja to connect with the broker (mosquitto)? I still can't figure it out until now.

Maybe that's my question for now. Thank you.

atiselsts, 2020-03-30:
Hi,

1. You need the border router functionality, so either use the rpl-border-router example or add BR as a module:

> Note that you do not have to use the rpl-border-router example to build a border router. In Contiki-NG, you can simply turn any project into a border router by adding to your Makefile the following module (and as always, make distclean):
> 
> MODULES += os/services/rpl-border-router

2. According to the Makefile in `examples/mqtt-client/`, the following platforms are supported:
`cc26x0-cc13x0 cc2538dk openmote zoul native simplelink`
None of them can be simulated in Cooja unfortunately. The z1 and sky probably don't have enough RAM.

3. Look at the wiki about information how to connect  border router in cooja:
https://github.com/contiki-ng/contiki-ng/wiki/Cooja:-simulating-a-border-router

xtrega, 2020-03-31:
> 
> 
> Hi,
> 
>     1. You need the border router functionality, so either use the rpl-border-router example or add BR as a module:
> 
> 
> > Note that you do not have to use the rpl-border-router example to build a border router. In Contiki-NG, you can simply turn any project into a border router by adding to your Makefile the following module (and as always, make distclean):
> > MODULES += os/services/rpl-border-router
> 
>     1. According to the Makefile in `examples/mqtt-client/`, the following platforms are supported:
>        `cc26x0-cc13x0 cc2538dk openmote zoul native simplelink`
>        None of them can be simulated in Cooja unfortunately. The z1 and sky probably don't have enough RAM.
> 
>     2. Look at the wiki about information how to connect  border router in cooja:
>        https://github.com/contiki-ng/contiki-ng/wiki/Cooja:-simulating-a-border-router

Thanks for the answer. I will try it out again and comment in here again for the result/problem that I have later.

xtrega, 2020-03-31:
> Hi,
>
>     1. You need the border router functionality, so either use the rpl-border-router example or add BR as a module:
>
>
>
>     1. According to the Makefile in `examples/mqtt-client/`, the following [...]

Hi. I tried it again, and here are the results:
[image]
[image]

1. I already tried to connect the border router and it succeed, then, I try to activate mosquitto broker on the same host as cooja. 
2. I started the simulation, but, the result is not like what I expected to be, based on the terminal, the mqtt and broker only receive CONNACK and PUBLISH, but there's no SUBSCRIBE in it and the mote output are too random.

Can you tell me what's the problem or the solution that you have? Thank you.

xtrega, 2020-03-31:
@atiselsts Hi. I tried it again, and here are the results:

[image]
[image]

1. I already tried to connect the border router and it succeed, then, I try to activate mosquitto broker on the same host as cooja.
2. I started the simulation, but, the result is not like what I expected to be, based on the terminal, the mqtt and broker only receive CONNACK and PUBLISH, but there's no SUBSCRIBE in it and the mote output on cooja are too random.

Can you tell me what's the problem or the solution that you have? Thank you.

g-oikonomou, 2020-03-31:
I cannot immediately think why the mqtt-client example does not support platform cooja - I'll need to have a look but I think it would be OK to simply add it under platforms-only.

Regarding the subscribe, please see the mqtt client example's README file - devices won't  subscribe under the default configuration:

https://github.com/contiki-ng/contiki-ng/tree/develop/examples/mqtt-client#subscribing

By the way, your node output in that cooja screenshot doesn't look quite right. What firmware have you loaded to device with ID 1?

xtrega, 2020-04-01:
> I cannot immediately think why the mqtt-client example does not support platform cooja - I'll need to have a look but I think it would be OK to simply add it under platforms-only.
>
> Regarding the [...]

@g-oikonomou Okay, when you already have a look at it, please let me know. 

Ohh okay, I'll carefully read the README file later and try it again on my cooja simulator.

Yeah, I think so too, the firmware that the node use is:
[image]

What do I have to do to make the node output right? Thank you.

g-oikonomou, 2020-04-15:
Tested the MQTT client example with a simulated cooja node and, as expected, it works as intended. So I am changing the example to allow this platform with #1230 

Closing this

closed 2020-04-15 by g-oikonomou

xtrega, 2020-04-17:
@g-oikonomou Hello, I want to ask you again about subscribing. I'm still confused about what and where to change it on project.conf.h ("Org ID" != 'quickstart'). Should I write that on #ifdefine MQTT_CLIENT_CONF_ORG_ID or #define MQTT_CLIENT_CONF_ORG_ID "quickstart"? Thank you.

PR #1230 (2020-04-19): Allow the MQTT client example for platform Cooja
There is no reason to disallow the cross-platform MQTT Client example for platform Cooja, but the platform is not listed under `PLATFORMS_ONLY`.

This pull fixes this.

Tested end-to-end inside docker, with:

* MQTT Client native cooja node
* Sky simulated BR inside Cooja
* tunslip6 outside Cooja talking to the simulated BR
* Mosquitto outside

Closes #1216