contiki-ng/contiki-ng #1217: Create my own nbr-table

philok93, 2020-03-30:
Hi, 
I am working for my thesis and I want to create my own neighbour table by using NBR_TABLE_GLOBAL. Is this going to create a new table where I can add entries without affecting the  normal nbr-table? Because I tried to do it but when I have 10 motes, I get null table and no space error. What should I do to reduce the space of a mote. I want to create a small table to keep some malicious nodes that are neighbours of a node.
Thank you!

atiselsts, 2020-03-30:
You can add your own neighbor table with the `NBR_TABLE` and `NBR_TABLE_GLOBAL` macros.

There are two constants that determine the dimensions of neighbor tables:

 * NBR_TABLE_CONF_MAX_NEIGHBORS - this is the max number of neighbors per table. This is platform-specific, but by default it is only 8.
* MAX_TABLE, the maximum number of neighbor or tables. This is not user-configurable and is equal to 8.

philok93, 2020-03-30:
Thank you for your fast reply. Ok I checked the macros but I wonder, in nbr-table.c: nbr_table_add_lladdr() I don't see the use of *table argument and it creates a new list so maybe is confusing the 2 tables. Is this list that is using a global keys list?
I have question about ip-nbr-ds6, can I remove some neighbours from this cache list to increase free space? 

Thank you!

atiselsts, 2020-03-30:
Hi,

I'm not sure I understand your follow up questions. Anyway, please use Gitter (https://gitter.im/contiki-ng/Developers) or Stack Overflow (https://stackoverflow.com/questions/tagged/contiki) for further discussionss, let's keep an GitHub issue focused to a single main point. If my reply helped you with the initial question, the issue should be closed.

closed 2020-03-30 by philok93