contiki-ng/contiki-ng #1220: Create motes in COOJA based on SimpleLink MCU platform

mChassignol, 2020-04-03:
Hello there!

I am new to Contiki-ng and Cooja and I would like to ask you a question about Cooja and SimpleLink MCU platform.

My device is based on CC1352 so I followed this tutorial to set up the environment (https://github.com/contiki-ng/contiki-ng/wiki/Platform-simplelink). I would like to use Cooja in my tests especially when I develop new features.

**How can I create motes in Cooja based on SimpleLink MCU platform ?**

Thank you in advance for your help!

arurke, 2020-04-03:
afaik the only HW emulation supported is for MSP-based platforms such as Sky and z1. Ref. https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-Running-Contiki‐NG-in-Cooja

mChassignol, 2020-04-03:
Thank you @arurke for your answer!

Is it possible to add a it with "import java mote" for example? if yes how?

rajeev1986, 2020-04-09:
@mChassignol ,
Yes, it is possible to add new platforms to COOJA. However, it is not straight forward and one needs to extend the simulator (Java code) with the following low-level MCU details before it can be simulated in Cooja.
- extending the mote interface to load the hex or elf file for emulating the  CPU instruction of that particular platform.
- memory interface
- timing details possibly
- on the radio side, radio channel model
- device peripherals

A good starting point will be understanding how the above are implemented in MSPSim for T-mote Sky and Zolerteria Z1 and then extending those.

I hope it helps.

closed 2020-11-25 by g-oikonomou