contiki-ng/contiki-ng #1224: Contiki-ng problem compiling "Hello-World"

dpapdpap, 2020-04-13:
I'm new to /linux/rasbian/contiki. 
I just installed contiki-ng. 
I had Java 11 but I faced problems building cooja. 
So I downgraded to 10 the same, to 9 the same. Finally it compiles (without errors or warnings) vith Java 8 openJDK.
But when I try to add a node (eg. cooja node) i receive this compilation error :

> make hello-world.cooja TARGET=cooja 
  CC        ../../arch/platform/cooja/./platform.c
../../arch/platform/cooja/./platform.c:38:10: fatal error: jni.h: No such file or directory
 #include <jni.h>
          ^~~~~~~
compilation terminated.
make: *** [../../Makefile.include:385: platform.o] Error 1
Process returned error code 2
I try G.Oikonomou suggestion to add this
-I/System/Library/Frameworks/JavaVM.framework/Headers to COMPILER_ARGS through the external tools dialogue
but without result.
The file jni.h exists in /home/usr/lib/jvm/java-8-openjdk-armhf/include and 
readlink -f `(type -p java)` gives /usr/lib/jvm/java-8-openjdk-armhf/jre/bin/java
echo $JAVA_HOME  --> /usr/lib/jvm/java-8-openjdk-armhf

java -version --> openjdk version "1.8.0_212"
OpenJDK Runtime Environment (build 1.8.0_212-8u212-b01-1+rpi1-b01)
OpenJDK Client VM (build 25.212-b01, mixed mode)

What is the problem, please ?

rajeev1986, 2020-04-14:
@dpapdpap 
Seems like your JAVA_HOME is not set correctly. I am on a Ubuntu machine where I can run COOJA properly and my JAVA_HOME is set to point
`/usr/lib/jvm/java-8-openjdk-amd64`

I have usually seen the above issue while trying to run cooja on macOS. Since current cooja implementation is for 32bit systems, on macOS which is 64-bit, it fails to run and gives jni.h is missing error.

Hope it help...

dpapdpap, 2020-04-15:
Rajeev, thank you for your help.
Finally the problem was that i was using sudo ant run but i have to execute as single user
Thank you again

rajeev1986, 2020-04-15:
@dpapdpap 
Glad to hear that you managed to solve the issue.
Just execute `ant run `and cooja will fire up...

closed 2020-04-29 by yatch

poornimagowda, 2022-07-01:
I am new to Contiki OS... I have installed Contiki3.0 and run examples.... I got following errors please any  one help me
> make hello-world.sky TARGET=sky 
make: msp430-gcc: Command not found
make: msp430-gcc: Command not found
fatal: Not a git repository: '../../.git'
  CC        ../../platform/sky/./contiki-sky-platform.c
make: msp430-gcc: Command not found
make: *** [obj_sky/contiki-sky-platform.o] Error 127
Process returned error code 2