contiki-ng/contiki-ng #1227: Unable to detect MAC layer ACK when run with cc1200 radio

m-srk, 2020-04-15:
**Outline:** When I try to run the rpl-udp client and server example from Contiki-NG, the client node continuously exits from the RPL network because it is unable to find a parent.

**Contiki-NG version:** release 4.4

**Hardware used:** Zolertia Remote revB

**Description:** I flashed the rpl-udp server and client on a couple of Zolertia Remote rev-B motes. The only changes I made to the code before flashing it:
1. Added a debug statement in the csma-output.c file to output the status of last tx (TX_OK or NOACK etc). 
2. Changed the contiki-conf.h so the cc1200 sub GHz driver is used.

After flashing, the server becomes the DAG root and the client node joins the RPL network created by the server. The client then proceeds to send out a unicast DIO probe message to the root to which it fails to receive the MAC layer ACK after tx of the frame. Since the status after each tx is NOACK (as shown in the output), the frame is sent out for the MAC_MAX_RETRIES (in my case 8). Because of this, the client is unable to find an RPL parent and hence it exits the DAG and sends out a DIS and this repeats all over again.

**Output:** Here is the serial output from the client node after booting up:

```                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0x5a87bf24)
rlwrap ../../tools/serial-io/serialdump -b115200 /dev/ttyUSB1
connecting to /dev/ttyUSB1 [OK]
[INFO: Main      ] Starting Contiki-NG-release/v4.4-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 24991
[INFO: Main      ] Link-layer address: 0012.4b00.060d.619f
[INFO: RPL       ] initializing
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:60d:619f
[INFO: Zoul      ] Zolertia RE-Mote revision B platform
[INFO: App       ] Not reachable yet
[INFO: RPL       ] sending a DIS to ff02::1a
[INFO: CSMA      ] sending to 0000.0000.0000.0000, len 10, seqno 111, queue length 1, free packets 7
[DBG : CSMA      ] scheduling transmission in 1 ticks, NB=0, BE=3
[INFO: CSMA      ] preparing packet for 0000.0000.0000.0000, seqno 111, tx 0, queue 1
[INFO: CSMA      ] tx to 0000.0000.0000.0000, seqno 111, status 0, tx 0, coll 0
[INFO: CSMA      ] status: TX_OK 
[INFO: CSMA      ] packet sent to 0000.0000.0000.0000, seqno 111, status 0, tx 1, coll 0
[DBG : CSMA      ] free_queued_packet, queue length 0, free packets 8
[INFO: CSMA      ] received packet from 0012.4b00.0615.9b5a, seqno 107, len 80
[INFO: RPL       ] received a multicast-DIO from fe80::212:4b00:615:9b5a, instance_id 0, DAG ID fd00::212:4b00:615:9b5a, version 240, dtsn 240, rank 128
[INFO: RPL       ] adding global IP address fd00::212:4b00:60d:619f
[INFO: RPL       ] reset MRHOF
[INFO: RPL       ] initialized DAG with instance ID 0, DAG ID fd00::212:4b00:615:9b5a, prexix fd00::/64, rank 65535
[WARN: RPL       ] just joined, no parent yet, setting timer for leaving
[INFO: RPL       ] refreshing lifetime
[INFO: RPL       ] best parent is not fresh, schedule urgent probing to fe80::212:4b00:615:9b5a
[INFO: RPL       ] probing fe80::212:4b00:615:9b5a (urgent) last tx 0 min ago
[INFO: RPL       ] sending a unicast-DIO with rank 65535 to fe80::212:4b00:615:9b5a
[INFO: CSMA      ] sending to 0012.4b00.0615.9b5a, len 79, seqno 112, queue length 1, free packets 7
[DBG : CSMA      ] scheduling transmission in 6 ticks, NB=0, BE=3
[INFO: CSMA      ] preparing packet for 0012.4b00.0615.9b5a, seqno 112, tx 0, queue 1
[INFO: CSMA      ] tx to 0012.4b00.0615.9b5a, seqno 112, status 2, tx 0, coll 0
[INFO: CSMA      ] status: NOACK 
[DBG : CSMA      ] scheduling transmission in 2 ticks, NB=0, BE=3
[DBG : CSMA      ] ignored ack
[INFO: CSMA      ] preparing packet for 0012.4b00.0615.9b5a, seqno 112, tx 1, queue 1
[INFO: CSMA      ] tx to 0012.4b00.0615.9b5a, seqno 112, status 2, tx 1, coll 0
[INFO: CSMA      ] status: NOACK 
[DBG : CSMA      ] scheduling transmission in 1 ticks, NB=0, BE=3
[DBG : CSMA      ] ignored ack
[INFO: CSMA      ] preparing packet for 0012.4b00.0615.9b5a, seqno 112, tx 2, queue 1
[INFO: CSMA      ] tx to 0012.4b00.0615.9b5a, seqno 112, status 2, tx 2, coll 0
[INFO: CSMA      ] status: NOACK 
[DBG : CSMA      ] scheduling transmission in 5 ticks, NB=0, BE=3
[DBG : CSMA      ] ignored ack
[INFO: CSMA      ] preparing packet for 0012.4b00.0615.9b5a, seqno 112, tx 3, queue 1
[INFO: CSMA      ] tx to 0012.4b00.0615.9b5a, seqno 112, status 2, tx 3, coll 0
[INFO: CSMA      ] status: NOACK 
[DBG : CSMA      ] scheduling transmission in 5 ticks, NB=0, BE=3
[DBG : CSMA      ] ignored ack
[INFO: CSMA      ] preparing packet for 0012.4b00.0615.9b5a, seqno 112, tx 4, queue 1
[INFO: CSMA      ] tx to 0012.4b00.0615.9b5a, seqno 112, status 2, tx 4, coll 0
[INFO: CSMA      ] status: NOACK 
[DBG : CSMA      ] scheduling transmission in 1 ticks, NB=0, BE=3
[DBG : CSMA      ] ignored ack
[INFO: CSMA      ] preparing packet for 0012.4b00.0615.9b5a, seqno 112, tx 5, queue 1
[INFO: CSMA      ] tx to 0012.4b00.0615.9b5a, seqno 112, status 2, tx 5, coll 0
[INFO: CSMA      ] status: NOACK 
[DBG : CSMA      ] scheduling transmission in 5 ticks, NB=0, BE=3
[DBG : CSMA      ] ignored ack
[INFO: CSMA      ] preparing packet for 0012.4b00.0615.9b5a, seqno 112, tx 6, queue 1
[INFO: CSMA      ] tx to 0012.4b00.0615.9b5a, seqno 112, status 2, tx 6, coll 0
[INFO: CSMA      ] status: NOACK 
[DBG : CSMA      ] scheduling transmission in 4 ticks, NB=0, BE=3
[DBG : CSMA      ] ignored ack
[INFO: CSMA      ] preparing packet for 0012.4b00.0615.9b5a, seqno 112, tx 7, queue 1
[INFO: CSMA      ] tx to 0012.4b00.0615.9b5a, seqno 112, status 2, tx 7, coll 0
[INFO: CSMA      ] status: NOACK 
[INFO: CSMA      ] packet sent to 0012.4b00.0615.9b5a, seqno 112, status 2, tx 8, coll 0
[DBG : CSMA      ] free_queued_packet, queue length 0, free packets 8
[INFO: RPL       ] packet sent to 0012.4b00.0615.9b5a, status 2, tx 8, new link metric 601
[DBG : CSMA      ] ignored ack
[INFO: CSMA      ] received packet from 0012.4b00.0615.9b5a, seqno 108, len 80
[INFO: RPL       ] received a multicast-DIO from fe80::212:4b00:615:9b5a, instance_id 0, DAG ID fd00::212:4b00:615:9b5a, version 240, dtsn 240, rank 128
[INFO: CSMA      ] received packet from 0012.4b00.0615.9b5a, seqno 109, len 80
[INFO: RPL       ] received a multicast-DIO from fe80::212:4b00:615:9b5a, instance_id 0, DAG ID fd00::212:4b00:615:9b5a, version 240, dtsn 240, rank 128
[INFO: RPL       ] sending a DIS to ff02::1a
[INFO: CSMA      ] sending to 0000.0000.0000.0000, len 10, seqno 113, queue length 1, free packets 7
[DBG : CSMA      ] scheduling transmission in 0 ticks, NB=0, BE=3
[INFO: CSMA      ] preparing packet for 0000.0000.0000.0000, seqno 113, tx 0, queue 1
[INFO: CSMA      ] tx to 0000.0000.0000.0000, seqno 113, status 0, tx 0, coll 0
[INFO: CSMA      ] status: TX_OK 
[INFO: CSMA      ] packet sent to 0000.0000.0000.0000, seqno 113, status 0, tx 1, coll 0
[DBG : CSMA      ] free_queued_packet, queue length 0, free packets 8
[INFO: CSMA      ] received packet from 0012.4b00.0615.9b5a, seqno 110, len 80
[INFO: RPL       ] received a multicast-DIO from fe80::212:4b00:615:9b5a, instance_id 0, DAG ID fd00::212:4b00:615:9b5a, version 240, dtsn 240, rank 128
[INFO: CSMA      ] received packet from 0012.4b00.0615.9b5a, seqno 111, len 80
[INFO: RPL       ] received a multicast-DIO from fe80::212:4b00:615:9b5a, instance_id 0, DAG ID fd00::212:4b00:615:9b5a, version 240, dtsn 240, rank 128
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:60d:619f, DAG state: initialized, MOP 1 OCP 1 rank 65535 max-rank 65535, dioint 0, nbr count 1 (Periodic)
[INFO: RPL       ] nbr: fe80::212:4b00:615:9b5a  128,   601 =>   729 --  8 r  f  (last tx 0 min ago, better since 0 min)
[INFO: RPL       ] nbr: end of list
[INFO: App       ] Not reachable yet
```
 Do let me know if anything else is needed. Thanks!

m-srk, 2020-04-15:
Just FYI,
ipv6 of **client** node : **60d:619f** and
that of **server** node (root) : **615:9b5a**
Only two nodes are present in the network and RPL-lite is being used.

Thanks

g-oikonomou, 2020-11-25:
This could be a bug. I'll flag it as such pending further investigation. Do update us here please if you have any more news.

tinstructor, 2021-03-15:
@g-oikonomou unfortunately, I ran into the very same issue in the current version of Contiki-NG (i.e., two releases later). It has been almost a year since this issue was reported and I was wondering if any progress had been made as it's kind of awkward to persist broken code / "functionality" through (multiple) consecutive releases? In addition I've found that a different (but likely related) problem occurs when using RPL-classic. However, in that case the udp-client does add the server as its parent but eventually drops it again as it keeps increasing its rank until the increase exceeds the max rank increase during the current DODAG version and a local repair is triggered, after which it is seemingly unable to add the server as a parent ever again.

In order to reproduce the issue reported by @m-srk I added the following `project-conf.h` file to the rpl-udp example directory:
```c
#ifndef PROJECT_CONF_H_
#define PROJECT_CONF_H_

#define NETSTACK_CONF_RADIO             cc1200_driver
#define CC1200_CONF_RF_CFG              cc1200_802154g_863_870_fsk_50kbps
#define ANTENNA_SW_SELECT_DEF_CONF      ANTENNA_SW_SELECT_SUBGHZ
#define CC1200_CONF_USE_GPIO2           0
#define CC1200_CONF_USE_RX_WATCHDOG     0
#define CC1200_CONF_802154G             0
#define CC1200_CONF_802154G_CRC16       0
#define CC1200_CONF_802154G_WHITENING   0

#define LOG_CONF_LEVEL_RPL      LOG_LEVEL_DBG
#define LOG_CONF_LEVEL_TCPIP    LOG_LEVEL_DBG
#define LOG_CONF_LEVEL_IPV6     LOG_LEVEL_DBG
#define LOG_CONF_LEVEL_MAC      LOG_LEVEL_DBG

#endif /* PROJECT_CONF_H_ */
```
In order to check whether it worked for RPL-classic (which it doesn't), I made the following changes to the Makefile of the rpl-udp example:
```
CONTIKI_PROJECT = udp-client udp-server
all: $(CONTIKI_PROJECT)

CONTIKI=../..

MAKE_MAC = MAKE_MAC_CSMA
MAKE_NET = MAKE_NET_IPV6
MAKE_ROUTING = MAKE_ROUTING_RPL_CLASSIC

include $(CONTIKI)/Makefile.include
```

I used one remote-revb as the server and a remote-reva as a client (I did however ask a colleague to run it on two remote-revb's which produced the same results). This produced the following terminal output for the udp-client:
```bash
$ make BOARD=remote-reva PORT=/dev/ttyUSB0 login
using saved target 'zoul'
rlwrap ../../tools/serial-io/serialdump -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 [OK]
[INFO: Main      ] Starting Contiki-NG-release/v4.6-110-g3ae1590f6-dirty
[INFO: Main      ] - Routing: RPL Classic
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 3929
[INFO: Main      ] Link-layer address: 0012.4b00.0616.0f59
[INFO: IPv6 DS   ] Init: 16 neighbors
[INFO: IPv6 DS   ] 2 default routers
[INFO: IPv6 DS   ] 3 prefixes
[INFO: IPv6 DS   ] 16 routes
[INFO: IPv6 DS   ] 3 unicast addresses
[INFO: IPv6 DS   ] 7 multicast addresses
[INFO: IPv6 DS   ] 2 anycast addresses
[INFO: IPv6 DS   ] Adding prefix fe80::length 64, flags 0, Valid lifetime 0, Preffered lifetime 0
[INFO: RPL       ] rpl-classic started
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:616:f59
[INFO: Zoul      ] Zolertia RE-Mote revision A platform
[INFO: App       ] Not reachable yet
[INFO: CSMA      ] received packet from 0012.4b00.09df.4dee, seqno 96, len 80
[INFO: TCP/IP    ] input: received 116 bytes
[INFO: IPv6      ] packet received from fe80::212:4b00:9df:4dee to ff02::1a
[INFO: IPv6      ] icmpv6 input length 116 type: 155 
[DBG : IPv6      ] Upper layer checksum len: 76 from: 40
[INFO: RPL       ] Received a DIO from fe80::212:4b00:9df:4dee
[DBG : RPL       ] Incoming DIO (id, ver, rank) = (30,240,128)
[... 1206 lines cut ...]
[INFO: IPv6      ] packet received from fe80::212:4b00:616:f59 to fe80::212:4b00:9df:4dee
[INFO: IPv6      ] icmpv6 input length 116 type: 155 
[DBG : IPv6      ] Upper layer checksum len: 76 from: 40
[INFO: RPL       ] Received a DIO from fe80::212:4b00:616:f59
[DBG : RPL       ] Incoming DIO (id, ver, rank) = (30,240,65535)
[DBG : RPL       ] Incoming DIO (dag_id, pref) = (fd00::212:4b00:9df:4dee, 0)
[DBG : RPL       ] Incoming DIO (option, length) = (4, 14)
[INFO: RPL       ] DAG conf:dbl=8, min=12 red=10 maxinc=896 mininc=128 ocp=1 d_l=30 l_u=60
[DBG : RPL       ] Incoming DIO (option, length) = (8, 30)
[INFO: RPL       ] Copying prefix information
[DBG : RPL       ] Prefix announced in DIO
[INFO: RPL       ] Prefix set - will announce this in DIOs
[WARN: CSMA      ] drop duplicate link layer packet from 0012.4b00.0616.0f59, seqno 45
[WARN: CSMA      ] drop duplicate link layer packet from 0012.4b00.0616.0f59, seqno 45
[WARN: CSMA      ] drop duplicate link layer packet from 0012.4b00.0616.0f59, seqno 45
[WARN: CSMA      ] drop duplicate link layer packet from 0012.4b00.0616.0f59, seqno 45
[WARN: CSMA      ] drop duplicate link layer packet from 0012.4b00.0616.0f59, seqno 45
[DBG : RPL       ] DIO Timer triggered
[DBG : RPL       ] DIO Timer interval doubled 15
[INFO: RPL       ] Scheduling DIO timer 2751 ticks in future (Interval)
[DBG : RPL       ] RPL: MOP 2 OCP 1 rank 128 dioint 15, nbr count 1
[DBG : RPL       ] RPL: end of list
[WARN: CSMA      ] drop duplicate link layer packet from 0012.4b00.0616.0f59, seqno 45
[WARN: CSMA      ] drop duplicate link layer packet from 0012.4b00.0616.0f59, seqno 45
```

tinstructor, 2021-03-15:
Additionally, this problem does not seem to occur in release 4.3, which may be useful information

tinstructor, 2021-03-15:
@g-oikonomou update: when using RPL-lite (i.e., the scenario of @m-srk) but with the RPL probing turned off (by adding `#define RPL_CONF_WITH_PROBING 0` to `project-conf.h`), I get the following output on the udp-client terminal. It seems that the udp-client kicks its RPL parent (i.e., the udp-server) because it's too impatient when waiting for a DAO-ACK

```
[INFO: Main      ] Starting Contiki-NG-release/v4.6-110-g3ae1590f6-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 3929
[INFO: Main      ] Link-layer address: 0012.4b00.0616.0f59
[INFO: RPL       ] initializing
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:616:f59
[INFO: Zoul      ] Zolertia RE-Mote revision A platform
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: RPL       ] received a multicast-DIO from fe80::212:4b00:9df:4dee, instance_id 0, DAG ID fd00::212:4b00:9df:4dee, version 240, dtsn 240, rank 128
[INFO: RPL       ] adding global IP address fd00::212:4b00:616:f59
[INFO: RPL       ] reset MRHOF
[INFO: RPL       ] initialized DAG with instance ID 0, DAG ID fd00::212:4b00:9df:4dee, prexix fd00::/64, rank 65535
[WARN: RPL       ] just joined, no parent yet, setting timer for leaving
[INFO: RPL       ] refreshing lifetime
[INFO: RPL       ] parent switch: (NULL IP addr) -> fe80::212:4b00:9df:4dee
[WARN: RPL       ] found parent: fe80::212:4b00:9df:4dee, staying in DAG
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:616:f59, DAG state: joined, MOP 1 OCP 1 rank 512 max-rank 1536, dioint 0, nbr count 1 (Parent switch)
[INFO: RPL       ] nbr: fe80::212:4b00:9df:4dee  128,   384 =>   512 --  0 rba p (no tx)
[INFO: RPL       ] nbr: end of list
[INFO: App       ] Not reachable yet
[INFO: RPL       ] sending a DAO seqno 241, tx count 1, lifetime 30, prefix fd00::212:4b00:616:f59 to fd00::212:4b00:9df:4dee, parent fe80::212:4b00:9df:4dee
[INFO: RPL       ] creating hop-by-hop option
[DBG : RPL       ] no SRH found
[INFO: RPL       ] packet sent to 0012.4b00.09df.4dee, status 2, tx 8, new link metric 601
[INFO: RPL       ] parent switch: fe80::212:4b00:9df:4dee -> (NULL IP addr)
[WARN: RPL       ] no parent, scheduling periodic DIS, will leave if no parent is found
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:616:f59, DAG state: initialized, MOP 1 OCP 1 rank 65535 max-rank 1536, dioint 0, nbr count 1 (Parent switch)
[INFO: RPL       ] nbr: fe80::212:4b00:9df:4dee  128,   601 =>   729 --  8 r  f  (last tx 0 min ago, better since 0 min)
[INFO: RPL       ] nbr: end of list
[INFO: RPL       ] received a DAO-ACK with seqno 241 (241 241) and status 0 from fd00::212:4b00:9df:4dee
```

tinstructor, 2021-03-15:
@g-oikonomou @m-srk This issue seems to be solved by requiring software-generated ACKs when using CSMA in conjunction with the cc1200 radio (much like with the cc13xx-cc26xx). This is easily achieved by setting `#define CSMA_CONF_SEND_SOFT_ACK 1`. However, further investigation is needed to make sure this doesn't create scenarios wherein ACKs are duplicated (i.e., because hardware-generated ACKs suddenly do come through). To reproduce my preliminary fix, you may include the following `project-conf.h` file in the rpl-udp example directory:
```c
#ifndef PROJECT_CONF_H_
#define PROJECT_CONF_H_

#define ZOUL_CONF_USE_CC1200_RADIO  0
#define CSMA_CONF_SEND_SOFT_ACK     1

#define LOG_CONF_LEVEL_RPL      LOG_LEVEL_DBG
#define LOG_CONF_LEVEL_TCPIP    LOG_LEVEL_INFO
#define LOG_CONF_LEVEL_IPV6     LOG_LEVEL_INFO
#define LOG_CONF_LEVEL_MAC      LOG_LEVEL_INFO

#endif /* PROJECT_CONF_H_ */
```
And change the `Makefile` in the rpl-udp example directory to (technically not required):
```
CONTIKI_PROJECT = udp-client udp-server
all: $(CONTIKI_PROJECT)

CONTIKI=../..

MAKE_MAC = MAKE_MAC_CSMA
MAKE_NET = MAKE_NET_IPV6
# MAKE_ROUTING = MAKE_ROUTING_RPL_CLASSIC

include $(CONTIKI)/Makefile.include
```
This gives the following terminal output at the udp-client node for RPL-lite:
```bash
$ make BOARD=remote-reva PORT=/dev/ttyUSB0 login
using saved target 'zoul'
rlwrap ../../tools/serial-io/serialdump -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 [OK]
[INFO: Main      ] Starting Contiki-NG-release/v4.6-110-g3ae1590f6-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 3929
[INFO: Main      ] Link-layer address: 0012.4b00.0616.0f59
[INFO: IPv6 DS   ] Init: 16 neighbors
[INFO: IPv6 DS   ] 2 default routers
[INFO: IPv6 DS   ] 3 prefixes
[INFO: IPv6 DS   ] 0 routes
[INFO: IPv6 DS   ] 3 unicast addresses
[INFO: IPv6 DS   ] 7 multicast addresses
[INFO: IPv6 DS   ] 2 anycast addresses
[INFO: IPv6 DS   ] Adding prefix fe80::length 64, flags 0, Valid lifetime 0, Preffered lifetime 0
[INFO: RPL       ] initializing
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:616:f59
[INFO: Zoul      ] Zolertia RE-Mote revision A platform
[INFO: CSMA      ] received packet from 0012.4b00.09df.4dee, seqno 122, len 80
[INFO: TCP/IP    ] input: received 116 bytes
[INFO: IPv6      ] packet received from fe80::212:4b00:9df:4dee to ff02::1a
[INFO: IPv6      ] icmpv6 input length 116 type: 155 
[INFO: RPL       ] received a multicast-DIO from fe80::212:4b00:9df:4dee, instance_id 0, DAG ID fd00::212:4b00:9df:4dee, version 240, dtsn 240, rank 128
[INFO: RPL       ] adding global IP address fd00::212:4b00:616:f59
[INFO: RPL       ] reset MRHOF
[INFO: RPL       ] initialized DAG with instance ID 0, DAG ID fd00::212:4b00:9df:4dee, prexix fd00::/64, rank 65535
[WARN: RPL       ] just joined, no parent yet, setting timer for leaving
[INFO: IPv6 Nbr  ] Adding neighbor with ip addr fe80::212:4b00:9df:4dee link addr 0012.4b00.09df.4dee state 1
[INFO: RPL       ] refreshing lifetime
[INFO: RPL       ] best parent is not fresh, schedule urgent probing to fe80::212:4b00:9df:4dee
[INFO: RPL       ] probing fe80::212:4b00:9df:4dee (urgent) last tx 0 min ago
[INFO: RPL       ] sending a unicast-DIO with rank 65535 to fe80::212:4b00:9df:4dee
[INFO: ICMPv6    ] Sending ICMPv6 packet to fe80::212:4b00:9df:4dee, type 155, code 1, len 72
[INFO: TCP/IP    ] output: processing 116 bytes packet from fe80::212:4b00:616:f59 to fe80::212:4b00:9df:4dee
[INFO: TCP/IP    ] output: destination is on link
[... 77 lines cut ...]
[INFO: TCP/IP    ] output: sending to (NULL LL addr)
[INFO: CSMA      ] sending to 0000.0000.0000.0000, len 80, seqno 121, queue length 1, free packets 7
[INFO: CSMA      ] preparing packet for 0000.0000.0000.0000, seqno 121, tx 0, queue 1
[INFO: CSMA      ] tx to 0000.0000.0000.0000, seqno 121, status 0, tx 0, coll 0
[INFO: CSMA      ] packet sent to 0000.0000.0000.0000, seqno 121, status 0, tx 1, coll 0
[INFO: App       ] Sending request 0 to fd00::212:4b00:9df:4dee
[INFO: IPv6      ] Sending packet with length 55 (15)
[INFO: RPL       ] creating hop-by-hop option
[INFO: TCP/IP    ] output: processing 63 bytes packet from fd00::212:4b00:616:f59 to fd00::212:4b00:9df:4dee
[DBG : RPL       ] no SRH found
[INFO: IPv6 Route] Default route, IP address fe80::212:4b00:9df:4dee
[INFO: IPv6 Route] Default route found, IP address fe80::212:4b00:9df:4dee
[INFO: TCP/IP    ] output: no route found, using default route: fe80::212:4b00:9df:4dee
[INFO: TCP/IP    ] output: sending to 0012.4b00.09df.4dee
[INFO: CSMA      ] sending to 0012.4b00.09df.4dee, len 25, seqno 122, queue length 1, free packets 7
[INFO: CSMA      ] preparing packet for 0012.4b00.09df.4dee, seqno 122, tx 0, queue 1
[INFO: CSMA      ] tx to 0012.4b00.09df.4dee, seqno 122, status 0, tx 0, coll 0
[INFO: CSMA      ] packet sent to 0012.4b00.09df.4dee, seqno 122, status 0, tx 1, coll 0
[INFO: RPL       ] packet sent to 0012.4b00.09df.4dee, status 0, tx 1, new link metric 128
[INFO: CSMA      ] received packet from 0012.4b00.09df.4dee, seqno 125, len 25
[INFO: TCP/IP    ] input: received 63 bytes
[INFO: IPv6      ] packet received from fd00::212:4b00:9df:4dee to fd00::212:4b00:616:f59
[INFO: IPv6      ] Receiving UDP packet
[INFO: App       ] Received response 'hello 0' from fd00::212:4b00:9df:4dee
```

tinstructor, 2021-03-15:
I propose to turn off hardware auto-ACKs for the cc1200 by default and make software-based ACKs the default (for now) so that the cc1200 may actually be used with RPL and CSMA. I shall abstract the need for software ACKs from people who mainly use Contiki-NG to write examples (**\*coughs\***) and make a pull request.

Conceivably, turning off hardware-ACKs for the cc1200 by default does not conflict with the TSCH MAC layer because that uses software ACKs anyway?

atiselsts, 2021-03-15:
Now I've never used the CC1200 radio myself, but I think if the config has `CSMA_CONF_SEND_SOFT_ACK` set to 1, then CSMA code also should to this to tell the radio driver to disable autoack:
```
NETSTACK_RADIO.get_value(RADIO_PARAM_RX_MODE, &radio_rx_mode);
radio_rx_mode &= ~RADIO_RX_MODE_AUTOACK;
NETSTACK_RADIO.set_value(RADIO_PARAM_RX_MODE, radio_rx_mode);
```

To me, this looks like a PR in CSMA.

Further, you need to change `CSMA_CONF_ACK_WAIT_TIME` - the default value is suitable for 250 kbps mode. Something like `RTIMER_SECOND / 300` would be ok as the value of  `CSMA_CONF_ACK_WAIT_TIME` for 50 kbps radios.

This could be another PR, in the CC1200 driver code.

TSCH is not affected as you say.

g-oikonomou, 2021-03-15:
I'll need to look at this in detail, but my immediate reaction is: The CC1200 has some automatic hardware ACK support, but it is not exactly the same as what 2.4GHz radios do. From what I've seen from other sub-ghz radios, I would have expected us to always use soft ACKs for the CC1200. Is this not the case?

tinstructor, 2021-03-15:
> Now I've never used the CC1200 radio myself, but I think if the config has `CSMA_CONF_SEND_SOFT_ACK` set to 1, then CSMA code also should to this to tell the radio driver to disable autoack:
>
> ``` [...]

Concerning an appropriate `CSMA_CONF_ACK_WAIT_TIME`, this handled by using `#define ZOUL_CONF_USE_CC1200_RADIO  1`. And concerning CSMA, given that it does work for the cc2538 I thought it'd be more appropriate to turn of hardware auto-ACKs in `cc1200.c`:

https://github.com/contiki-ng/contiki-ng/blob/3ae1590f6a76e5612fec6917c35483cd49e81caa/arch/dev/radio/cc1200/cc1200.c#L681

by changing this line to 
```c
 /* Enable address filtering + disable auto ack */
    rx_mode_value = (~RADIO_RX_MODE_AUTOACK | RADIO_RX_MODE_ADDRESS_FILTER);
```

tinstructor, 2021-03-15:
I just noticed I may have made a critical error so before we continue this discussion, let me first check that

atiselsts, 2021-03-15:
Whatever the conclusion for the rest of the discussion is, the CSMA change should be applied - it's just sensible defensive programming, and cannot possibly break anything as long as the radio driver is sane.

I've made a PR for this: https://github.com/contiki-ng/contiki-ng/pull/1531

tinstructor, 2021-03-15:
@atiselsts @g-oikonomou any ideas as to what may cause these issues with the cc1200? I'm sort of out of inspiration atm

> I'll need to look at this in detail, but my immediate reaction is: The CC1200 has some automatic hardware ACK support, but it is not exactly the same as what 2.4GHz radios do. From what I've seen [...]

This is indeed not the case it seems. Not that it matters, because it doesn't seem to fix anything anyway

tinstructor, 2021-03-18:
@atiselsts @g-oikonomou As indicated by @m-srk there is indeed an ACK-related problem here. However, the problem is not so much that the udp-client is not receiving the ACKs to DIO probes, but that the ACKs are received much later than anticipated, causing them to be parsed by [`input_packet()`](https://github.com/contiki-ng/contiki-ng/blob/3ae1590f6a76e5612fec6917c35483cd49e81caa/os/net/mac/csma/csma.c#L75-L121) as if they were an ordinary packet rather than immediately after sending the probes within the scope of [`send_one_packet()`](https://github.com/contiki-ng/contiki-ng/blob/3ae1590f6a76e5612fec6917c35483cd49e81caa/os/net/mac/csma/csma-output.c#L169-L254).  This of course causes the incoming ACK to be ignored as indicated here:

https://github.com/contiki-ng/contiki-ng/blob/3ae1590f6a76e5612fec6917c35483cd49e81caa/os/net/mac/csma/csma.c#L75-L83

Naturally, I immediately thought of playing with `CSMA_CONF_ACK_WAIT_TIME` and `CSMA_CONF_AFTER_ACK_DETECTED_WAIT_TIME` such that the ACK is in fact detected and received before [`send_one_packet()`](https://github.com/contiki-ng/contiki-ng/blob/3ae1590f6a76e5612fec6917c35483cd49e81caa/os/net/mac/csma/csma-output.c#L169-L254) returns and incoming ACKs are treated as ordinary packets and thus go through [`input_packet()`](https://github.com/contiki-ng/contiki-ng/blob/3ae1590f6a76e5612fec6917c35483cd49e81caa/os/net/mac/csma/csma.c#L75-L121) (which, again, causes them to be ignored). Given that I added the following [`project-conf.h`](https://github.com/tinstructor/contiki-ng/blob/9a719ff8ebd49b888d7de18a16a8e38516fbe53c/examples/rpl-udp/project-conf.h) file to the rpl-udp example:

```c
#ifndef PROJECT_CONF_H_
#define PROJECT_CONF_H_

#define ZOUL_CONF_USE_CC1200_RADIO          1
#define IEEE802154_CONF_DEFAULT_CHANNEL     2

#define LOG_CONF_LEVEL_RPL      LOG_LEVEL_INFO
#define LOG_CONF_LEVEL_TCPIP    LOG_LEVEL_INFO
#define LOG_CONF_LEVEL_IPV6     LOG_LEVEL_INFO
#define LOG_CONF_LEVEL_MAC      LOG_LEVEL_DBG

#endif /* PROJECT_CONF_H_ */
```

I changed around these constants here:

https://github.com/contiki-ng/contiki-ng/blob/3ae1590f6a76e5612fec6917c35483cd49e81caa/arch/platform/zoul/contiki-conf.h#L126-L127

However, after unsuccessfully trying countless combinations of these values to the point they no longer made sense, I was kind of stuck for a while. So then I went on a little witch-hunt here and I've noticed an issue with the following line when the cc1200 is in use:

https://github.com/contiki-ng/contiki-ng/blob/3ae1590f6a76e5612fec6917c35483cd49e81caa/os/net/mac/csma/csma-output.c#L206

When I changed it to

```c
      radio_result_t foo = NETSTACK_RADIO.transmit(packetbuf_totlen());
      RTIMER_BUSYWAIT(RTIMER_SECOND / 200);
      switch(foo) {
```

And made [these changes to cc1200.c](https://github.com/tinstructor/contiki-ng/commit/9a719ff8ebd49b888d7de18a16a8e38516fbe53c?branch=9a719ff8ebd49b888d7de18a16a8e38516fbe53c&diff=split#diff-a48d6eedc6a5b20f5848e18ef878a9e97f243716847d738b182aa9a5dea36abc) to mimic the more strict timing requirements for the TSCH MAC, for some reason (related to timing presumably), ACKs to DIO probes are received correctly and operation of the example proceeds as expected. It is however besides me why this "works" at all and it is probably a good idea to look deeper into it.

Everything you need to replicate my setup can be found[ in this branch](https://github.com/tinstructor/contiki-ng/tree/cc1200_csma_troubles). Normally, I'd make a pull request but as I'm almost certain that this is not a proper fix, I've resorted to just linking the branch here for now.