contiki-ng/contiki-ng #1231: Compiling cooja mote as 32-bit

philok93, 2020-04-17:
Hi all,
I want to simulate udp-server (from rpl-udp example) as cooja mote so that doesn't have limitations. Is that possible in a 64bit linux to build 32-bit cooja mote? Actually the output of compiler is the following:

```
> make udp-server.cooja TARGET=cooja
  CC        udp-server.c
  MKDIR     build/cooja/server/obj
  CC        ../../arch/platform/cooja/./platform.c
mv platform.o build/cooja/server/mtype619.o
  CC        ../../arch/platform/cooja/lib/simEnvChange.c
  CC        ../../arch/platform/cooja/sys/cooja_mt.c
  CC        ../../arch/platform/cooja/sys/cooja_mtarch.c
  CC        ../../arch/platform/cooja/./rtimer-arch.c
  CC        ../../os/dev/slip.c
  CC        ../../arch/platform/cooja/dev/watchdog.c
  CC        ../../arch/platform/cooja/./int-master.c
  CC        ../../arch/platform/cooja/dev/beep.c
  CC        ../../arch/platform/cooja/dev/ip.c
  CC        ../../arch/platform/cooja/dev/leds-arch.c
  CC        ../../arch/platform/cooja/dev/moteid.c
  CC        ../../arch/platform/cooja/dev/pir-sensor.c
  CC        ../../arch/platform/cooja/dev/rs232.c
  CC        ../../arch/platform/cooja/dev/vib-sensor.c
  CC        ../../arch/platform/cooja/sys/clock.c
  CC        ../../arch/platform/cooja/sys/cooja-log.c
  CC        ../../arch/platform/cooja/cfs/cfs-cooja.c
  CC        ../../arch/platform/cooja/dev/cooja-radio.c
  CC        ../../arch/platform/cooja/dev/eeprom.c
  CC        ../../arch/platform/cooja/./slip-arch.c
  CC        ../../os/lib/random.c
  CC        ../../os/lib/sensors.c
  CC        ../../os/dev/leds.c
  CC        ../../arch/platform/cooja/dev/gpio-hal-arch.c
  CC        ../../arch/platform/cooja/dev/buttons.c
  CC        ../../os/net/routing/ids-app/ids.c
  CC        ../../os/contiki-main.c
  CC        ../../os/sys/autostart.c
  CC        ../../os/sys/ctimer.c
  CC        ../../os/sys/energest.c
  CC        ../../os/sys/etimer.c
  CC        ../../os/sys/log.c
  CC        ../../os/sys/atomic.c
  CC        ../../os/sys/mutex.c
  CC        ../../os/sys/stimer.c
  CC        ../../os/sys/compower.c
  CC        ../../os/sys/process.c
  CC        ../../os/sys/rtimer.c
  CC        ../../os/sys/node-id.c
  CC        ../../os/sys/timer.c
  CC        ../../os/sys/stack-check.c
  CC        ../../os/lib/dbl-circ-list.c
  CC        ../../os/lib/crc16.c
  CC        ../../os/lib/ifft.c
  CC        ../../os/lib/ccm-star.c
  CC        ../../os/lib/dbl-list.c
  CC        ../../os/lib/ringbufindex.c
  CC        ../../os/lib/list.c
  CC        ../../os/lib/circular-list.c
  CC        ../../os/lib/aes-128.c
  CC        ../../os/lib/trickle-timer.c
  CC        ../../os/lib/memb.c
  CC        ../../os/lib/heapmem.c
  CC        ../../os/lib/assert.c
  CC        ../../os/lib/ringbuf.c
  CC        ../../os/dev/button-hal.c
  CC        ../../os/dev/serial-line.c
  CC        ../../os/dev/spi.c
  CC        ../../os/dev/nullradio.c
  CC        ../../os/dev/gpio-hal.c
  CC        ../../os/net/queuebuf.c
[... 43 lines cut ...]
  CC        ../../os/net/routing/rpl-lite/rpl-nbr-policy.c
  CC        ../../os/net/routing/rpl-lite/rpl-mrhof.c
  CC        ../../os/net/routing/rpl-lite/rpl-dag-root.c
ar rcf build/cooja/mtype619.a build/cooja/server/obj/simEnvChange.o build/cooja/server/obj/cooja_mt.o build/cooja/server/obj/cooja_mtarch.o build/cooja/server/obj/rtimer-arch.o build/cooja/server/obj/slip.o build/cooja/server/obj/watchdog.o build/cooja/server/obj/int-master.o build/cooja/server/obj/beep.o build/cooja/server/obj/ip.o build/cooja/server/obj/leds-arch.o build/cooja/server/obj/moteid.o build/cooja/server/obj/pir-sensor.o build/cooja/server/obj/rs232.o build/cooja/server/obj/vib-sensor.o build/cooja/server/obj/clock.o build/cooja/server/obj/cooja-log.o build/cooja/server/obj/cfs-cooja.o build/cooja/server/obj/cooja-radio.o build/cooja/server/obj/eeprom.o build/cooja/server/obj/slip-arch.o build/cooja/server/obj/random.o build/cooja/server/obj/sensors.o build/cooja/server/obj/leds.o build/cooja/server/obj/gpio-hal-arch.o build/cooja/server/obj/buttons.o build/cooja/server/obj/ids.o build/cooja/server/obj/contiki-main.o build/cooja/server/obj/autostart.o build/cooja/server/obj/ctimer.o build/cooja/server/obj/energest.o build/cooja/server/obj/etimer.o build/cooja/server/obj/log.o build/cooja/server/obj/atomic.o build/cooja/server/obj/mutex.o build/cooja/server/obj/stimer.o build/cooja/server/obj/compower.o build/cooja/server/obj/process.o build/cooja/server/obj/rtimer.o build/cooja/server/obj/node-id.o build/cooja/server/obj/timer.o build/cooja/server/obj/stack-check.o build/cooja/server/obj/dbl-circ-list.o build/cooja/server/obj/crc16.o build/cooja/server/obj/ifft.o build/cooja/server/obj/ccm-star.o build/cooja/server/obj/dbl-list.o build/cooja/server/obj/ringbufindex.o build/cooja/server/obj/list.o build/cooja/server/obj/circular-list.o build/cooja/server/obj/aes-128.o build/cooja/server/obj/trickle-timer.o build/cooja/server/obj/memb.o build/cooja/server/obj/heapmem.o build/cooja/server/obj/assert.o build/cooja/server/obj/ringbuf.o build/cooja/server/obj/button-hal.o build/cooja/server/obj/serial-line.o build/cooja/server/obj/spi.o build/cooja/server/obj/nullradio.o build/cooja/server/obj/gpio-hal.o build/cooja/server/obj/queuebuf.o build/cooja/server/obj/nbr-table.o build/cooja/server/obj/packetbuf.o build/cooja/server/obj/netstack.o build/cooja/server/obj/linkaddr.o build/cooja/server/obj/net-debug.o build/cooja/server/obj/link-stats.o build/cooja/server/obj/mac-sequence.o build/cooja/server/obj/mac.o build/cooja/server/obj/frame802154e-ie.o build/cooja/server/obj/framer-802154.o build/cooja/server/obj/nullframer.o build/cooja/server/obj/frame802154.o build/cooja/server/obj/csma-security.o build/cooja/server/obj/anti-replay.o build/cooja/server/obj/csma.o build/cooja/server/obj/csma-output.o build/cooja/server/obj/ccm-star-packetbuf.o build/cooja/server/obj/uipbuf.o build/cooja/server/obj/uip-icmp6.o build/cooja/server/obj/uip-sr.o build/cooja/server/obj/uip-ds6.o build/cooja/server/obj/uip-packetqueue.o build/cooja/server/obj/tcp-socket.o build/cooja/server/obj/ip64-addr.o build/cooja/server/obj/uip-udp-packet.o build/cooja/server/obj/uip-ds6-route.o build/cooja/server/obj/psock.o build/cooja/server/obj/uip-ds6-nbr.o build/cooja/server/obj/tcpip.o build/cooja/server/obj/uip6.o build/cooja/server/obj/uip-nameserver.o build/cooja/server/obj/uip-nd6.o build/cooja/server/obj/udp-socket.o build/cooja/server/obj/simple-udp.o build/cooja/server/obj/sicslowpan.o build/cooja/server/obj/uiplib.o build/cooja/server/obj/rpl-icmp6.o build/cooja/server/obj/rpl-of0.o build/cooja/server/obj/rpl-neighbor.o build/cooja/server/obj/rpl-dag.o build/cooja/server/obj/rpl-timers.o build/cooja/server/obj/rpl-ext-header.o build/cooja/server/obj/rpl.o build/cooja/server/obj/rpl-nbr-policy.o build/cooja/server/obj/rpl-mrhof.o build/cooja/server/obj/rpl-dag-root.o 
Redefining printf
objcopy --redefine-sym printf=log_printf udp-server.o;   objcopy --redefine-sym printf=log_printf build/cooja/server/mtype619.o;   objcopy --redefine-sym printf=log_printf build/cooja/server/mtype619.a; 
objcopy: 'build/cooja/server/mtype619.a': No such file
../../arch/platform/cooja/Makefile.customrules-cooja:21: recipe for target 'build/cooja/server/mtype619.cooja' failed
make: [build/cooja/server/mtype619.cooja] Error 1 (ignored)
objcopy --redefine-sym puts=log_puts udp-server.o;   objcopy --redefine-sym puts=log_puts build/cooja/server/mtype619.o;   objcopy --redefine-sym puts=log_puts build/cooja/server/mtype619.a; 
objcopy: 'build/cooja/server/mtype619.a': No such file
../../arch/platform/cooja/Makefile.customrules-cooja:21: recipe for target 'build/cooja/server/mtype619.cooja' failed
make: [build/cooja/server/mtype619.cooja] Error 1 (ignored)
objcopy --redefine-sym putchar=log_putchar udp-server.o;   objcopy --redefine-sym putchar=log_putchar build/cooja/server/mtype619.o;   objcopy --redefine-sym putchar=log_putchar build/cooja/server/mtype619.a; 
objcopy: 'build/cooja/server/mtype619.a': No such file
../../arch/platform/cooja/Makefile.customrules-cooja:21: recipe for target 'build/cooja/server/mtype619.cooja' failed
gcc -m32 -I'/usr/lib/jvm/java-8-oracle/include' -I'/usr/lib/jvm/java-8-oracle/include/linux' -shared -Wl,-Map=build/cooja/mtype619.map -o build/cooja/mtype619.cooja udp-server.o build/cooja/server/mtype619.o build/cooja/server/mtype619.a 
make: [build/cooja/server/mtype619.cooja] Error 1 (ignored)
gcc: error: build/cooja/server/mtype619.a: No such file or directory
../../arch/platform/cooja/Makefile.customrules-cooja:21: recipe for target 'build/cooja/server/mtype619.cooja' failed
make: *** [build/cooja/server/mtype619.cooja] Error 1
Process returned error code 2
```

It seems that files are not compiled at the end. What should be a fix for this? Thank you.

alexrayne, 2020-05-09:
try this  [PR#1211](https://github.com/contiki-ng/contiki-ng/pull/1211)

alexrayne, 2020-05-09:
or better you should exprlore why 

> ar rcf build/cooja/mtype619.a ....
does generated  `build/cooja/mtype619.a` other rules expects `build/cooja/server/mtype619.a`

joakimeriksson, 2022-06-22:
Closing as Cooja now supports 64-bits.

closed 2022-06-22 by joakimeriksson