contiki-ng/contiki-ng #1235: powertrace in contiki-ng

jabberimaher, 2020-04-19:
hello,
i want to dispaly energy consumption using powertrace in contiki-ng, it's possible
best regards

rajeev1986, 2020-04-20:
Hi @jabberimaher,
You can use the following to compute current consumption and later power by muitlying with the input voltage:

```
avg_current_cons = ((delta_tx * current_radio_tx) + (delta_rx * current_radio_rx) + 
(delta_cpu * current_cpu) + (delta_lpm * current_lpm) + 
(delta_deep_lpm * current_deep_lpm))/ (RTIMER_ARCH_SECOND * (delta_time/ENERGEST_SECOND));
```

jabberimaher, 2020-04-21:
Thank you **rajeev1986** for your reply, but firstly i need to know how to configure powertrace in contiki-ng to display data or save a file containing informations about enerfor example, if i want to dispaly enrgy consumption of rpl-udp project, i ask how to modify and configure files of rpl-udp (makefile, confi.h...).
thank you 
I

rajeev1986, 2020-04-21:
Hello there @jabberimaher,
So there are two ways to activate the energest module in Contiki-NG.
- In your Makefile add this line: `MODULES += os/services/simple-energest` **or** 

- In the project-conf.h add `#define ENERGEST_CONF_ON      1`

either of the above will activate the energest logging after you recompile the code and will display the energest data every **60 seconds** as configured by default in `simple-energest.h`
You can change the `#define SIMPLE_ENERGEST_PERIOD (CLOCK_SECOND * 60)` to other values as you like to periodically log the data.

Let me know how it goes....

jabberimaher, 2020-04-22:
Hello **rajeev1986**
i try energest with hello-wols axample, it works nad dispaly, but by the same way (In your Makefile add this line: MODULES += os/services/simple-energest) with rpl-udp-client.z1 and rpl-udp-sever.z1 don't dispaly anything.(first issue)....
the second issue i need to implement aes128 and sha1 and estimate energy consumption  of node when executing them..... is demand too how to add them in rpl-udp ..the idea is to estimate energy in rpl-udp befoe adding aes/sha1  and afte and see the difference.
i am really sorry to have disturbed you Sir, but i need to achieve a paper before 21 june.
best regards

rajeev1986, 2020-04-23:
Hi @jabberimaher ,
By default, `SIMPLE_ENERGEST_CONF_LOG_LEVEL` is set to `LOG_LEVEL_NONE`, meaning it will not print out the logging info. So you also need to add the following in your ` project-conf.h` 
`#define SIMPLE_ENERGEST_CONF_LOG_LEVEL             LOG_LEVEL_INFO`

After that execute `make clean` and then build your firmware. Now you should see energest log info every 60 seconds.

Regarding aes128 and sha1, I am not aware of that. However, the energy demand to do the computation should already reflect in the CPU on time which is computed by the energest as `delta_cpu`.  More computation means the CPU was on for longer period...

jabberimaher, 2020-04-28:
Hello rajeev1986,
i am very grateful for your help, one last request please, energest displays information every 60 seconds (cpu, lpm tx, rx), how to use this information to calculate the energy consumption in joules or watts ...... and it is possible to use energest to display the execution time.

yatch, 2020-04-29:
@jabberimaher Measure the power consumption of your board in each mode (energest type), then you can do simple math.

https://github.com/contiki-ng/contiki-ng/wiki/Documentation:-Energest

> By tracking the time various hardware components such as the radio is turned on, and by knowing the power consumption of the component, it is possible to estimate the energy consumption.

@rajeev1986 Thank you for your contribution here.