contiki-ng/contiki-ng #1237: promiscuous mode

iliar-phd, 2020-04-20:
Hello,

I need to enable the promiscuous mode on a Z1 mote in Cooja (not on real mote). Actually my end goal is to measure some stats at the sniffer and send them to the sink.
I tried to change the sensniff example and to know which part of code is executed when a packet is received.
I did put some printf commands in the <<char_in(unsigned char c)>> callback function and after the PROCESS_YIELD function as below, but they did not run.

`while(1) {
    printf("in the while:\n");
    PROCESS_YIELD();

    if(ev == PROCESS_EVENT_POLL) {
      process_incoming_data();
    }
  }
`
Any idea?

alexrayne, 2020-04-22:
but who polls your process? somebody need call process_poll to `if(ev == PROCESS_EVENT_POLL)` see incoming poll event

g-oikonomou, 2020-05-09:
On real hardware what happens is the sensniff example sets the radio in promiscuous mode by calling `NETSTACK_RADIO.set_value(RADIO_PARAM_RX_MODE, 0)`. This is supported by the CC2420 radio driver (which is what the Z1 uses).

When the radio picks up a frame, what will eventually happen is `sensniff_output_frame()`.

The `char_in()`callback, which is what polls the process, will only get polled when the device receives data over the serial line, which is how the sensniff python scripts sends commands to the sniffer e.g. to change to a different channel.

All of that on real hardware.

Inside Cooja I have only ever run those things back in the day when the sniffing was implemented inside the actual radio driver, before I wrote this particular example. I would first check `sensniff_output_frame()` as above and I would also see what is going on inside `cc2420_process()` in `cc2420.c`. See if you can see frame reception activity there.

closed 2020-11-25 by g-oikonomou