contiki-ng/contiki-ng #1241: cant run tcpdump in contiki-ng container

alexrayne, 2020-04-22:
Hallow!
run docker container on ubutnu19.10 host. 
inside container run `tcpdump` answer:
```
tcpdump -i tun0 -n -q
tcpdump: error while loading shared libraries: libcrypto.so.1.0.0: cannot open shared object file: Permission denied
```

in `/user/lib/i386-linux-gnu` have:
```
libcrypt.a			  
libcrypt.so		
libcrypto.a		
libcrypto.so		
libcups.so.2
```

in ldcache have:
```
libcrypto.so.1.0.0 (libc6) => /lib/i386-linux-gnu/libcrypto.so.1.0.0
	libcrypto.so (libc6) => /usr/lib/i386-linux-gnu/libcrypto.so
	libcrypt.so.1 (libc6, OS ABI: Linux 2.6.32) => /lib/i386-linux-gnu/libcrypt.so.1
	libcrypt.so (libc6, OS ABI: Linux 2.6.32) => /usr/lib/i386-linux-gnu/libcrypt.so
```

copy or sym-link from libcrypt.so -> libcrypt.so.1.0.0 doesn`t help.
what i can helps?

yatch, 2020-04-29:
`contiker/contiki-ng` doesn't have `tcpdump` installed in the first place. So, you may be using a different container.

```
$ docker run -ti contiker/contiki-ng apt list --installed | grep tcpdump
```

closed 2020-04-29 by yatch

alexrayne, 2020-04-29:
> `contiker/contiki-ng` doesn't have `tcpdump` installed in the first place. So, you may be using a different container.
> 
> ```
> $ docker run -ti contiker/contiki-ng apt list --installed | grep tcpdump
> ```

Yes, therefore i have install it in running container. And even when it installed - not works. Reason was described before.