contiki-ng/contiki-ng #1253: AES128 in contiki-ng

jabberimaher, 2020-04-29:
Hi,
please help me to add and implement AES128 in contiki-ng under rpl-udp project (encryption(udp-client)/decryption(server)).
best regards

alexrayne, 2020-05-09:
what platform?

jabberimaher, 2020-05-09:
thank you for your reply
but what do you mean with platform?

alexrayne, 2020-05-09:
for what device/cpu you want implement AES128 support? for most devices it is alredy has.

jabberimaher, 2020-05-09:
i want to implement aes128 on mote z1, i want to use cooja simulator to measure energy consumption when z1 execute encryption / decryption aes128, i want to know how to add aes128 to udp-client.c and udp-server.c (under examples / rpl-udp).

alexrayne, 2020-05-13:
i don`t see nor in z1, nor in msp430cpu any support of hardware AES - therefore, it wil be used software one. BUT software, on my exploration, was too heavy for cpu even on ARM32 of cc1310 (about 80MHz) - it takes a few ms to encode single frame.

And event for such software sim, you need to force cooja use mspsim to accurately simulate this. For generic c simulation cooja have no instrumentation to evaluate CPU usage.

atiselsts, 2020-05-13:
Hi @jabberimaher, you can enable IEEE 802.15.4 standard encryption and authentication in Contiki-NG. it will use AES-128 under the hood.

To do that, add a `project-conf.h` file to the examples/rpl-udp/ with the following contents:

    #define LLSEC802154_CONF_ENABLED 1

You should also redefine the value of `CSMA_CONF_LLSEC_DEFAULT_KEY0` if you want to use a secret encryption key. That is for CSMA, the keys used by TSCH are `TSCH_SECURITY_CONF_K1` and `TSCH_SECURITY_CONF_K2`.

Be aware that Cooja / mspsim does not implement hardware acceleration for AES-128, so the operations will take quite a lot of time. All actual hardware platforms supported by Contiki-NG **do have** hardware acceleration, even though some (such as Simplelink) as the moment seem to lack a driver that would enable the use of these hardware features.

closed 2020-05-13 by atiselsts