contiki-ng/contiki-ng #1264: tun0 error when trying to set prefix

mwm93, 2020-05-15:
Hello,

I'm tryin to get rpl-border-ruter to work.

When I compile it (with make distclean && make TARGET=zoul BOARD=remote-revb PORT=/dev/ttyUSB0 border-router.upload login) and put it to a zolertia ReMote, Revision B, it is waiting for the prefix, as it should:

```
  CC        border-router.c
  LD        build/zoul/remote-revb/border-router.elf
  OBJCOPY   build/zoul/remote-revb/border-router.elf --> build/zoul/remote-revb/border-router.bin
Flashing /dev/ttyUSB0
Opening port /dev/ttyUSB0, baud 460800
Reading data from build/zoul/remote-revb/border-router.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
CC2538 PG2.0: 512KB Flash, 32KB SRAM, CCFG at 0x0027FFD4
Primary IEEE Address: 00:12:4B:00:18:D6:F8:17
Erasing 524288 bytes starting at address 0x00200000
    Erase done
Writing 516096 bytes starting at address 0x00202000
Write 8 bytes at 0x0027FFF8F00
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xf2e95d89)
rlwrap ../../tools/serial-io/serialdump -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 [OK]
[INFO: Main      ] Starting Contiki-NG-release/v4.4-97-g55dc08aab-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 63511
[INFO: Main      ] Link-layer address: 0012.4b00.18d6.f817
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:18d6:f817
[INFO: Zoul      ] Zolertia RE-Mote revision B platform
[INFO: RPL BR    ] Contiki-NG Border Router started
[INFO: BR        ] RPL-Border router started
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
```

But when I try to set the prefix, the following occurs: 

```
# ./setprefix.sh 
********SLIP started on ``/dev/ttyUSB0''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` mtu 1500 up
ifconfig tun0 add fd::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 127.0.1.1  netmask 255.255.255.255  destination 127.0.1.1
        inet6 fe80::1  prefixlen 64  scopeid 0x20<link>
        inet6 fd::1  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::bb19:413d:1e37:634b  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

tunslip6: serial_to_tun: read: Success
ifconfig tun0 down
netstat -nr | awk '{ if ($2 == "tun0") print "route delete -net "$1; }' | sh
```

The Border is still waiting to get the prefix. The script setprefix.sh simply contains:

`sudo ../../tools/serial-io/tunslip6 -s /dev/ttyUSB0 fd::1/64`

Any suggestions? Thanks in advance.

g-oikonomou, 2020-05-15:
Instead of running setprefix, what happens if you run the tunslip6 command?

mwm93, 2020-05-15:
Than it shows the following:

  CC        border-router.c
  LD        build/zoul/remote-revb/border-router.elf
  OBJCOPY   build/zoul/remote-revb/border-router.elf --> build/zoul/remote-revb/border-router.bin
Flashing /dev/ttyUSB0
Opening port /dev/ttyUSB0, baud 460800
Reading data from build/zoul/remote-revb/border-router.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
CC2538 PG2.0: 512KB Flash, 32KB SRAM, CCFG at 0x0027FFD4
Primary IEEE Address: 00:12:4B:00:18:D6:F8:17
Erasing 524288 bytes starting at address 0x00200000
    Erase done
Writing 516096 bytes starting at address 0x00202000
Write 8 bytes at 0x0027FFF8F00
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xf2e95d89)
rlwrap ../../tools/serial-io/serialdump -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 [OK]
[INFO: Main      ] Starting Contiki-NG-release/v4.4-97-g55dc08aab-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 63511
[INFO: Main      ] Link-layer address: 0012.4b00.18d6.f817
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:18d6:f817
[INFO: Zoul      ] Zolertia RE-Mote revision B platform
[INFO: RPL BR    ] Contiki-NG Border Router started
[INFO: BR        ] RPL-Border router started
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Waiting for prefix
��?P�serial device disconnected: Bad file descriptor
../../Makefile.embedded:36: recipe for target 'login' failed
make: *** [login] Error 255
rm border-router.o build/zoul/remote-revb/obj/startup-gcc.o

ping6 instead shows answers:
# ping6 fd00::212:4b00:18d6:f817
PING fd00::212:4b00:18d6:f817(fd00::212:4b00:18d6:f817) 56 data bytes
64 bytes from fd00::212:4b00:18d6:f817: icmp_seq=1 ttl=64 time=19.8 ms
64 bytes from fd00::212:4b00:18d6:f817: icmp_seq=2 ttl=64 time=19.8 ms
64 bytes from fd00::212:4b00:18d6:f817: icmp_seq=3 ttl=64 time=19.9 ms
^C
--- fd00::212:4b00:18d6:f817 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2002ms
rtt min/avg/max/mdev = 19.868/19.897/19.951/0.167 ms

Now i can even ping and fetch an index.html from it. Thank you. I will leave the issue open if i run into problems with it again. 

But why doesnt it work with setprefix or why does it disconnect?

g-oikonomou, 2020-05-15:
I just noticed the "login" part of your make command. You cannot combine `make login` with `tunslip6` since that results in two separate processes trying to write to the same character device. Try removing the `login` part from your make command.

closed 2020-09-07 by mwm93