contiki-ng/contiki-ng #1267: tsch channel selection replaces channels with no interference

ChristianMorup, 2020-05-18:
If you create a network consisting of only one node in Cooja as a tsch coordinator, and calls tsch_cs_adaptations_init, then it will evaluate several channels (and always the same channels) to be busy even when they aren't used. Thus, several channels are replaced even if they shouldn't be.

alexrayne, 2020-05-18:
may you provide sample project, or instruction how tu build it?

ChristianMorup, 2020-05-18:
We basically took the example from nullnett, configured it with tsch:

`#include "contiki.h"
#include "net/netstack.h"
#include "net/nullnet/nullnet.h"
#include <string.h>
#include <stdio.h> /* For printf() */

/* Log configuration */
#include "sys/log.h"
#define LOG_MODULE "App"
#define LOG_LEVEL LOG_LEVEL_INFO

/* Configuration */
#define SEND_INTERVAL (8 * CLOCK_SECOND)

#if MAC_CONF_WITH_TSCH
#include "net/mac/tsch/tsch.h"

#include "tsch-cs.h"
static linkaddr_t coordinator_addr = {{0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}};
#endif /* MAC_CONF_WITH_TSCH */

/*---------------------------------------------------------------------------*/
PROCESS(nullnet_example_process, "NullNet broadcast example");
AUTOSTART_PROCESSES(&nullnet_example_process);

/*---------------------------------------------------------------------------*/
void input_callback(const void *data, uint16_t len,
                    const linkaddr_t *src, const linkaddr_t *dest)
{
  if (len == sizeof(unsigned))
  {
    unsigned count;
    memcpy(&count, data, sizeof(count));
    LOG_INFO("Received %u from ", count);
    LOG_INFO_LLADDR(src);
    LOG_INFO_("\n");
  }
}
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(nullnet_example_process, ev, data)
{
  static struct etimer periodic_timer;
  static unsigned count = 0;

  PROCESS_BEGIN();

#if MAC_CONF_WITH_TSCH
  tsch_set_coordinator(linkaddr_cmp(&coordinator_addr, &linkaddr_node_addr));
  NETSTACK_MAC.on();
  tsch_cs_adaptations_init();
#endif /* MAC_CONF_WITH_TSCH */

  
  /* Initialize NullNet */
  nullnet_buf = (uint8_t *)&count;
  nullnet_len = sizeof(count);
  nullnet_set_input_callback(input_callback);

  etimer_set(&periodic_timer, SEND_INTERVAL);
  while (1)
  {
    PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&periodic_timer));

    etimer_reset(&periodic_timer);
  }

  PROCESS_END();
}`

And our configuration is as follows: 

`#ifndef __PROJECT_CONF_H__
#define __PROJECT_CONF_H__

#include <stdint.h>
#include <stdbool.h>

/* Set to enable TSCH security */
#ifndef WITH_SECURITY
#define WITH_SECURITY 0
#endif /* WITH_SECURITY */

/* USB serial takes space, free more space elsewhere */
#define SICSLOWPAN_CONF_FRAG 0
#define UIP_CONF_BUFFER_SIZE 160

/*******************************************************/
/******************* Configure TSCH ********************/
/*******************************************************/

/* IEEE802.15.4 PANID */
#define IEEE802154_CONF_PANID 0x81a5

/* Do not start TSCH at init, wait for NETSTACK_MAC.on() */

/* 6TiSCH minimal schedule length.
 * Larger values result in less frequent active slots: reduces capacity and saves energy. */
#define TSCH_SCHEDULE_CONF_DEFAULT_LENGTH 3

#if WITH_SECURITY

/* Enable security */
#define LLSEC802154_CONF_ENABLED 1

#endif /* WITH_SECURITY */

/* Enable TSCH statistics: must be on for channel selection to work */
#define TSCH_STATS_CONF_ON 1

/* Enable periodic RSSI sampling for TSCH statistics */
#define TSCH_STATS_CONF_SAMPLE_NOISE_RSSI 1

/* Reduce the TSCH stat "decay to normal" period to get printouts more often */
#define TSCH_STATS_CONF_DECAY_INTERVAL (60 * CLOCK_SECOND)

/* For adaptive channel selection */
extern void tsch_cs_channel_stats_updated(uint8_t updated_channel, uint16_t old_busyness_metric);
extern bool tsch_cs_process(void);
/* These will be called from the core TSCH code */
#define TSCH_CALLBACK_CHANNEL_STATS_UPDATED tsch_cs_channel_stats_updated
#define TSCH_CALLBACK_SELECT_CHANNELS tsch_cs_process

/* The coordinator will update the network nodes with new hopping sequences */
#define TSCH_PACKET_CONF_EB_WITH_HOPPING_SEQUENCE 1

/* Reduce the EB period in order to update the network nodes with more agility */
#define TSCH_CONF_EB_PERIOD (4 * CLOCK_SECOND)
#define TSCH_CONF_MAX_EB_PERIOD (4 * CLOCK_SECOND)

#define TSCH_CONF_AUTOSTART 0
/*******************************************************/
/************* Other system configuration **************/
/*******************************************************/

/* Logging */
#define LOG_CONF_LEVEL_RPL LOG_LEVEL_INFO
#define LOG_CONF_LEVEL_TCPIP LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_IPV6 LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_6LOWPAN LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_MAC LOG_LEVEL_DBG
#define LOG_CONF_LEVEL_FRAMER LOG_LEVEL_INFO
#define TSCH_LOG_CONF_PER_SLOT 1

#endif /* __PROJECT_CONF_H__ */`

And the makefile: 

`CONTIKI_PROJECT = nullnet_tsch
all: $(CONTIKI_PROJECT)

CONTIKI = ../..

PLATFORMS_EXCLUDE = nrf52dk

MAKE_MAC = MAKE_MAC_TSCH
MAKE_NET = MAKE_NET_NULLNET
MODULES += $(CONTIKI_NG_SERVICES_DIR)/tsch-cs

include $(CONTIKI)/Makefile.include`

alexrayne, 2020-05-18:
use 3times ` ` ` for mark code

alexrayne, 2020-05-18:
looks ending ` ` ` must be on start of line - new line

alexrayne, 2020-05-18:
What should i look in mote debug log?

ChristianMorup, 2020-05-18:
Exactly, As I interpret that data, it thinks that channel 17-26 are busy, even though there is no interference?
You could also filter for "cs: replacing channel" and you will see that it start swapping channels because they are mistakenly seen as busy channels. 

In addition to this, if you add a mote which works as a constant jammer on channel 11, then it still measures channel 11 as free even though it is obviously wrong.

atiselsts, 2020-05-18:
Hi @ChristianMorup , you don't mention which platform you are using, but from Alex's comment I interpret that it is the Cooja mote.

By defauilt, channels with > -85 dBm background noise RSSI will be interpreted as busy. To change that `#define TSCH_STATS_BUSY_CHANNEL_RSSI` to something else.

On the Cooja mote platform, realistic RSSI is not implemented and a constant value depending on the channel is always returned. If you want to improve the results, use another platform such as the Z1 node.

closed 2020-05-18 by atiselsts

ChristianMorup, 2020-05-18:
Yes, We did it with the cooja motes. Just tried out with Z1 and works perfectly. Thanks a lot.