contiki-ng/contiki-ng #1268: How can a multicast sink node send message to root ?

dpapdpap, 2020-05-18:
I try the the multicast example and it works properly. The root sends a message and all (sink) nodes receive it. But I want the nodes to be able to send messages back to the root. How can I do it ?
I have to add another uip pair (server-client) or I can use the existing ?

g-oikonomou, 2020-05-18:
You can use a separate UDP port pair, but it should also be OK (and perahps preferable) to use the existing UDP "connection".

On the sink, you will need to change the source of the response to be the sink's routable IPv6 address. Destination should be the source of the incoming message. You can do that inside the `tcpip_handler()`.

On the root, you will of course need a similar handler such that it can notify you about reception. One thing to note is, that the UDP "connection" on the root uses the mcast address as destination. I am not sure whether that will prevent reception from other IPs, you will need to check.

dpapdpap, 2020-05-18:
Thanks George
I tried the separate UDP port pair and it works as I want. The only negative is that consumes about 6KB and that does matter in IoT. That's why I would prefer tou use the existing pair but  it's difficult to me to implement the root's tcp_handler (I don't need it to send back a response to node. Just to listen the node). 
I tried and I failed with these :
node's site : 
`NETSTACK_ROUTING.get_root_ipaddr(&borderRoot_ipaddr);
uip_udp_packet_sendto(&mcast_conn,str,strlen(str), &borderRoot_ipaddr,UDP_PORT);` 
where as `UDP_PORT` i tried the multicast's port and a different port with no success
rooter's site : 
 `udp_bind(mcast_conn, UIP_HTONS(UDP_PORT));`
`tcpip_handler(void) {
    if(uip_newdata()) {....}...}`
Any suggestions ?

closed 2020-11-25 by g-oikonomou