contiki-ng/contiki-ng #1270: Connect router cooja using contiki-ng docker on wsl

thymbahutymba, 2020-05-18:
I'm trying to set up rpl-border-router on cooja using contiki-ng on docker WSL. I've followed the [wiki](https://github.com/contiki-ng/contiki-ng/wiki/Docker) but once i try to execute: `make TARGET=cooja connect-router-cooja`
what I get is:
```
../../os/services/rpl-border-router/embedded/Makefile.embedded:13: warning: overriding recipe for target 'connect-router-cooja'
../../arch/platform/cooja/Makefile.cooja:34: warning: ignoring old recipe for target 'connect-router-cooja'
sudo ../../tools/serial-io/tunslip6 -a 127.0.0.1 fd00::1/64
[sudo] password for thymbahutymba:
client: connect: Connection refused
tunslip6: can't connect to ``127.0.0.1:60001'': Connection refused
make: *** [../../os/services/rpl-border-router/embedded/Makefile.embedded:13: connect-router-cooja] Error 1
```
I've tried also to add to the `docker run ...` the option `--network host` with the same results. How I can get this working.

g-oikonomou, 2020-05-18:
So you have the serial socket plugin inside cooja running? And the sim also running?

We'll have to check, I _think_ this is expected to work.

yatch, 2020-05-18:
Can you try `--sysctl net.ipv6.conf.all.disable_ipv6=0` option of `docker run`?

yatch, 2020-05-18:
Ah, it may be irrelevant.

thymbahutymba, 2020-05-18:
> Can you try `--sysctl net.ipv6.conf.all.disable_ipv6=0` option with `docker run`?

Is already included in the docker run command of you wiki (docker under windows). What I've tried is with and without `--sysctl ...` and with and without `--network host`

yatch, 2020-05-18:
Are you sure Cooja is running when you hit `make TARGET=cooja connect-router-cooja`? Realized  I'm asking the same as @g-oikonomou...:

> So you have the serial socket plugin inside cooja running? And the sim also running?

thymbahutymba, 2020-05-18:
Just to give you a full overview of what I've done:

`docker run --name contiki-ng-dev --rm --privileged --sysctl net.ipv6.conf.all.disable_ipv6=0 --mount type=bind,source=/c/Users/thymbahutymba/Documents/contiki-ng,destination=/home/user/contiki-ng -e DISPLAY="$DISPLAY" -ti contiker/contiki-ng:develop`
then execute cooja and create the cooja mote of the rpl-border-router example following this [link](https://github.com/contiki-ng/contiki-ng/wiki/Cooja:-simulating-a-border-router)

yatch, 2020-05-18:
@thymbahutymba OK, thanks. Put my name on this issue. I don't a Windows machine having WSL enabled, though.

thymbahutymba, 2020-05-18:
Let me know if I can help somehow.

yatch, 2020-12-01:
@thymbahutymba Sorry for this late response. You might have solved this issue by yourself, but...

I think, you forgot to click "start" button on "Serial Socket" window before running `make TARGET=cooja connect-router-cooja`.

Here is what I got. I'm not using Windows, though.

Without clicking "start" button,
```
$ make TARGET=cooja connect-router-cooja
../../os/services/rpl-border-router/embedded/Makefile.embedded:13: warning: overriding recipe for target 'connect-router-cooja'
../../arch/platform/cooja/Makefile.cooja:34: warning: ignoring old recipe for target 'connect-router-cooja'
sudo ../../tools/serial-io/tunslip6 -a 127.0.0.1 fd00::1/64
client: connect: Connection refused
tunslip6: can't connect to ``127.0.0.1:60001'': Connection refused
../../os/services/rpl-border-router/embedded/Makefile.embedded:13: recipe for target 'connect-router-cooja' failed
make: *** [connect-router-cooja] Error 1
```
After clicking "start" button,
```
$ make TARGET=cooja connect-router-cooja
../../os/services/rpl-border-router/embedded/Makefile.embedded:13: warning: overriding recipe for target 'connect-router-cooja'
../../arch/platform/cooja/Makefile.cooja:34: warning: ignoring old recipe for target 'connect-router-cooja'
sudo ../../tools/serial-io/tunslip6 -a 127.0.0.1 fd00::1/64
slip connected to ``127.0.0.1:60001''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` mtu 1500 up
ifconfig tun0 add fd00::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 172.17.0.2  netmask 255.255.255.255  destination 172.17.0.2
        inet6 fe80::1  prefixlen 64  scopeid 0x20<link>
        inet6 fe80::2b4:908a:b9ec:6c40  prefixlen 64  scopeid 0x20<link>
        inet6 fd00::1  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
Hope this helps.

I'm going to close this issue. If the issue persists on your side, feel free to reopen.

closed 2020-12-01 by yatch