contiki-ng/contiki-ng #1274: [COOJA] overwrite CONFIG_DIR in csc files

MaudCh, 2020-05-22:
Hi everyone!

I would like to use symlinks on csc files in order to use my own makefile and avoid to duplicate tests in my project. 
My problem is that COOJA consider that the test path is the source of the symlink path and not the symlink path.

I know that is link to this CONFIG_DIR variable in csc files but I don't know how COOJA define this variable and so how we can overwrite it. 

Is it possible to do it? if yes, how?

_code extract from tests/14-rpl-lite/01-rpl-up-route.csc_
`<motetype>
      org.contikios.cooja.contikimote.ContikiMoteType
      <identifier>mtype488</identifier>
      <description>Sender</description>
      <source>[CONFIG_DIR]/code/sender-node.c</source>
      <commands>make TARGET=cooja clean
make -j sender-node.cooja TARGET=cooja</commands>`

alexrayne, 2020-05-23:
Can you describe more , what you wanna do? maybe step by step, some example? 
can't understand what is the problem?

MaudCh, 2020-05-25:
Hi @alexrayne,
I would like to use some contiki tests which use cooja in my own environment (based on contiki-ng so I have included all contiki-ng in my repo). For that, I took _14-rpl-lite_ test and copy paste in my environment and only change the makefile to include my environment. 

This is working but it mean that I have to duplicate a lot of files that exists somewhere else in my project. I tried to create symlinks for the files I did not need to change and keep the same structure in my folder like that : 
`.
├── 01-rpl-up-route.csc -> symlink to contiki-ng/tests/14-rpl-lite/01-rpl-up-route.csc
├── 02-rpl-root-reboot-2.csc -> symlink to contiki-ng/tests/14-rpl-lite/02-rpl-root-reboot-2.csc
├── 03-rpl-28-hours.csc -> symlink to contiki-ng/tests/14-rpl-lite/03-rpl-28-hours.csc
├── 05-rpl-up-and-down-routes.csc -> symlink to contiki-ng/tests/14-rpl-lite/05-rpl-up-and-down-routes.csc
├── 06-rpl-temporary-root-loss.csc -> symlink to contiki-ng/tests/14-rpl-lite/06-rpl-temporary-root-loss.csc
├── 07-rpl-random-rearrangement.csc -> symlink to contiki-ng/tests/14-rpl-lite/07-rpl-random-rearrangement.csc
├── 08-rpl-dao-route-loss-0.csc -> symlink to contiki-ng/tests/14-rpl-lite/08-rpl-dao-route-loss-0.csc
├── 08-rpl-dao-route-loss-1.csc -> symlink to contiki-ng/tests/14-rpl-lite/08-rpl-dao-route-loss-1.csc
├── 08-rpl-dao-route-loss-2.csc -> symlink to contiki-ng/tests/14-rpl-lite/08-rpl-dao-route-loss-2.csc
├── 08-rpl-dao-route-loss-3.csc -> symlink to contiki-ng/tests/14-rpl-lite/08-rpl-dao-route-loss-3.csc
├── 08-rpl-dao-route-loss-4.csc -> symlink to contiki-ng/tests/14-rpl-lite/08-rpl-dao-route-loss-4.csc
├── 08-rpl-dao-route-loss-5.csc -> symlink to contiki-ng/tests/14-rpl-lite/08-rpl-dao-route-loss-5.csc
├── 09-rpl-probing.csc -> symlink to contiki-ng/tests/14-rpl-lite/09-rpl-probing.csc
├── 10-rpl-resetting-dio-timer-by-dis.csc -> symlink to contiki-ng/tests/14-rpl-lite/10-rpl-resetting-dio-timer-by-dis.csc
├── code
│   ├── dis-sender.c -> symlink to contiki-ng/tests/14-rpl-lite/code/dis-sender.c
│   ├── Makefile
│   ├── project-conf.h -> symlink to contiki-ng/tests/14-rpl-lite/code/project-conf.h
│   ├── receiver-node.c -> symlink to contiki-ng/tests/14-rpl-lite/code/receiver-node.c
│   ├── root-node.c -> symlink to contiki-ng/tests/14-rpl-lite/code/root-node.c
│   └── sender-node.c -> symlink to contiki-ng/tests/14-rpl-lite/code/sender-node.c
├── js -> symlink to contiki-ng/tests/14-rpl-lite/js
└── Makefile`

My problem is that CSC files has a variable named [CONFIG_DIR] that is define by COOJA somewhere. by default, [CONFIG_DIR] = CSC path and does not care about symlinks so in my example if I run ./01-rpl-up-route.csc, [CONFIG_DIR] = contiki-ng/tests/14-rpl-lite/01-rpl-up-route.csc instead of ./01-rpl-up-route.csc.

I am looking for a way to overwrite this CONFIG_DIR to match with my symlinks.

alexrayne, 2020-05-25:
Your explanation mades more questions. Work via symlinks, imho - disaster. maybe your goal can be handled another way?

- what you mean "your environment" ? it is modified contiki-ng repo?

- why via symlinks, why not work exactly in original contiki-ng?

[CONFIG_DIR] and  [CONTIKI_DIR] takes from settings/external tools path ...  from PATH_COJA and PATH_CONTIKI

MaudCh, 2020-06-02:
I cannot modify contiki-ng folder because it is a sub-module of my project and "my environment" = all the features I implemented based on contiki-ng.

CONTIKI_DIR and COOJA_DIR are based on PATH_CONTIKI and PATH_COOJA. 
I want to know how Cooja get the value of CONFIG_DIR in its csc files.

alexrayne, 2020-06-02:
> CONTIKI_DIR and COOJA_DIR are based on PATH_CONTIKI and PATH_COOJA.
> I want to know how Cooja get the value of CONFIG_DIR in its csc files.

CONFIG_DIR - substitues to canonical PATH_COOJA, as i remember. and therefore [CONFIG_DIR] MUST starts node.c file name

look to cooja.java:restoreContikiRelativePath for sure

MaudCh, 2020-06-04:
Nice thank you very much for your help !

closed 2020-06-04 by MaudCh

alexrayne, 2020-06-04:
try inspect on my patches at [fix-compiledialog-command-update](https://github.com/alexrayne/cooja/tree/fix-compiledialog-command-update)
There is a bit fixes on [CONTIKI/CONFIG_DIR] evaluation and expantion. 
at least , a can't build my csc files on original version, only my fixes helps. Maybe it help you a bit?