contiki-ng/contiki-ng #1284: Cooja crashes with error 134 as soon as button "start" pressed

dpapdpap, 2020-05-30:
In a clear installation of contiki-ng (Win10, WSL2, UBUNTU 20.4, openjdk-8-amd64) I try the Hello-world example.
I create a cooja-node. It compiles without errors or warnings but, as soon as the start is pressed, the IDE closes by giving the following error messages :
**[java] # A fatal error has been detected by the Java Runtime Environment:
[java] #
[java] # SIGSEGV (0xb) at pc=0x00007f6719ecbf4d, pid=4215, tid=0x00007f66eae3a700
[java] #
[java] # JRE version: OpenJDK Runtime Environment (8.0_252-b09) (build 1.8.0_252-8u252-b09-1ubuntu1-b09)
[java] # Java VM: OpenJDK 64-Bit Server VM (25.252-b09 mixed mode linux-amd64 compressed oops)
[java] # Problematic frame:
[java] # C [libc.so.6+0x96f4d]
[java] #
[java] # Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
[java] #
[java] # An error report file with more information is saved as:
[java] # /home/dpap/contiki-ng/tools/cooja/build/hs_err_pid4215.log
[java] #
[java] # If you would like to submit a bug report, please visit:
[java] # http://bugreport.java.com/bugreport/crash.jsp
[java] # The crash happened outside the Java Virtual Machine in native code.
[java] # See problematic frame for where to report the bug.
[java] #
[java] Java Result: 134**

Can someone help me please?
It's a shame to get there and not be able to continue!

g-oikonomou, 2020-05-30:
So your linux subsystem is 64-bit, right?

dpapdpap, 2020-05-30:
yes

g-oikonomou, 2020-05-30:
> Note that Cooja motes only work on 32-bit OSes.

https://github.com/contiki-ng/contiki-ng/wiki/Platform-cooja

dpapdpap, 2020-05-30:
I have read this but I also read this 
"When working with a 64-bit virtual machine, you may run into problems with the `serialdump-linux` executable because it may have been compiled for 32-bit machines. Install the following package to fix this issue.
 sudo apt-get install lib32ncurses5" (https://anrg.usc.edu/contiki/index.php/Installation)
 and assumed that the problem would be solved. But obviously not for ng, right?

g-oikonomou, 2020-05-30:
`serialdump-linux` is no longer part of the source distribution, we removed it a long while ago and replaced it with easy ways of compiling it for the user's target operating system. So this comment you quote above is no longer applicable.

The problem with native cooja motes is entirely unrelated. This has been a sticking point for quite some time and one that we really need to fix because it is holding a number of other things back.

For now I would suggest either running cooja motes inside docker, or using some other mote type in Cooja.

dpapdpap, 2020-05-30:
Thank you very much George. You are very kind. 
I will try the doker and I hope I can do it (although I really liked the ease of WSL2)

closed 2020-05-30 by dpapdpap

g-oikonomou, 2020-05-30:
If your project code does not have too high code/RAM requirements and you prefer WSL, Isuggest trying Z1 motes inside Cooja. They could well prove adequate for your needs unless you end up getting linker errors.

dpapdpap, 2020-05-30:
Ok, I'll try with Z1. Thanks again

dpapdpap, 2020-05-30:
Z1 motes can run "Hello-world" but CoAP and rpl router examples, for what I'm inderested, don't work on Z1 (pitty)

g-oikonomou, 2020-05-30:
RPL border router should be OK if you force-build it. CoAP server would need a bit of work to get rid of stuff before it'll fit.