contiki-ng/contiki-ng #1294: cc2650, start boot loader without press the button?

fengjikui, 2020-06-02:
hello, I am using the-rom-bootloader to program the cc2650, and very time I need to press the user button and reset button, is there any way to start boot loader without press the button?
[ROM boot loader](https://github.com/contiki-ng/contiki-ng/wiki/Platform-srf06-cc26xx#using-the-rom-bootloader)

fengjikui, 2020-06-03:
@George Oikonomou

g-oikonomou, 2020-06-08:
As per the wiki page, if the flash is not empty then the only way to enter the bootloader is by setting a specific pin (configurable) to a specific logical state (hi or low, also configurable).

If you are interested in more details, the bootloader is documented in full in the chip's Technical Reference Manual (Section 8.1)

https://www.ti.com/lit/pdf/swcu117

closed 2020-06-08 by g-oikonomou