contiki-ng/contiki-ng #1303: is energy metric implement in RPL protocol? 2

matinpf, 2020-06-08:
I‌ already submit issue but did not get my answer (#1281)

my point is update_metric_container function in RPL implementation depend on pas_cost and pas_cost call "parent_path_cost" function. also "parent_path_cost"  call parent_link_metric that base on ETX.

As far as I know, in this chain, the energy has not any role, but I expected that to use "sys/energest.h" library in this process.

alexrayne, 2020-06-13:
Have you realy energy difference for communication between nodes? as i understand, all communication are equal, and energy depends only on packet size.
Can you describe scenario, where consumption may be ralated to route?

matinpf, 2020-06-14:
Yes, I think energy is depending on both size and number of packets a mote routing them. If you imagine the scenario like below simulation, the mote 12 has to route many packets; therefore, its battery decreases rapidly. In my view, if we have a proper implementation of energy metric after slightly decreases in a battery, the routing path of some other mote most changes to 9 and 3 mote.

[image]

By the way, I also think the energy consumption may depend on some programs that run in the application layer. 

My suggestion is to use the formula like the below formula to calculate the energy consumption in the Cooja simulator. 
[image]

source :‌SVELTE: Real-time intrusion detection in the Internet of Things

Whether the real mote, like sky mote, have access to the remaining amount of battery?

alexrayne, 2020-06-14:
> In my view, if we have a proper implementation of energy metric after slightly decreases in a battery, the routing path of some other mote most changes to 9 and 3 mote.

imho, for "proper implementation" there is no need to use metric of energy. Amount of connections,  amount of traffic, amount of time-slots for listen/send packets - can do the same route pathes allocation.
as for energy, maybe battery level can be meaningful as route metris - to avoid pathet via low-battery nodes.

matinpf, 2020-06-14:
> imho, for "proper implementation" there is no need to use metric of energy. Amount of connections, amount of traffic, amount of time-slots for listen/send packets - can do the same route pathes allocation.
> as for energy, maybe battery level can be meaningful as route metric - to avoid pathes via low-battery nodes.
I agree with you

I think if anyone will want to implement this metric, the [RFC 6551](https://tools.ietf.org/html/rfc6551#section-3.2) is a useful resource but until then, maybe better deactivate this option (in the current implementation) in order to prevent confusing people like me :) .

atiselsts, 2020-06-15:
Hello,

> update_metric_container function in RPL implementation depend on pas_cost and pas_cost call "parent_path_cost" function. also "parent_path_cost" call parent_link_metric that is base on ETX.

I agree, I think it's non-conformance to the standard. I would like some of the RPL experts to join this discussion, but they seem to be quiet, so here are my views. Note that to be conformant to the RFC may mean simply this:

>   The simplest solution relies on a 2-bit field encoding three types of
>   power sources: "powered", "battery", and "scavenger".  This simple
>   approach may be sufficient for many applications.

This info is already provided in the `flags` field of the MC object (admittedly, based on guesswork).
My impression is that it was never the intention of the developers to provide realistic remaining energy values, merely a RFC-conformant packet structure and parsing, so that anyone interested in this can come up with their own energy-based routing metric without having to hack the core RPL code.

**To sum up:** there is support for the MC energy object; the metric value appears to be filled incorrectly (a probable bug); the implementation does not aim to provide full solution, just (part of) an API for a solution.

> As far as I know, in this chain, the energy has not any role, but I expected that to use "sys/energest.h" library in this process.

Your broader question about node's energy budget is a different one. The unfortunately named Energest library does not track the node's energy consumption, merely the time spent in each state - as I'm sure you know. Even so, you can use Energest to track the charge consumption in mAh quite accurately, assuming you have a good hardware charge-consumption model. It can even be used to track the energy consumption in mW assuming the voltage is stable or known (though the energy consumption is quite useless for most cases as batteries provide their capacity in mAh).

Contiki-NG certainly does not have a remaining battery lifetime estimation library, nor do I know any Contiki-NG class embedded OS that does this (but I'm interested in any counterexamples). You can certainly come up with some models, but they will be (1) specific to a combination of battery, device, and potentially even application workload and/or (2) very inaccurate for real-life applications. Battery estimation is a complicated business. Even my laptop does not have a good idea about the remaining battery, despite having many highly-qualified engineers working on this for many years! I'm not sure the RPL IETF working group fully appreciated these difficulties when they come up with those standards.

closed 2020-11-25 by g-oikonomou