contiki-ng/contiki-ng #1315: RPL-UDP example downward issue

CharalamposDimaridis, 2020-06-17:
Hello,
I'm doing a research for my Thesis about RPL-Lite and RPL-Classic implementation and i have an issue with downwards messages. I want to measure control overhead and packet delivery ratio between the two implementations. I'm using 9 examples of 10, 25 and 50 nodes with RPL-Classic, storing and non-storing mode, and RPL-Lite using sky mote. The problem is, while sink takes all the messages (upward messages) from all clients and sends respond, only 16 nodes are taking this respond from the sink both on 25 and 50 nodes example, for all RPL implementations. The examples ran for 2 hours each, the nodes that wasn't take respond was sending messages for this 2 hours, the sink was responding on this nodes but the response never arrived on this clients.I tried to use different mote, z1 mote, but it was worse, only 10/25 and 10/50 nodes took response (downward message) from sink while sink took message from all client nodes. Is this normal? 
   
Thank you for your time.

atiselsts, 2020-06-18:
By default, only 16 routes are available. This depends on the configuration constant `NETSTACK_MAX_ROUTE_ENTRIES`.

If you are using sky or z1 (very low RAM size) then it's unlikely you can get much more routes than that. Use a more modern platform or simulated (Cooja) motes

CharalamposDimaridis, 2020-06-18:
Thank you for your help!

closed 2020-06-25 by atiselsts