contiki-ng/contiki-ng #1319: MQTT 5 Compilation error

veracl, 2020-06-28:
I saw that MQTT 5 support has recently been added to contiki-ng (https://github.com/contiki-ng/contiki-ng/pull/1250). I am working on a project using TI CC2538 boards that would benefit from this, so I wanted to try it out. However, when I tried compiling the mqtt-client example with MQTT 5 enabled, I got these compilation errors (using arm-none-eabi-gcc version 9.2.1):

```
vagrant@ubuntu-bionic:~/iot/contiki-ng/examples/mqtt-client$ make
TARGET not defined, using target 'native'
  CC        ../../os/net/app-layer/mqtt/mqtt.c
../../os/net/app-layer/mqtt/mqtt.c: In function ‘write_out_props’:
../../os/net/app-layer/mqtt/mqtt.c:486:29: error: ‘prop’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
                             prop->val,
                             ^~~~
../../os/net/app-layer/mqtt/mqtt.c: In function ‘process_thread_mqtt_process’:
../../os/net/app-layer/mqtt/mqtt.c:582:5: error: ‘will_props’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
     write_out_props(pt, conn, will_props);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../os/net/app-layer/mqtt/mqtt.c:507:26: note: ‘will_props’ was declared here
   struct mqtt_prop_list *will_props = MQTT_PROP_LIST_NONE;
                          ^~~~~~~~~~
cc1: all warnings being treated as errors
../../Makefile.include:356: recipe for target 'build/native/obj/mqtt.o' failed
make: *** [build/native/obj/mqtt.o] Error 1
```

I have tried both `TARGET=cc2538dk` and target native (see above). To enable MQTT 5, I added the following line to `project-conf.h`:

```
#define MQTT_CONF_VERSION MQTT_PROTOCOL_VERSION_5
```

I realise that the MQTT 5 support is not part of an official release yet, but I had thought it should at least compile. Is this error perhaps caused by some misconfiguration on my part? How can I get it to compile?
Thanks for any assistance!

g-oikonomou, 2020-06-28:
Bug

Try with #1320 please

veracl, 2020-06-29:
It now compiles for target native without error, but when I try compiling for cc2538dk, I get this:

```
vagrant@ubuntu-bionic:~/iot/contiki-ng/examples/mqtt-client$ make TARGET=cc2538dk
  CC        ../../arch/cpu/cc2538/dev/cc2538-rf.c
  CC        ../../arch/cpu/cc2538/./ieee-addr.c
  CC        ../../os/net/app-layer/mqtt/mqtt.c
  CC        ../../os/net/app-layer/mqtt/mqtt-prop.c
../../os/net/app-layer/mqtt/mqtt-prop.c: In function 'mqtt_get_next_in_prop':
../../os/net/app-layer/mqtt/mqtt-prop.c:508:30: error: converting a packed 'mqtt_vhdr_prop_t' {aka 'enum <anonymous>'} pointer (alignment 1) to a 'uint16_t' {aka 'short unsigned int'} pointer (alignment 2) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  508 |                              NULL, NULL, (uint16_t *)prop_id);
      |                              ^~~~
In file included from ../../os/net/app-layer/mqtt/mqtt-prop.c:35:
../../os/net/app-layer/mqtt/mqtt.h:307:14: note: defined here
  307 | typedef enum {
      |              ^
cc1: all warnings being treated as errors
../../Makefile.include:356: recipe for target 'build/cc2538dk/obj/mqtt-prop.o' failed
make: *** [build/cc2538dk/obj/mqtt-prop.o] Error 1
```

g-oikonomou, 2020-06-29:
I cannot reproduce this.

Could it be the toolchain version?

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I'll have a look at the code anyway

veracl, 2020-06-29:
Yes, that seems to be the case. It is compiling for me using v7.3.1 instead of v9.2.1. Testing it out now. Thanks for the quick fix!

PR #1320 (2020-07-14): Fix uninitialised variable errors under MQTT 5
Fixes #1319
closed 2020-07-14 by g-oikonomou