contiki-ng/contiki-ng #1321: How to compile contiki-ng for for CC2538DK to get elf and bin files?

ahmed-balata, 2020-06-28:
i want to build elf file and bin file for for CC2538DK.

When i run make i get
make: *** No targets specified and no makefile found.  Stop.

and when i run gcc contiki-main.c i get
In file included from contiki-main.c:43:0:
contiki.h:37:10: fatal error: contiki-conf.h: No such file or directory
 #include "contiki-conf.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.

Thanks

g-oikonomou, 2020-06-28:
From inside the example's directory (e.g. `examples/hello-world`) you will need to run `make TARGET=cc2538dk`. This will create a firmware file `hello-world.cc2538dk` for you to programme your device with. This is an ELF file.

You will also find the `.elf`, `.bin` and `.hex` under `build/cc2538dk`

closed 2020-06-28 by g-oikonomou

ahmed-balata, 2020-06-28:
@ g-oikonomou

when i run     make TARGET=cc2538dk. inside   examples/platform-specific/cc2538-common   i get

MKDIR     build/cc2538dk/obj
  CC        ../../arch/platform/cc2538dk/dev/leds-arch.c
make: arm-none-eabi-gcc: Command not found
../../Makefile.include:356: recipe for target 'build/cc2538dk/obj/leds-arch.o' failed
make: *** [build/cc2538dk/obj/leds-arch.o] Error 127

g-oikonomou, 2020-06-28:
Please spend some time reading and understanding guides in our wiki:
https://github.com/contiki-ng/contiki-ng/wiki

Depending on your environment (Linux, macOS, virtual machine, docker container) you will likekly need to spend some time on those:
https://github.com/contiki-ng/contiki-ng/wiki/Docker
https://github.com/contiki-ng/contiki-ng/wiki/Vagrant
https://github.com/contiki-ng/contiki-ng/wiki/Toolchain-installation-on-Linux
https://github.com/contiki-ng/contiki-ng/wiki/Toolchain-installation-on-macOS

Suggest you also familiarise yourself with the build system:
https://github.com/contiki-ng/contiki-ng/wiki/The-Contiki%E2%80%90NG-build-system

Since you are interested in the CC2538DK:
https://github.com/contiki-ng/contiki-ng/wiki/Platform-cc2538dk

Any feedback on the guides always welcome