contiki-ng/contiki-ng #1325: Where to report security issues for Contiki-OS?

standash, 2020-06-28:
Hi,

I tried to find a proper contact for reporting security issues in the historical version of Contiki-OS, but I could not find anything on the website. I also opened an issue at the Contiki-OS repository, but there is no response.

Could anyone of the maintainers suggest whom should I contact please?

Thank you!

mjurczak, 2020-07-11:
@standash, there is a security policy indicating an e-mail address to report vulnerabilities privately: https://github.com/contiki-ng/contiki-ng/blob/develop/SECURITY.md.

I have reported an issue that way, but unfortunately there was no response or confirmation of reception.
@g-oikonomou - what would is the right way to report security issues?

g-oikonomou, 2020-11-25:
The right way to report security vulnerabilities privately is indeed as shown on that page. Do note that this address is for issues specific to Contiki-NG, and not for the legacy Contiki OS.

closed 2020-11-25 by g-oikonomou