contiki-ng/contiki-ng #1328: CC1310 startup

amxbc, 2020-07-03:
Hi,
I was testing contiki-ng with the CC1310 and started with two launchpads (border router + hello world) which worked just fine. I then switched to a custom board that does not have the 32kHz oscillator and the hello-world code wont start.
In arch/platform/cc26x0-cc13x0/platform.c from line 134 it reads
`void
platform_init_stage_one()
{
  /* Enable flash cache and prefetch. */
  ti_lib_vims_mode_set(VIMS_BASE, VIMS_MODE_ENABLED);
  ti_lib_vims_configure(VIMS_BASE, true, true);

  ti_lib_int_master_disable();

  /* Set the LF XOSC as the LF system clock source */
  oscillators_select_lf_xosc();`
In my opinion this should be optional or is there a reason why contiki requires the 32kHz external oscillator?
Btw, I checked this in the debugger and the "while(ti_lib_osc_clock_source_get(OSC_SRC_CLK_LF) != OSC_XOSC_LF);" repeats forever.

Thanks and regards
amxbc

g-oikonomou, 2020-07-03:
The internal RC Osc is bugged on the CC26x0/CC13x0

https://www.ti.com/lit/er/swrz088/swrz088.pdf
https://www.ti.com/lit/pdf/swrz062

amxbc, 2020-07-03:
I see, thanks for the comment g-oikonomou. I am not to familiar with the platform, is the LF oscillator mandatory? For most other MCUs I know you don't need the LF/32kHz oscillator at all (just for low power, rtc etc.).
Anyway, I soldered a crystal with two caps onto the board and now it gets past that point. It still hangs at some point but now in the board realm. That's something I can live with (defining a new board). Having to reinvent the CC1310 platform when the 32kHz is not mandatory would be different.

g-oikonomou, 2020-07-03:
You always need a low frequency oscillator to feed clocks when the device is deep sleeping. If you are planning to keep the device running/sleeping at all times then you could probably always feed all clocks from the high frequency osc.

amxbc, 2020-07-04:
Thanks again. I will familiarize myself a little more with contiki-ng for cc1310. Maybe I will come up with a suggestion. :-)

closed 2020-07-04 by amxbc