contiki-ng/contiki-ng #1347: Contiki-ng rpl udp example using CCS + Contiki-NG + LAUNCHXL-CC1310 + LAUNCHXL-CC1350 not communicating over RF Sub-1 GHz

PrashantTeke, 2020-08-04:
I am working on contiki-ng rpl udp example using CCS  + Contiki-NG + LAUNCHXL-CC1310 + LAUNCHXL-CC1350. Board gets booted successfully, but boards are not communicating on RF. I think we are missing some basic settings. Could please check the below logs of UDP-Server and UDP-Client and let us know why these are not communicating over Sub GHz

Contiki-NG UDP Server
[INFO: Main      ] Starting Contiki-NG-release/v4.5-176-g06773d345-dirty
[INFO: Main      ] UDP Server 
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 50859
[INFO: Main      ] Link-layer address: 0012.4b00.12cd.c6ab
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:12cd:c6ab
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 26
[INFO: CC13xx/CC26xx] Node ID: 50859

Contiki UDP Client
[INFO: Main      ] Starting Contiki-NG-release/v4.5-176-g06773d345-dirty
[INFO: Main      ] UDP Client 
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 388
[INFO: Main      ] Link-layer address: 0012.4b00.1532.0184
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:1532:184
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 26
[INFO: CC13xx/CC26xx] Node ID: 388

Thanks for your help!

Thanks,
Prashant

PrashantTeke, 2020-08-04:
Please find the below Makefile of Contiki-ng rpl udp udp-client udp-server example, let me know if anythings is missing in this,
CONTIKI_PROJECT = udp-client udp-server

TARGET = simplelink
BOARD = launchpad/cc1350
CFLAGS += -g

all: $(CONTIKI_PROJECT)

CONTIKI=../..
include $(CONTIKI)/Makefile.include

Thanks for your help!

Thanks,
Prashant

atiselsts, 2020-08-05:
We fixed a bug in the CSMA on CC13xx on simplelink platform. Can you update to the latest `develop` branch commit and try again?

PrashantTeke, 2020-08-05:
Hi,

Thank you very much for your reply.

I have tested it but for first few messages we are not receiving any reply. 

UDP Client
[INFO: Main      ] Starting Contiki-NG-release/v4.5-180-gbda187630-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 388
[INFO: Main      ] Link-layer address: 0012.4b00.1532.0184
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:1532:184
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 26
[INFO: CC13xx/CC26xx] Node ID: 388
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Sending request 0 to fd00::212:4b00:1532:a801
[INFO: App       ] Received response 'hello 0' from fd00::212:4b00:1532:a801
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Sending request 0 to fd00::212:4b00:1532:a801
[INFO: App       ] Received response 'hello 0' from fd00::212:4b00:1532:a801
[INFO: App       ] Sending request 1 to fd00::212:4b00:1532:a801
[INFO: App       ] Received response 'hello 1' from fd00::212:4b00:1532:a801
[INFO: App       ] Sending request 2 to fd00::212:4b00:1532:a801
[INFO: App       ] Received response 'hello 2' from fd00::212:4b00:1532:a801
[INFO: App       ] Sending request 3 to fd00::212:4b00:1532:a801
[INFO: App       ] Received response 'hello 3' from fd00::212:4b00:1532:a801
[INFO: App       ] Sending request 4 to fd00::212:4b00:1532:a801
[INFO: App       ] Received response 'hello 4' from fd00::212:4b00:1532:a801
[INFO: App       ] Sending request 5 to fd00::212:4b00:1532:a801
[INFO: App       ] Received response 'hello 5' from fd00::212:4b00:1532:a801
[INFO: App       ] Sending request 6 to fd00::212:4b00:1532:a801
[INFO: App       ] Received response 'hello 6' from fd00::212:4b00:1532:a801

UDP Server
[INFO: Main      ] Starting Contiki-NG-release/v4.5-180-gbda187630-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 43009
[INFO: Main      ] Link-layer address: 0012.4b00.1532.a801
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:1532:a801
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 26
[INFO: CC13xx/CC26xx] Node ID: 43009
[INFO: App       ] Received request 'hello 0' from fd00::212:4b00:1532:184
[INFO: App       ] Sending response.
[INFO: App       ] Received request 'hello 0' from fd00::212:4b00:1532:184
[INFO: App       ] Sending response.
[INFO: App       ] Received request 'hello 0' from fd00::212:4b00:1532:184
[INFO: App       ] Sending response.
[INFO: App       ] Received request 'hello 1' from fd00::212:4b00:1532:184
[INFO: App       ] Sending response.
[INFO: App       ] Received request 'hello 2' from fd00::212:4b00:1532:184
[INFO: App       ] Sending response.
[INFO: App       ] Received request 'hello 3' from fd00::212:4b00:1532:184
[INFO: App       ] Sending response.

Thanks,
Prashant

atiselsts, 2020-08-05:
It takes some time for the other node to join the RPL network. If you want an immediate response, try pinging the link-local address.

I'm closing this now that things are working as they should.

closed 2020-08-05 by atiselsts

andynnd, 2020-08-19:
Hi all,
I tested the rpl-udp example using a LAUNCHXL-CC1352P-2 and a LAUNCHXL-CC1310 at the newest develop branch (413fb45) but these boards cannot connect each other. 

UDP-server:
[INFO: Main      ] Starting Contiki-NG-413fb45-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 44916
[INFO: Main      ] Link-layer address: 0012.4b00.1ca1.af74
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:1ca1:af74
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 26
[INFO: CC13xx/CC26xx] Node ID: 44916

UDP-client:
[INFO: Main      ] Starting Contiki-NG-413fb45-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 25902
[INFO: Main      ] Link-layer address: 0012.4b00.0e09.652e
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:e09:652e
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 26
[INFO: CC13xx/CC26xx] Node ID: 25902
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
[INFO: App       ] Not reachable yet
...

I turn on debug flags on rf/prop-mode.c but it showed nothing. Seem like they cannot work on Sub-1GHz.

atiselsts, 2020-08-19:
@andynnd A number of people have reported an issue with CC1352 - you are the third or fourth that I see. Unfortunately Contiki-NG maintainers do not have any of those board at the moment. You should try to contact Texas Instrument directly and inquire if they can contribute to the support of the Simplelink platform in Contiki-NG.

This is however off-topic for this issue which is already fixed and closed. Can you create a new one about your problem?

andynnd, 2020-08-19:
@atiselsts Thanks for your reply. Yes, I can run some examples with 2 CC1310 launchpad boards but for some cannot work with cc1352p at Sub-1Ghz. I will try to test some and create a new issue.

andresblanco06, 2020-08-27:
@andynnd @atiselsts 

I am having the same issue as everyone else.

The main fix for the CC1352P boards is to apply cap tuning, since the launchpad is designed differently and is using external clock sources, there is a slight frequency drift when compared to cc1312.
I do this by adding the following to the makefile:

```
CFLAGS += -DSET_CCFG_MODE_CONF_XOSC_CAP_MOD=0x0
CFLAGS += -DSET_CCFG_MODE_CONF_XOSC_CAPARRAY_DELTA=0xD5 
``` 

Now, After I apply the fix I am able to associate but I cannot communicate between them... if that makes any sense.

I try pinging the rpl-border router from the cc1312 node and it always timesout, I already updated to latest develop branch and I am still getting this. @atiselsts any help?