contiki-ng/contiki-ng #1352: Buffer overflow in .bss section due to SNMP request overflow

mjurczak, 2020-08-17:
### Description of defect

**References:**

https://github.com/contiki-ng/contiki-ng/tree/release/v4.5
https://github.com/contiki-ng/contiki-ng/tree/release/v4.4

**File:**

[snmp-engine.c](https://github.com/contiki-ng/contiki-ng/blob/release/v4.5/os/net/app-layer/snmp/snmp-engine.c)
[snmp-message.c](https://github.com/contiki-ng/contiki-ng/blob/release/v4.5/os/net/app-layer/snmp/snmp-message.c)

**Analysis:**
Memory access out of buffer boundaries may occur if an SNMP  request with number of OIDs larger than supported by the engine is received and processed.

The OIDs listed in a request are processed by snmp_message_decode() function without verification of the varbinds buffer capacity.
The buffer is allocated in .bss as a static variable:
https://github.com/contiki-ng/contiki-ng/blob/23db95746afcaa40df3e742a88fc4877d1ec4dd2/os/net/app-layer/snmp/snmp-engine.c#L208

The varbinds memory buffer is written with the values provided in SNMP request:
https://github.com/contiki-ng/contiki-ng/blob/23db95746afcaa40df3e742a88fc4877d1ec4dd2/os/net/app-layer/snmp/snmp-message.c#L245

The buffer capacity is determined at compile time by the following definition:
https://github.com/contiki-ng/contiki-ng/blob/23db95746afcaa40df3e742a88fc4877d1ec4dd2/os/net/app-layer/snmp/snmp-conf.h#L81-L87

If the number of variables in the request exceeds the allocated buffer a memory write out of the buffer boundaries occurs. The write operation beyond the buffer capacity provides possibility to overwrite other variables allocated in the .bss section by the application. 
As the sender of the frame is in controll of the content that will be written beyond the buffer limits and there is no strict process memory separation in contiki-ng, this issue may allow overwriting of sensitive memory areas of IoT device.

**Type:**

- Out-of-bounds memory write

**Result:**

- Memory corruption
- Memory write to initialized variables segment with arbitrary data

#### Target(s) affected by this defect ?

- contiki-ng v4.5
- contiki-ng v4.4

#### Fix

Rudimentary fix to address the most critical aspect of the issue:
https://github.com/mjurczak/contiki-ng/tree/bugfix/snmp-engine 

#### How is this defect reproduced ? 

An example hex-encoded SNMP request causing out-of-bounds memory write to varbinds:

```
306002010004067075626C6963A0530201290201000201003048301606122B06010401817D0840040201070A86DEB7380500301606122B06010401817D084004
0201070A86DEB7360500301606122B06010401817D0840040201050A86DEB9600500
```

g-oikonomou, 2020-11-25:
@Yagoor @mjurczak: Am I right to assume that this has been fixed in #1355 and/or #1397? Can we close?

nvt, 2026-09-30:
Fixed by #1355 and #1397.

closed 2026-09-30 by nvt