contiki-ng/contiki-ng #1353: Stack overflow in SNMP bulk request processing

mjurczak, 2020-08-17:
### Description of defect
**References:**

https://github.com/contiki-ng/contiki-ng/tree/release/v4.5
https://github.com/contiki-ng/contiki-ng/tree/release/v4.4

**File:**

[snmp-engine.c](https://github.com/contiki-ng/contiki-ng/blob/release/v4.5/os/net/app-layer/snmp/snmp-engine.c)
[snmp-message.c](https://github.com/contiki-ng/contiki-ng/blob/release/v4.5/os/net/app-layer/snmp/snmp-message.c)

**Analysis:**

Memory access out of buffer boundaries may occur if an SNMP bulk get request with number of OIDs larger than supported by the engine is received and processed.

The OIDs listed in a request are processed by snmp_message_decode() function without verification of the varbinds buffer capacity.
The varbinds memory buffer is written with the values provided in SNMP request:
https://github.com/contiki-ng/contiki-ng/blob/23db95746afcaa40df3e742a88fc4877d1ec4dd2/os/net/app-layer/snmp/snmp-message.c#L245
The buffer capacity is determined by:
https://github.com/contiki-ng/contiki-ng/blob/23db95746afcaa40df3e742a88fc4877d1ec4dd2/os/net/app-layer/snmp/snmp-conf.h#L81-L87

SNMP get bulk requests are processed by snmp_engine_get_bulk() function that allocates a local stack buffer for buffering OIDs of the requested variables.
https://github.com/contiki-ng/contiki-ng/blob/23db95746afcaa40df3e742a88fc4877d1ec4dd2/os/net/app-layer/snmp/snmp-engine.c#L116-L121

The stack buffer in snmp_engine_get_bulk() is populated with OIDs as a first step before any further processing of the data.
https://github.com/contiki-ng/contiki-ng/blob/23db95746afcaa40df3e742a88fc4877d1ec4dd2/os/net/app-layer/snmp/snmp-engine.c#L123-L130

The varbinds_length variable value is not verified against the capacity of the temporary oid stack buffer. If the number of requested OIDs exceeds the buffer capacity a stack buffer overflow condition occurs and stack memory beyond the allocated oid buffer is overwritten with OIDs received in SNMP get bulk request.

As the OIDs are supplied in the request content it may be possible to alter the return address from the snmp_engine_get_bulk() function. If the target architecture uses common addressing space for program and data memory (which is common in IoT devices) it may also be possible to supply code in the SNMP request payload and redirect the execution path to the injected code by modification of the return address.

**Type:**

- Out-of-bounds memory write
- Stack memory overwrite
- Return address alteration
- Remote altering of code execution path
- Remote executable code injection
- Remote code execution

**Result:**

- Memory corruption
- Remote code execution

#### Target(s) affected by this defect ?

- contiki-ng v4.5
- contiki-ng v4.4

#### Fix

Rudimentary fix to address the most critical aspect of the issue:
https://github.com/mjurczak/contiki-ng/tree/bugfix/snmp-engine 

#### How is this defect reproduced ? 

An example SNMP request causing stack overwrite:

```
306102010104067075626C6963A554020431D065A702010402010A3046300C06082B060102010102000500300C06082B060102010102010500300C06082B0601
02010102020500300C06082B060102010102030500300C06082B060102010102040500
```

g-oikonomou, 2020-11-25:
@Yagoor @mjurczak: Am I right to assume that this has been fixed in #1355 and/or #1397? Can we close?

Karthikdhondi, 2023-04-16:
Hey! I am working on this issue, as a project for my course work. Can you help me out with this?

nvt, 2026-09-30:
Fixed by #1355 and #1397 .

closed 2026-09-30 by nvt