contiki-ng/contiki-ng #1354: Out of bounds memory access in SNMP BER decoder/encoder routines

mjurczak, 2020-08-17:
### Description of defect
**References:**
https://github.com/contiki-ng/contiki-ng/tree/release/v4.5
https://github.com/contiki-ng/contiki-ng/tree/release/v4.4

**File:**

[snmp-engine.c](https://github.com/contiki-ng/contiki-ng/blob/release/v4.5/os/net/app-layer/snmp/snmp-ber.c)

**Analysis:**

Memory access out of buffer boundaries may occur if an SNMP ASN.1 BER encoder/decoder routines.

The length of provided input/output buffers is insufficiently verified when encoding and decoding data. Lack of boundary checks may lead to out-of-bounds buffer read or write access.

Example functions that make access to memory without prior verification of sufficient input data length:
https://github.com/contiki-ng/contiki-ng/blob/31753fed81d980ef1b0de477921ab977f4c1a552/os/net/app-layer/snmp/snmp-ber.c#L129
https://github.com/contiki-ng/contiki-ng/blob/31753fed81d980ef1b0de477921ab977f4c1a552/os/net/app-layer/snmp/snmp-ber.c#L138
https://github.com/contiki-ng/contiki-ng/blob/31753fed81d980ef1b0de477921ab977f4c1a552/os/net/app-layer/snmp/snmp-ber.c#L147
https://github.com/contiki-ng/contiki-ng/blob/31753fed81d980ef1b0de477921ab977f4c1a552/os/net/app-layer/snmp/snmp-ber.c#L183
https://github.com/contiki-ng/contiki-ng/blob/31753fed81d980ef1b0de477921ab977f4c1a552/os/net/app-layer/snmp/snmp-ber.c#L264

**Type:**

- Out-of-bounds memory access

**Result:**

- Memory corruption
- Invalid memory read access

#### Target(s) affected by this defect ?

- contiki-ng v4.5
- contiki-ng v4.4

g-oikonomou, 2020-11-25:
@Yagoor @mjurczak: Am I right to assume that this has been fixed in #1355 and/or #1397? Can we close?

nvt, 2026-09-30:
Fixed by #1355 and #1397.

closed 2026-09-30 by nvt