contiki-ng/contiki-ng #1359: tunslip6 will broken in some platform

GeneKong, 2020-08-26:
After test I found some verbose like 2,3,5 will case tunslip6 broken in my embedded board.
When it received string like:
```
Packet from TUN of length 104 - write SLIP
0000 60 0a 7c 92 00 40 3a 40 fd 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 fd 00 00 00 00 00 00 00 02 12 4b 00 1c a7 6e a4 80 00 fb 23 71 f8 00 01 99 fc c6 5d eb a9 09 00 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
`
Packet from SLIP of length 102 - write TUN
0000 7c 92 00 40 3a 3f fd 00 00 00 00 00 00 00 02 12 4b 00 1c a7 6e a4 fd 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 81 00 fa 23 71 f8 00 01 99 fc c6 5d eb a9 09 00 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
tunslip6: serial_to_tun: write: Invalid argument
ifconfig tun0 down
netstat -nr | awk '{ if ($2 == "tun0") print "route delete -net "$1; }' | sh
```
After review code I found this:
```c
 /* Echo lines as they are received for verbose=2,3,5+ */
    /* Echo all printable characters for verbose==4 */
    if((verbose==2) || (verbose==3) || (verbose>4)) {
      if(c=='\n') {
        if(is_sensible_string(uip.inbuf, inbufptr)) {
          if (timestamp) stamptime();
          fwrite(uip.inbuf, inbufptr, 1, stdout);
          inbufptr=0;
        }
      }
    } else if(verbose==4) {
      if(c == 0 || c == '\r' || c == '\n' || c == '\t' || (c >= ' ' && c <= '~')) {
        fwrite(&c, 1, 1, stdout);
        if(c=='\n') if(timestamp) stamptime();
      }
    }
```
`0a` in stream maybe broken work stream in some platform, I test linux PC, it's work fine.
Please check why this problem occur, how to solved it.
Thanks.

g-oikonomou, 2020-08-27:
I think you may be onto something. I am not 100% sure what you are trying to show with the packet bytes in the tunslip output.

GeneKong, 2020-08-28:
just ping address like
```
ping fe80::212:4b00:1ca1:9464
```

alexstanoev, 2020-09-17:
This seems to be related to automatic flow label generation on Linux combined with tunslip's heuristic to demux printed text from packets. Since the flow label field occupies bits 12 to 31 in the header, and the first byte is 60 (ASCII \`) all we need is the second byte to be `0a` to hit this.

I have seen this happen on packets starting with `60 0a`, which is actually \`\n and a perfectly sensible string according to `is_sensible_string`.
From ` __skb_get_hash` in `net/core/flow_dissector.c` in Linux, "This function calculates a flow hash based on src/dst addresses and src/dst port numbers.", hence why we only see this (consistently, but very rarely) on random boards.

Disabling automatic flow labels with `echo 0 > /proc/sys/net/ipv6/auto_flowlabels` turns the first 4 bytes to `60 00 00 00` and things work fine.

How do we want to work around this? While `60 0a` hits this, it's technically possible to get `60 XX 0a` where XX is valid ASCII, or other combinations.

g-oikonomou, 2020-09-17:
> How do we want to work around this?

IMHO we fix the heuristic in tunslip

closed 2020-10-22 by g-oikonomou