contiki-ng/contiki-ng #1365: Byte loss over slip on platform nrf52840

g-oikonomou, 2020-08-31:
There appears to be something fishy going on over SLIP for the nrf52840 platform.

Here is how to reproduce:

* Build the rpl-border-router example for platform nrf52840
* Program the device
* run tunslip6
* ping6 -c1 the device (won't work)
* ping6 -c1 the device again (it will start working after a few attempts)
* restart tunslip with the device running. Again, ping6 will stop working for a bit
* restart the device with tunslip running. This seems to fix things for a few pings, until we get some losses again.

Some of those pings result in unexplained backtick characters appearing in the tunslip console:

```
$ sudo ../../tools/serial-io/tunslip6 -s /dev/tty.usbmodem0006839521871 fd00::1/64
********SLIP started on ``/dev/tty.usbmodem0006839521871''
opened tun device ``/dev/tun0''
ifconfig tun0 inet6 mtu 1500 up
ifconfig tun0 inet6 fd00::1/64 add
sysctl -w net.inet6.ip6.forwarding=1
net.inet6.ip6.forwarding: 1 -> 1
ifconfig tun0

tun0: flags=8851<UP,POINTOPOINT,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	inet6 fe80::8e85:90ff:fe07:d9db%tun0 prefixlen 64 optimistic scopeid 0xf 
	inet6 fd00::1 prefixlen 64 tentative 
	nd6 options=201<PERFORMNUD,DAD>
	open (pid 66015)
`
`
`
`
```

This is on macOS Catalina - v10.15.6 (19G2021) with #1316 

@Robert-Lasater @alexstanoev

g-oikonomou, 2020-08-31:
```
$ sudo ../../tools/serial-io/tunslip6 -B 9600 -s /dev/tty.usbmodem0006839521871 fd00::1/64 
Password:
********SLIP started on ``/dev/tty.usbmodem0006839521871''
opened tun device ``/dev/tun0''
ifconfig tun0 inet6 mtu 1500 up
ifconfig tun0 inet6 fd00::1/64 add
sysctl -w net.inet6.ip6.forwarding=1
net.inet6.ip6.forwarding: 1 -> 1
ifconfig tun0

tun0: flags=8851<UP,POINTOPOINT,RUNNING,SIMPLEX,MULTICAST> mtu 1500
    inet6 fe80::8e85:90ff:fe07:d9db%tun0 prefixlen 64 optimistic scopeid 0xf 
    inet6 fd00::1 prefixlen 64 tentative 
    nd6 options=201<PERFORMNUD,DAD>
    open (pid 67365)
[INFO: BR        ] Waiting for prefix
*** Address:fd00::1 => fd00:0000:0000:0000
[INFO: BR        ] Waiting for prefix
*** Address:fd00::1 => fd00:0000:0000:0000
[INFO: BR        ] Waiting for prefix
*** Address:fd00::1 => fd00:0000:0000:0000
[INFO: BR        ] Waiting for prefix
[INFO: BR        ] Server IPv6 addresses:
[INFO: BR        ]   fd00::f6ce:3676:7654:84e8
[INFO: BR        ]   fe80::f6ce:3676:7654:84e8
```

This `[INFO: BR        ] Waiting for prefix` should not be happening x3 either and I suspect this to be the same root cause.

Robert-Lasater, 2020-08-31:
Thank you for opening this!

First a description of the problem as I am experiencing it.

Linux laptop with Mint 18, nR52840 board connected over USB, programmed with the RPL Border Router,  Reset the board; it comes up and tunslip6 sends the board its IPv6 prefix.  No problem.

ping6 command though always fails.  No packet received from the board.

Now a work-around.  Rework tunslip6 source code so each byte in a packet is written separately, with a 1 microsecond delay between each write.  Now ping6 works reliably and the HTTP server in the RPL Border Router can be accessed from Linux.

This is NOT meant to be a fix.  It suggests that there is a race condition, most likely in the nRF52480 serial driver or device.

Robert-Lasater, 2020-09-01:
Another update: I switch the host linux to a Raspberry PI.  Here short pin6 packets (-s10) worked; longer packets (-s64, the default) failed.  With the change to tunslip6 described in my previous comment, ping6 works (so far) without incident.

This observation suggests the failure is in the nRF52840 board, either the UART driver or the UART itself.

alexstanoev, 2020-09-22:
@Robert-Lasater could you please give the following branch a try: 
https://github.com/toshiba-bril/contiki-ng/tree/feature/nrf52840-uart-hwfc

I haven't been able to reliably reproduce this issue, but this will enable flow control on the DK, as well as hopefully handle RX errors better. 
If you still see issues, could you try adding `-v0` to tunslip6 too.

Robert-Lasater, 2020-09-23:
Hello @alexstanoev I want to acknowledge your request.  It may take a day or two before I can try it out.  The target nRF52840 board right now connects to a Raspberry PI and Nordic (unfortunately) does not provide a ready way to program their boards over a serial/ USB port from the ARM platform.  I will let you know.

Also, I want to test this for a while, since its surely a timing problem or a race condition.

alexstanoev, 2020-09-23:
Hi @Robert-Lasater, thanks for taking the time to test this!

If it helps, you can also program the nRF52840 boards from a Raspberry Pi using SEGGER JLink directly.
You need J-Link Software and Documentation pack for Linux ARM from https://www.segger.com/downloads/jlink/#J-LinkSoftwareAndDocumentationPack

You can then program a board by running `JLinkExe -device nrf52840_xxaa -if swd -speed 1000 -autoconnect 1` and typing the following commands:
```
loadbin yourfile.hex, 0
r
g
exit
```

Robert-Lasater, 2020-09-25:
Hello @alexstanoev, thank you for working this problem. Unfortunately with your build I still cannot ping the RPL border router build after enabling it with the tunslip6 program.

This build is a bit of an improvement.  The RPL border router does get its IPv6 address from tunslip6; that was not happening previously.

I regret I don't have the bandwidth to debug this further on my own.  Let me know if you have something you want me to try.

g-oikonomou, 2020-10-22:
#1400

closed 2020-10-22 by g-oikonomou

ais65, 2021-04-28:
@Robert-Lasater and @alexstanoev I experience the same problem with pinging the RPL Border router when I'm using NRF52840 and Raspberry 4 running Ubuntu 20. Did you conclude any root cause? ping6 -s 12 works fine but with no -s extension the links seems to stop.

alexstanoev, 2021-04-28:
Hi @ais65,

Could you please provide some more information about your setup?
Are you using an nRF52840 devkit, dongle, or custom hardware?

If this is a DK board could you please try the native USB interface by connecting a second cable to the nRF USB connector on the board and building with the `NRF52840_NATIVE_USB=1` make option (you'll need to run make clean beforehand too).

As this has been a very intermittent issue I haven't been able to rule out the possibility that it's caused by the UART to USB conversion inside the JLink on the devkit.

ais65, 2021-04-28:
It is a DK so I can test with NRF52840_NATIVE_USB instead.

ais65, 2021-04-28:
@alexstanoev that works much better. No problem to ping any longer. Thanks for your feedback.

MBradbury, 2022-02-17:
I recently encountered this and found that my cause for byte loss was described here: https://github.com/openthread/openthread/issues/2857. The solution was to issue `msddisable` in JLinkExe, which then allows up to 512 bytes to be sent instead of the 64 bytes allowed when the mass storage device is enabled. Some more information is available here: https://wiki.segger.com/J-Link_OB#Limitations_under_Mac_OS_X

alexstanoev, 2022-02-17:
Thanks for reporting this @MBradbury, great to have this one figured out! 
I have added a note about disabling MSD on the wiki [here](https://github.com/contiki-ng/contiki-ng/wiki/Platform-nrf52840#virtual-com-and-real-time-transfer).

This lines up with my inability to reproduce the issue, as all of my devkits have had MSD disabled.