contiki-ng/contiki-ng #1368: TSCH CC1352P1 + CC1312 do not communicate

andresblanco06, 2020-09-04:
Setup:
CC1352P1 as TSCH Border router 

_make sure you compile the border router with this define:
CFLAGS += -DSET_CCFG_MODE_CONF_XOSC_CAP_MOD=0x0
CFLAGS += -DSET_CCFG_MODE_CONF_XOSC_CAPARRAY_DELTA=0xC1_

CC1312R1 as SimpleNode

The CC1312 is able to receive EB from the CC1352P1 but when the CC1312R1 responds and tries to finish the association process the CC1352P1 never sees the packets. I know the packets are flowing by using sniffers and I am also sure they are both listening at the same frequencies by jamming and forcing them to go into the rx_cb using an interferer in smartRfstudio.

I think the issue might have to do with the synchronizations when timestamps are received. TSCH uses some parameters to adjust for crystal PPM and these boards use different crystals on board.

andresblanco06, 2020-09-04:
CC1312R1 print out:

```
[INFO: TSCH      ] scanning on channel 20
[WARN: Radio     ] Radio is already on
[WARN: Radio     ] Radio is already on
[INFO: Radio     ] Reading (35): 40eba581ffff80c2a11c004b1200003f1188061a2c0100000000011c0001c800011b00
[INFO: Radio     ] rssi: -15
[INFO: Radio     ] poll Mode?: 1
[INFO: TSCH      ] scan: received packet (35 bytes) on channel 20
[DBG : Frame 15.4] ie type 0, current state 0
[DBG : Frame 15.4] header ie: len 0 id 7e
[DBG : Frame 15.4] list termination 1, look for payload IEs
[DBG : Frame 15.4] ie type 1, current state 1
[DBG : Frame 15.4] payload ie: len 17 id 1
[DBG : Frame 15.4] entering MLME ie with len 17
[DBG : Frame 15.4] ie type 0, current state 2
[DBG : Frame 15.4] short mlme ie len 6 id 1a
[DBG : Frame 15.4] ie type 0, current state 2
[DBG : Frame 15.4] short mlme ie len 1 id 1c
[DBG : Frame 15.4] ie type 1, current state 2
[DBG : Frame 15.4] long mlme ie len 1 id 9
[DBG : Frame 15.4] ie type 0, current state 2
[DBG : Frame 15.4] short mlme ie len 1 id 1b
[DBG : Frame 15.4] end of MLME IE parsing
[INFO: TSCH      ] parse_eb: no schedule, setting up minimal schedule
[INFO: TSCH Sched] add_slotframe 0 3
[INFO: TSCH Sched] add_link sf=0 opt=Tx|Rx|Sh type=ADV ts=0 ch=0 addr=ffff.ffff.ffff.ffff
[INFO: TSCH Queue] update time source: (NULL LL addr) -> 0012.4b00.1ca1.c280
[INFO: TSCH      ] association done (1), sec 0, PAN ID 81a5, asn-0.12c, jp 1, timeslot id 0, hopping id 0, slotframe len 0 with 0 links, from 0012.4b00.1ca1.c280
[INFO: TSCH-LOG  ] {asn 00.0000012f link  0   3   0  0  0 ch  0} !dl-miss assoc 16408 7864
[INFO: TSCH-LOG  ] {asn 00.00000132 link  0   3   0  0  0 ch  0} !dl-miss assoc 8550 7864
[DBG : Stack     ] Check ok: 880 vs. 59256
```

andresblanco06, 2020-09-04:
CC1352P1 Print out:
```
[INFO: RPL       ] IPv6 addresses:
[INFO: RPL       ] -- fd00::212:4b00:1ca1:c280
[INFO: RPL       ] -- fe80::212:4b00:1ca1:c280
[INFO: RPL       ] reset DIO timer (Init root)
[INFO: RPL       ] created DAG with instance ID 0, DAG ID fd00::212:4b00:1ca1:c280, rank 128
[INFO: RPL       ] created a new RPL DAG
[INFO: BR        ] Waiting for prefix
[INFO: TSCH Sched] add_slotframe 0 3
[INFO: TSCH Sched] add_link sf=0 opt=Tx|Rx|Sh type=ADV ts=0 ch=0 addr=ffff.ffff.ffff.ffff
[INFO: TSCH      ] starting as coordinator, PAN ID 81a5, asn-0.0
[INFO: TSCH      ] starting as coordinator
[INFO: BR        ] Server IPv6 addresses:
[INFO: BR        ]   fd00::212:4b00:1ca1:c280
[INFO: BR        ]   fe80::212:4b00:1ca1:c280
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 126, status 0, tx 1
[INFO: RPL       ] sending a multicast-DIO with rank 128 to ff02::1a
[INFO: ICMPv6    ] Sending ICMPv6 packet to ff02::1a, type 155, code 1, len 72
[INFO: TCP/IP    ] output: sending to (NULL LL addr)
[INFO: 6LoWPAN   ] output: sending IPv6 packet with len 116
[INFO: 6LoWPAN   ] output: header len 40 -> 4, total len 116 -> 80, MAC max payload 111, frag_needed 0
[INFO: TSCH      ] send packet to ffff.ffff.ffff.ffff with seqno 196, queue 1/8 1/8, len 14 94
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 127, status 0, tx 1
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 128, status 0, tx 1
[INFO: RPL       ] sending a multicast-DIO with rank 128 to ff02::1a
[INFO: ICMPv6    ] Sending ICMPv6 packet to ff02::1a, type 155, code 1, len 72
[INFO: TCP/IP    ] output: sending to (NULL LL addr)
[INFO: 6LoWPAN   ] output: sending IPv6 packet with len 116
[INFO: 6LoWPAN   ] output: header len 40 -> 4, total len 116 -> 80, MAC max payload 111, frag_needed 0
[INFO: TSCH      ] send packet to ffff.ffff.ffff.ffff with seqno 196, queue 1/8 1/8, len 14 94
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 129, status 0, tx 1
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 130, status 0, tx 1
[INFO: RPL       ] sending a multicast-DIO with rank 128 to ff02::1a
[INFO: ICMPv6    ] Sending ICMPv6 packet to ff02::1a, type 155, code 1, len 72
[INFO: TCP/IP    ] output: sending to (NULL LL addr)
[INFO: 6LoWPAN   ] output: sending IPv6 packet with len 116
[INFO: 6LoWPAN   ] output: header len 40 -> 4, total len 116 -> 80, MAC max payload 111, frag_needed 0
[INFO: TSCH      ] send packet to ffff.ffff.ffff.ffff with seqno 196, queue 1/8 1/8, len 14 94
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 131, status 0, tx 1
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 132, status 0, tx 1
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 133, status 0, tx 1
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 134, status 0, tx 1
[INFO: RPL       ] sending a multicast-DIO with rank 128 to ff02::1a
[INFO: ICMPv6    ] Sending ICMPv6 packet to ff02::1a, type 155, code 1, len 72
[INFO: TCP/IP    ] output: sending to (NULL LL addr)
[INFO: 6LoWPAN   ] output: sending IPv6 packet with len 116
[INFO: 6LoWPAN   ] output: header len 40 -> 4, total len 116 -> 80, MAC max payload 111, frag_needed 0
[INFO: TSCH      ] send packet to ffff.ffff.ffff.ffff with seqno 196, queue 1/8 1/8, len 14 94
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 135, status 0, tx 1
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:1ca1:c280, DAG state: reachable, MOP 1 OCP 1 rank 128 max-rank 65535, dioint 15, nbr count 0 (Periodic)
[INFO: RPL       ] nbr: end of list
[INFO: RPL       ] No routing links
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 136, status 0, tx 1
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 137, status 0, tx 1
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 138, status 0, tx 1
```

atiselsts, 2020-09-04:
Does CC1312R1 ever attempt sending any packets after the association?

andresblanco06, 2020-09-04:
They start drifting apart from each other until the CC1312 resync with another EB.
Yes, the CC1312 is constantly sending messages after it syncs with the EB but the CC1352P1 does not respond because it never listens to them.

Sometimes the packets sent from CC1312 do not line up with the RX window of the CC1352P. 
but when they do line up, the CC1352P does not enter the read function inside prop_mode.c

andresblanco06, 2020-09-04:
UPDATE:
I just went ahead and tested a CC1312 + CC1312 network and I am having the same issue. 
The node device tries to finish association and never gets responses from coordinator.

See bellow the logs:

COORDINATOR:
```
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 169, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.000006ab link  0   3   0  0  0 ch 20} bc-0-0 tx LL-74c7->LL-NULL, len  35, seq 169, st 0  1
[DBG : Stack     ] Check ok: 1392 vs. 58312
[DBG : Stack     ] Check ok: 1392 vs. 58312
[DBG : TSCH Queue] packet is added put_index 0, packet #0x20003840
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 170, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.000007dd link  0   3   0  0  0 ch 20} bc-0-0 tx LL-74c7->LL-NULL, len  35, seq 170, st 0  1
[DBG : Stack     ] Check ok: 1392 vs. 58312
[DBG : TSCH Queue] packet is added put_index 1, packet #0x20003840
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 171, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.00000960 link  0   3   0  0  0 ch 20} bc-0-0 tx LL-74c7->LL-NULL, len  35, seq 171, st 0  1
[DBG : Stack     ] Check ok: 1392 vs. 58312
[INFO: RPL       ] sending a multicast-DIO with rank 128 to ff02::1a
[DBG : IPv6      ] Upper layer checksum len: 76 from: 40
[INFO: ICMPv6    ] Sending ICMPv6 packet to ff02::1a, type 155, code 1, len 72
[INFO: TCP/IP    ] output: sending to (NULL LL addr)
[INFO: 6LoWPAN   ] output: sending IPv6 packet with len 116
[DBG : 6LoWPAN   ] compression: before (76): 60000000004c3a40fe8000000000000002124b001e1774c7ff02000000000000000000000000001a9b01454700f0008008f00000fd0000000000000002124b001e1774c7040e00080c00040000800001001e003c081e4040ffffffffffffffff00000000fd000000000000000000000000000000
[DBG : 6LoWPAN   ] compression: first header: 58
[DBG : 6LoWPAN   ] compression: last header could is not compressed: 58
[DBG : 6LoWPAN   ] compression: after (4): 7a3b3a1a
[INFO: 6LoWPAN   ] output: header len 40 -> 4, total len 116 -> 80, MAC max payload 111, frag_needed 0
[DBG : TSCH Queue] packet is added put_index 4, packet #0x20003840
[INFO: TSCH      ] send packet to ffff.ffff.ffff.ffff with seqno 195, queue 1/8 1/8, len 14 94
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 172, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.00000a7d link  0   3   0  0  0 ch 20} bc-1-0 tx LL-74c7->LL-NULL, len  94, seq 172, st 0  1
[DBG : Stack     ] Check ok: 1392 vs. 58312
[DBG : TSCH Queue] packet is added put_index 2, packet #0x20003840
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 173, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.00000ab9 link  0   3   0  0  0 ch 20} bc-0-0 tx LL-74c7->LL-NULL, len  35, seq 173, st 0  1
[DBG : Stack     ] Check ok: 1392 vs. 58312
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:1e17:74c7, DAG state: reachable, MOP 1 OCP 1 rank 128 max-rank 65535, dioint 16, nbr count 0 (Periodic)
[INFO: RPL       ] nbr: end of list
[INFO: RPL       ] No routing links
[DBG : TSCH Queue] packet is added put_index 3, packet #0x20003840
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 174, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.00000c24 link  0   3   0  0  0 ch 20} bc-0-0 tx LL-74c7->LL-NULL, len  35, seq 174, st 0  1
[DBG : Stack     ] Check ok: 1392 vs. 58312
[DBG : Stack     ] Check ok: 1392 vs. 58312
[DBG : TSCH Queue] packet is added put_index 4, packet #0x20003840
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 175, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.00000d8c link  0   3   0  0  0 ch 20} bc-0-0 tx LL-74c7->LL-NULL, len  35, seq 175, st 0  1
[DBG : Stack     ] Check ok: 1392 vs. 58312
[DBG : TSCH Queue] packet is added put_index 5, packet #0x20003840
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 176, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.00000ee8 link  0   3   0  0  0 ch 20} bc-0-0 tx LL-74c7->LL-NULL, len  35, seq 176, st 0  1
[DBG : Stack     ] Check ok: 1392 vs. 58312
[DBG : TSCH Queue] packet is added put_index 6, packet #0x20003840
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 177, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.00001017 link  0   3   0  0  0 ch 20} bc-0-0 tx LL-74c7->LL-NULL, len  35, seq 177, st 0  1
[DBG : Stack     ] Check ok: 1392 vs. 58312
[DBG : TSCH Queue] packet is added put_index 7, packet #0x20003840
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 178, status 0, tx 1
[... 140 lines cut ...]
[DBG : 6LoWPAN   ] compression: last header could is not compressed: 58
[DBG : 6LoWPAN   ] compression: after (4): 7a3b3a1a
[INFO: 6LoWPAN   ] output: header len 40 -> 4, total len 46 -> 10, MAC max payload 111, frag_needed 0
[INFO: Frame 15.4] Out:  1 ffff.0000.0000.0000 14 10 (24)
[DBG : TSCH Queue] packet is added put_index 7, packet #0x200038ac
[INFO: TSCH      ] send packet to ffff.ffff.ffff.ffff with seqno 220, queue 1/8 2/8, len 14 24
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 195, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.00001671 link  0   3   0  0  0 ch 20} bc-1-0 tx LL-7b1d->LL-NULL, len  24, seq 195, st 0  1
[... cut ...]
[INFO: TSCH      ] send packet to 0012.4b00.1e17.74c7 with seqno 221, queue 1/8 1/8, len 21 21
[INFO: TSCH      ] sending KA to 0012.4b00.1e17.74c7
[INFO: TSCH-LOG  ] {asn 00.00001761 link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-74c7, len  21, seq 221, st 2  1
[DBG : Stack     ] Check ok: 1336 vs. 59256
[INFO: TSCH-LOG  ] {asn 00.000017ac link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-74c7, len  21, seq 221, st 2  2
[INFO: TSCH-LOG  ] {asn 00.000017be link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-74c7, len  21, seq 221, st 2  3
```

atiselsts, 2020-09-04:
Hi,

I suggest configuring your logs like this to reduce the spam:

```
#define LOG_CONF_LEVEL_MAIN                        LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_RPL                         LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_TCPIP                       LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_IPV6                        LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_6LOWPAN                     LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_MAC                         LOG_LEVEL_INFO
#define LOG_CONF_LEVEL_FRAMER                      LOG_LEVEL_WARN
#define TSCH_LOG_CONF_PER_SLOT                     1

```

Also, correct me if I'm wrong, but from the log it seems that the node is able to sync with the coordinator at least once, otherwise it would give up faster. However, the sync is not visible in the logs. Can you include the part where the node associates with the coordinator and what happens immediately (say 10 seconds) afterwards? It should include that first attemp to sync. That will give you a clue about the  RAT time offset correction needed to fix the timestamps.

andresblanco06, 2020-09-04:
HEy Atis,

I di das suggested and that helped.

here is the output you wanted:

```
[INFO: TSCH      ] scanning on channel 20
[INFO: TSCH      ] scanning on channel 20
[INFO: TSCH      ] scanning on channel 20
[INFO: TSCH      ] scan: received packet (35 bytes) on channel 20
[INFO: TSCH      ] parse_eb: no schedule, setting up minimal schedule
[INFO: TSCH Sched] add_slotframe 0 3
[INFO: TSCH Sched] add_link sf=0 opt=Tx|Rx|Sh type=ADV ts=0 ch=0 addr=ffff.ffff.ffff.ffff
[INFO: TSCH Queue] update time source: (NULL LL addr) -> 0012.4b00.1ca1.c280
[INFO: TSCH      ] association done (1), sec 0, PAN ID 81a5, asn-0.54, jp 1, timeslot id 0, hopping id 0, slotframe len 0 with 0 links, from 0012.4b00.1ca1.c280
[INFO: TSCH      ] received from 0012.4b00.1ca1.c280 with seqno 65535
[INFO: TSCH-LOG  ] {asn 00.000000d5 link  0   3   0  0  0 ch 20} bc-1-0 rx LL-c280->LL-NULL, len  94, seq   0, edr   3, dr  -3
[INFO: TSCH-LOG  ] {asn 00.00000108 link  0   3   0  0  0 ch 20} bc-0-0 rx LL-c280->LL-NULL, len  35, seq   0, edr   1, dr  -1
[INFO: TSCH      ] send packet to 0012.4b00.1ca1.c280 with seqno 234, queue 1/8 1/8, len 21 100
[INFO: TSCH-LOG  ] {asn 00.0000011a link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len 100, seq 234, st 2  1
[INFO: TSCH-LOG  ] {asn 00.0000011d link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len 100, seq 234, st 2  2
[INFO: TSCH-LOG  ] {asn 00.00000123 link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len 100, seq 234, st 2  3
[INFO: TSCH-LOG  ] {asn 00.0000012c link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len 100, seq 234, st 2  4
[INFO: TSCH-LOG  ] {asn 00.0000013b link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len 100, seq 234, st 2  5
[INFO: TSCH-LOG  ] {asn 00.0000018f link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len 100, seq 234, st 2  6
[INFO: TSCH-LOG  ] {asn 00.000001c8 link  0   3   0  0  0 ch 20} drift -4 ppm (min/max delta seen: -3/0)
[INFO: TSCH-LOG  ] {asn 00.000001c8 link  0   3   0  0  0 ch 20} bc-0-0 rx LL-c280->LL-NULL, len  35, seq   0, edr   2, dr  -2
[INFO: TSCH-LOG  ] {asn 00.000001d7 link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len 100, seq 234, st 2  7
[INFO: TSCH      ] packet sent to 0012.4b00.1ca1.c280, seqno 234, status 2, tx 8
[INFO: TSCH-LOG  ] {asn 00.000001dd link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len 100, seq 234, st 2  8
[INFO: TSCH      ] received from 0012.4b00.1ca1.c280 with seqno 65535
[INFO: TSCH-LOG  ] {asn 00.000002af link  0   3   0  0  0 ch 20} drift -4 ppm (min/max delta seen: -3/0)
[INFO: TSCH-LOG  ] {asn 00.000002af link  0   3   0  0  0 ch 20} bc-1-0 rx LL-c280->LL-NULL, len  94, seq   0, edr   0, dr   0
[INFO: TSCH-LOG  ] {asn 00.0000030c link  0   3   0  0  0 ch 20} bc-0-0 rx LL-c280->LL-NULL, len  35, seq   0, edr   0, dr   0
[INFO: TSCH      ] send packet to ffff.ffff.ffff.ffff with seqno 234, queue 1/8 1/8, len 14 24
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 78, status 0, tx 1
[INFO: TSCH-LOG  ] {asn 00.00000417 link  0   3   0  0  0 ch 20} bc-1-0 tx LL-7b1d->LL-NULL, len  24, seq  78, st 0  1
[INFO: TSCH      ] send packet to 0012.4b00.1ca1.c280 with seqno 235, queue 1/8 1/8, len 21 21
[INFO: TSCH      ] sending KA to 0012.4b00.1ca1.c280
[INFO: TSCH-LOG  ] {asn 00.00000420 link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 235, st 2  1
[INFO: TSCH-LOG  ] {asn 00.0000043b link  0   3   0  0  0 ch 20} drift -4 ppm (min/max delta seen: -3/0)
[INFO: TSCH-LOG  ] {asn 00.0000043b link  0   3   0  0  0 ch 20} bc-0-0 rx LL-c280->LL-NULL, len  35, seq   0, edr   1, dr  -1
[INFO: TSCH-LOG  ] {asn 00.00000450 link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 235, st 2  2
[INFO: TSCH-LOG  ] {asn 00.00000489 link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 235, st 2  3
[INFO: TSCH      ] received from 0012.4b00.1ca1.c280 with seqno 65535
[INFO: TSCH-LOG  ] {asn 00.000004d1 link  0   3   0  0  0 ch 20} drift -5 ppm (min/max delta seen: -3/0)
[INFO: TSCH-LOG  ] {asn 00.000004d1 link  0   3   0  0  0 ch 20} bc-1-0 rx LL-c280->LL-NULL, len  94, seq   0, edr   2, dr  -2
[INFO: TSCH-LOG  ] {asn 00.000004da link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 235, st 2  4
[INFO: TSCH-LOG  ] {asn 00.0000052b link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 235, st 2  5
[INFO: TSCH-LOG  ] {asn 00.0000052e link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 235, st 2  6
[INFO: TSCH-LOG  ] {asn 00.0000055b link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 235, st 2  7
[INFO: TSCH      ] packet sent to 0012.4b00.1ca1.c280, seqno 235, status 2, tx 8
[INFO: TSCH      ] KA sent to 0012.4b00.1ca1.c280, st 2-8
[WARN: TSCH      ] re-synchronizing on 0012.4b00.1ca1.c280
[INFO: TSCH-LOG  ] {asn 00.0000055e link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 235, st 2  8
[INFO: TSCH      ] send packet to 0012.4b00.1ca1.c280 with seqno 236, queue 1/8 1/8, len 21 21
[INFO: TSCH      ] sending KA to 0012.4b00.1ca1.c280
[INFO: TSCH-LOG  ] {asn 00.0000056a link  0   3   0  0  0 ch 20} drift -4 ppm (min/max delta seen: -3/0)
[INFO: TSCH-LOG  ] {asn 00.0000056a link  0   3   0  0  0 ch 20} bc-0-0 rx LL-c280->LL-NULL, len  35, seq   0, edr  -3, dr   3
[INFO: TSCH-LOG  ] {asn 00.000005b5 link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 236, st 2  1
[... cut ...]
[INFO: TSCH      ] sending KA to 0012.4b00.1ca1.c280
[INFO: TSCH-LOG  ] {asn 00.0000068a link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 236, st 2  5
[INFO: TSCH-LOG  ] {asn 00.00000699 link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 236, st 2  6
[INFO: TSCH-LOG  ] {asn 00.000006e4 link  0   3   0  0  0 ch 20} uc-1-0 tx LL-7b1d->LL-c280, len  21, seq 236, st 2  7
[INFO: TSCH-LOG  ] {asn 00.000006f0 link  0   3   0  0  0 ch 20} drift -4 ppm (min/max delta seen: -3/3)
```