contiki-ng/contiki-ng #1370: CC13X2 TSCH RPL Border Router Crashing with Radio Module LOG

andresblanco06, 2020-09-04:
When enabling the RADIO Logging module in the RPL BORDER router example cause the device to crash and restart continuously after scheduling the EB.

See below the print outs:

```
[[INFO: Radio     ] TX power configured to 126 dBm
[INFO: Main      ] Starting Contiki-NG-release/v4.5-189-g29b4961ba-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: TSCH
[INFO: Main      ] - 802.15.4 PANID: 0x81a5
[INFO: Main      ] - 802.15.4 TSCH default hopping sequence length: 1
[INFO: Main      ] Node ID: 29895
[INFO: Main      ] Link-layer address: 0012.4b00.1e17.74c7
[INFO: IPv6 DS   ] Init: 16 neighbors
[INFO: IPv6 DS   ] 2 default routers
[INFO: IPv6 DS   ] 3 prefixes
[INFO: IPv6 DS   ] 0 routes
[INFO: IPv6 DS   ] 3 unicast addresses
[INFO: IPv6 DS   ] 7 multicast addresses
[INFO: IPv6 DS   ] 2 anycast addresses
[INFO: IPv6 DS   ] Adding prefix fe80::length 64, flags 0, Valid lifetime 0, Preffered lifetime 0
[INFO: RPL       ] initializing
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:1e17:74c7
[INFO: CC13xx/CC26xx] Configuring Pins
[INFO: CC13xx/CC26xx] Done with Pins 
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 20
[INFO: CC13xx/CC26xx] Node ID: 29895
[WARN: Radio     ] Radio is already off
[INFO: RPL BR    ] Contiki-NG Border Router started
[INFO: BR        ] RPL-Border router started
*** Address:fd00::1 => fd00:0000:0000:0000
[INFO: RPL       ] IPv6 addresses:
[INFO: RPL       ] -- fd00::212:4b00:1e17:74c7
[INFO: RPL       ] -- fe80::212:4b00:1e17:74c7
[INFO: RPL       ] reset DIO timer (Init root)
[INFO: RPL       ] created DAG with instance ID 0, DAG ID fd00::212:4b00:1e17:74c7, rank 128
[INFO: RPL       ] created a new RPL DAG
[INFO: BR        ] Waiting for prefix
[INFO: TSCH Sched] add_slotframe 0 3
[INFO: TSCH Sched] add_link sf=0 opt=Tx|Rx|Sh type=ADV ts=0 ch=0 addr=ffff.ffff.ffff.ffff
[INFO: TSCH      ] starting as coordinator, PAN ID 81a5, asn-0.0
[INFO: TSCH      ] starting as coordinator
[INFO: BR        ] Server IPv6 addresses:
[INFO: BR        ]   fd00::212:4b00:1e17:74c7
[INFO: BR        ]   fe80::212:4b00:1e17:74c7
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[[INFO: Radio     ] TX power configured to 126 dBm
[INFO: Main      ] Starting Contiki-NG-release/v4.5-189-g29b4961ba-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: TSCH
[INFO: Main      ] - 802.15.4 PANID: 0x81a5
[INFO: Main      ] - 802.15.4 TSCH default hopping sequence length: 1
[INFO: Main      ] Node ID: 29895
[INFO: Main      ] Link-layer address: 0012.4b00.1e17.74c7
[INFO: IPv6 DS   ] Init: 16 neighbors
[INFO: IPv6 DS   ] 2 default routers
[INFO: IPv6 DS   ] 3 prefixes
[INFO: IPv6 DS   ] 0 routes
[INFO: IPv6 DS   ] 3 unicast addresses
[INFO: IPv6 DS   ] 7 multicast addresses
[INFO: IPv6 DS   ] 2 anycast addresses
[INFO: IPv6 DS   ] Adding prefix fe80::length 64, flags 0, Valid lifetime 0, Preffered lifetime 0
[INFO: RPL       ] initializing
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:1e17:74c7
[INFO: CC13xx/CC26xx] Configuring Pins
[INFO: CC13xx/CC26xx] Done with Pins 
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 20
[INFO: CC13xx/CC26xx] Node ID: 29895
[WARN: Radio     ] Radio is already off
[INFO: RPL BR    ] Contiki-NG Border Router started
[INFO: BR        ] RPL-Border router started
*** Address:fd00::1 => fd00:0000:0000:0000
[INFO: RPL       ] IPv6 addresses:
[INFO: RPL       ] -- fd00::212:4b00:1e17:74c7
[INFO: RPL       ] -- fe80::212:4b00:1e17:74c7
[INFO: RPL       ] reset DIO timer (Init root)
[INFO: RPL       ] created DAG with instance ID 0, DAG ID fd00::212:4b00:1e17:74c7, rank 128
[INFO: RPL       ] created a new RPL DAG
[INFO: BR        ] Waiting for prefix
[INFO: TSCH Sched] add_slotframe 0 3
[INFO: TSCH Sched] add_link sf=0 opt=Tx|Rx|Sh type=ADV ts=0 ch=0 addr=ffff.ffff.ffff.ffff
[INFO: TSCH      ] starting as coordinator, PAN ID 81a5, asn-0.0
[INFO: TSCH      ] starting as coordinator
[INFO: BR        ] Server IPv6 addresses:
[INFO: BR        ]   fd00::212:4b00:1e17:74c7
[INFO: BR        ]   fe80::212:4b00:1e17:74c7
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[[INFO: Radio     ] TX power configured to 126 dBm
[INFO: Main      ] Starting Contiki-NG-release/v4.5-189-g29b4961ba-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: TSCH
[INFO: Main      ] - 802.15.4 PANID: 0x81a5
[INFO: Main      ] - 802.15.4 TSCH default hopping sequence length: 1
[INFO: Main      ] Node ID: 29895
[INFO: Main      ] Link-layer address: 0012.4b00.1e17.74c7
[INFO: IPv6 DS   ] Init: 16 neighbors
[INFO: IPv6 DS   ] 2 default routers
[INFO: IPv6 DS   ] 3 prefixes
[INFO: IPv6 DS   ] 0 routes
[INFO: IPv6 DS   ] 3 unicast addresses
[INFO: IPv6 DS   ] 7 multicast addresses
[INFO: IPv6 DS   ] 2 anycast addresses
[INFO: IPv6 DS   ] Adding prefix fe80::length 64, flags 0, Valid lifetime 0, Preffered lifetime 0
[INFO: RPL       ] initializing
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:1e17:74c7
[INFO: CC13xx/CC26xx] Configuring Pins
[INFO: CC13xx/CC26xx] Done with Pins 
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 20
[INFO: CC13xx/CC26xx] Node ID: 29895
[WARN: Radio     ] Radio is already off
[INFO: RPL BR    ] Contiki-NG Border Router started
[INFO: BR        ] RPL-Border router started
*** Address:fd00::1 => fd00:0000:0000:0000
[INFO: RPL       ] IPv6 addresses:
[INFO: RPL       ] -- fd00::212:4b00:1e17:74c7
[INFO: RPL       ] -- fe80::212:4b00:1e17:74c7
[INFO: RPL       ] reset DIO timer (Init root)
[INFO: RPL       ] created DAG with instance ID 0, DAG ID fd00::212:4b00:1e17:74c7, rank 128
[INFO: RPL       ] created a new RPL DAG
[INFO: BR        ] Waiting for prefix
[INFO: TSCH Sched] add_slotframe 0 3
[INFO: TSCH Sched] add_link sf=0 opt=Tx|Rx|Sh type=ADV ts=0 ch=0 addr=ffff.ffff.ffff.ffff
[INFO: TSCH      ] starting as coordinator, PAN ID 81a5, asn-0.0
[INFO: TSCH      ] starting as coordinator
[INFO: BR        ] Server IPv6 addresses:
[INFO: BR        ]   fd00::212:4b00:1e17:74c7
[INFO: BR        ]   fe80::212:4b00:1e17:74c7
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
```

andresblanco06, 2020-09-04:
Does not happen when I enable it on the simpleNode examples

atiselsts, 2020-10-14:
No updates on the issue, but I assume it is fixed by #1379.

closed 2020-10-14 by atiselsts