contiki-ng/contiki-ng #1371: accessing the external flash from the simplelink platform

hseokman, 2020-09-04:
I would like to use the external flash implementation in Contiki, but there are number of issues with the simplelink platform.
Header files order and inclusion and typo need to be fixed.
Also, I do not understand why bit banging on SPI in the init to put the flash into low power mode.
Here is my fix to make external flash to work on CC1312R1.
https://github.com/hseokman/contiki-ng/commit/313a8eb0c88a82bc422edea4631064a77a62adf1

g-oikonomou, 2020-11-25:
I just noticed this.

Do you have time to take your commit and turn it into a pull request so we can test things properly?

Do you have a particular device with an external flash on it in mind?

hseokman, 2020-12-02:
I only tested on CC1312R1, and I don't have other devices.
All other devices seem follow the similar code structure.
I was just exercising the external flash on the launchpad.
I can create pull request, but it is only for CC1312R1, I can make modification to all other platform, but they would be untested.

jeppenodgaard, 2021-07-07:
Anyone working on this or can I attempt to make a pr?

arurke, 2021-10-05:
@jeppenodgaard You got something brewing here or should I have a stab at it?

jeppenodgaard, 2021-10-05:
@arurke I got nothing. Stab away :+1:

David-Wallis-Seneco, 2021-10-06:
I got the test-coffee.c example partially running. I just added:

1. #include "Board.h" into board-conf.h.
2. #include "cc13xx-cc26xx-conf.h" into Board.h
3. MODULES += $(CONTIKI_NG_DRIVERS_STORAGE_DIR)/ext-flash
    CONTIKI_TARGET_SOURCEFILES += xmem.c
   into Makefile.cc13xx-cc26xx

The example worked partially. I had to put watchdog-periodic() in the loops, and it failed when the FILE_SIZE was left at 4096 (reducing to 2048 worked).