contiki-ng/contiki-ng #1382: MSPGCC right shift with weird behaviour

Yagoor, 2020-09-13:
### Description

When using the MSPGCC compiler on the cooja I noticed a weird behaviour in the right shift operation.

The expected outcome of the code below is:

1156212811
4516456
17642
68
0

However when using the MSPGCC I get the following outcome:

870914 node-1 1156212811
873918 node-1 15395432
876859 node-1 15395562
879861 node-1 15395562
882801 node-1 15395562

If the >> 8 operation is broken into >> 4 and >> 4 the code works fine.

I believe this is a compiler issue, but I am not sure.

### Example

```c

PROCESS(bug_process, "Bug process");
AUTOSTART_PROCESSES(&bug_process);

void
__MSPGCC__TEST(uint32_t number)
{
  uint32_t examples[5], i = 0;

  examples[i++] = number;

  do {
    if(i == 5) {
      break;
    }
    /* I'm not sure why but on MSPGCC the >> 8 operation goes haywire here */
    number >>= 8;
    examples[i++] = number;
  } while(number);

  printf("%" PRIu32 "\n", examples[0]);
  printf("%" PRIu32 "\n", examples[1]);
  printf("%" PRIu32 "\n", examples[2]);
  printf("%" PRIu32 "\n", examples[3]);
  printf("%" PRIu32 "\n", examples[4]);
}

/*---------------------------------------------------------------------------*/
PROCESS_THREAD(bug_process, ev, data)
{
  PROCESS_BEGIN();

  __MSPGCC__TEST(1156212811);

  PROCESS_END();
}
```

atiselsts, 2020-09-13:
mspgcc has had many bugs over the years. There is not much we can do except trying to test our code and trying avoiding writing code that breaks it.

Compiler upgrade is not realistically going to happen - it would be too much effort. It would only make sense if there's suddenly a new msp430 platform we want to support.

To verify that its compiler issue, you can try to generate assembly code (with -S switch) or by looking the binary with `msp430-objdump`.

Yagoor, 2020-09-13:
Thank you for your answer.
I assumed it was a compiler issue.
I'll close this issue then.

closed 2020-09-13 by Yagoor