contiki-ng/contiki-ng #1389: lwm2m build does not report its protocol version

Robert-Lasater, 2020-09-20:
Problem is at line 729 of `os/services/lwm2m/lwm2m-rd-client.c`:

`snprintf(query_data, sizeof(query_data) - 1, "?ep=%s&lt=%d&b=%s",
session_info->ep, session_info->lifetime, session_info->binding);`

Notice the endpoint name and the binding mode are included, but not the protocol version.  The wakaama LWM2M server thus refuses to accept a connection from this contiki-ng client.  Changing that line to:

'snprintf(query_data, sizeof(query_data) - 1, "?lwm2m=1.1&ep=%s&lt=%d&b=%s", session_info->ep, session_info->lifetime, session_info->binding);`

fixes this problem.  Of course a real fix should separate the protocol version, e.g., including it as a preprocessor definition.

g-oikonomou, 2020-09-20:
Thanks Robert, can you please also include the things you mentioned about truncated packets?

Robert-Lasater, 2020-09-20:
Yes; I prefer to create a separate issue for that.  The two are not really connected; I just happened to encounter both in my effort to get the contiki-ng build to work with that wakaama server.

g-oikonomou, 2020-09-20:
@bitvijays Hi Vijay this is what I was telling you about yesterday, Robert is having issues with Wakama and it doesn't look like  platform-specific.

Robert-Lasater, 2020-09-20:
Guys, at the risk of getting out of line, I have a proposed fix and would appreciate the opportunity to make a pull request.  It should be one changed line (see above) and one new line (defining the protocol version).

Thanks, Robert Lasater

joakimeriksson, 2020-09-20:
Yes, please do a pull request on this singe line. But maybe also add a constant somewhere - with the protocol-version (will turn into a few more lines.

joakimeriksson, 2020-09-20:
BTW: Do you also happen to work with improving Wakaama?

Robert-Lasater, 2020-09-20:
>BTW: Do you also happen to work with improving Wakaama?

No. What is that?
Also, George O pointed out that the "master" branch in the wakaama repository has not had a commit since February, 2019.

Robert-Lasater, 2020-09-20:
Regarding a pull request, contributing guidelines emphasize selecting the correct branch.  But I only see two branches, develop and master; nothing relating to individual releases.  Any pull request to address this issue should be done on the develop branch, correct?

g-oikonomou, 2020-09-20:
Yes please, pull against branch develop

Robert-Lasater, 2020-09-21:
One quick question / confirmation before I complete the pull request: the protocol version should be "1.1", correct?

joakimeriksson, 2020-09-21:
I would say the current version is closer to 1.0, there are new features in 1.1 that are not there yet. But allow configuration with a constant and it can easily be changed depending on the need of the server, etc.

Robert-Lasater, 2020-09-21:
OK, thanks for the clarification!

Robert-Lasater, 2020-09-21:
There should be a pull request to address this issue.  Regret any uncertainty but this is my first time and the github UI is not 100% clear.  Thanks.

Robert-Lasater, 2020-09-21:
Pull request complete; regret any confusion.

I see the CI process runs some 17 tests.  Is there anyway a potential contributor can run those tests offline?

g-oikonomou, 2020-09-21:
>  Is there anyway a potential contributor can run those tests offline?

Inside the docker container, run `make` under `tests/`

PR #1391 (2020-09-22): Include LWM2M protocol version in registration message
Adds a preprocessor definition LWM2M_PROTOCOL_VERSION; set it to "1.0".  Extend query_data to include this with the key "lwm2m=".

Fixes #1389
closed 2020-09-22 by nfi