contiki-ng/contiki-ng #1390: The value of COAP_MAX_CHUNK_SIZE in examples/lwm2m-ipso-objects/project-conf.h might be too small

Robert-Lasater, 2020-09-21:
Preprocessor symbol `COAP_MAX_CHUNK_SIZE` is set to 64 in `examples/lwm2m-ipso-objects/project-conf.h`.  But the payload, for which this symbol is the maximum length, is now 66 bytes.  The last two bytes thus get truncated. This causes a problem when registering with the wakaama LWM2M server.

g-oikonomou, 2020-09-21:
This could be a bug, will let Niclas and Joakim comment on it. Now that we've been discussing this, I think that native execution tests (similar to mqtt and others) could make sense

g-oikonomou, 2020-09-21:
Having said that, we do have `18-coap-lwm2m`, so perhaps we just need to enrich those if this is indeed a bug

Robert-Lasater, 2020-10-05:
As suggested, solving this just required changing `COAP_MAX_CHUNK_SIZE` in project_conf.h.  So addressing it in the contiki-ng code base has a bit lower priority for me.

This comment though bothers me:

    /* Increase rpl-border-router IP-buffer when using more than 64. */
    #define COAP_MAX_CHUNK_SIZE            96

as it suggests increasing `COAP_MAX_CHUNK_SIZE` could cause a buffer overrun or memory corruption.  But as I recall the buffer that the parameter `COAP_MAX_CHUNK_SIZE` works in the LWM2M build with has dimension 128.  If this comment is obsolete or incorrect, it should be removed.  If another buffer or #define needs to be changed, that needs to be identified better.