contiki-ng/contiki-ng #1401: cc1310 device reboot with 2.5kbps TSCH sub-GHz mode

lili1218, 2020-10-02:
hi everyone,
I am working on 2.5kbps  TSCH sub-GHz mode with rpl-lite and  target is cc26x0-cc13x0. my test scenario is 1 gateway with 15 nodes. it can successfully form a network , but some devices will reboot  at the same place after few hours running
 
**in prop-mode.c transmit()**
`    
while((cmd_tx_adv->status & RF_CORE_RADIO_OP_MASKED_STATUS)
          == RF_CORE_RADIO_OP_MASKED_STATUS_RUNNING) {
      if(!rf_core_poll_mode) {
        printf("PM ");
        lpm_sleep();
      }
    }`

it seems when device tries to transmit packet it will in this while loop until watchdog timeout( around 2 mins) causing device rebooting.
I tried to fix by replace while loop by 
`    RTIMER_BUSYWAIT_UNTIL((cmd_tx_adv->status & RF_CORE_RADIO_OP_MASKED_STATUS) !=
                         RF_CORE_RADIO_OP_MASKED_STATUS_RUNNING, 1000 * RF_CORE_TURN_OFF_TIMEOUT);`

with this fix, device won't reboot but will hang even watchdog is not working. Error message is as below

[INFO: TSCH-LOG  ] {asn 00.00074009 link  0  53   0  0  0 ch 13} bc-0-0 rx LL-0000->LL-NULL, len  35, seq  44, edr -355, dr 355
prop-mode:  prop_div_radio_setup: DIV_SETUP, CMDSTA=0x00000086, status=0x0000 
prop-mode:  on: prop_div_radio_setup() failed
prop-mode:  transmit: on() failed
rf-core: fs_powerdown: CMDSTA=0x00000086
rf-core: rf_core_get_rat_rtc_offset: SYNC_STOP_RAT fail, ret 0 CMDSTA=0x00000086                    
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 144, status 4, tx 1                    
[INFO: TSCH-LOG  ] {asn 00.0007400a link  0  53   0  1  0 ch 13} bc-0-0 tx LL-5a36->LL-NULL, len  35, seq 144, st 4  1
device hang here...

may I know what is the possible reason and way to fix it ? thanks

rajeev1986, 2020-10-02:
First of all, how did you configure the physical layer settings (smartrf-settings) for using 2.5kbps data rate?  Any particular reason for using this data rate?
Secondly, the TSCH is configured to work with 50kbps. Anything lower will require reconfiguration of TSCH timeslots as the radio will need to be kept on for longer to RX and ACK the packets properly, i.e., radio and network timings need to be tested and tuned accordingly.

The above incorrect settings will cause the radio driver to fail possibly locking the OS into some bad state, hence, reboots.

lili1218, 2020-10-05:
hi,
2.5kbps smartrf-setting file is copy from cc1310-lrm branch [https://github.com/tbriggs6/contiki-ng/tree/cc1310-lrm/arch/cpu/cc26x0-cc13x0/rf-core/lrm](url). in side contiki-ng/arch/cpu/cc26x0-cc13x0/rf-core/lrm/ there is smartrf-setting file for 5kbps.

we do a further modification in override to make it become 2.5kbps, one line change 

`
HW_REG_OVERRIDE(0x505C,0x0303)
`

and since range limitation, I want to have 2.5 kbps which has longer transmission range compared to 50 kbps . As for the 2.5kbps related TSCH timeslot, we did some tuning accordingly to make it works.

currently, it can successfully have rpl parent and transmitted data, but it will reboot after few days running. cans it still be TSCH timing issue?  I thought TSCH timing will only cause device failed to talk to other node if it is not properly designed.

alexrayne, 2020-10-05:
can you provide actual value of `cmd_tx_adv->status` when it fails?

try to print it like:
```
if ((cmd_tx_adv->status & RF_CORE_RADIO_OP_MASKED_STATUS) != RF_CORE_RADIO_OP_MASKED_STATUS_RUNNING) 
    PRINTF(..??? .)
```

lili1218, 2020-10-05:
hi alexrayne,

when it fails, it will return
`prop-mode:  transmit: Not Sent OK status=0x0002`
`prop-mode:  rf_cmd_prop_rx: send_cmd ret=0, CMDSTA=0x00000086, status=0x0000`

above log is based on replacing while loop with RTIMER_BUSYWAIT_UNTIL()

alexrayne, 2020-10-05:
our `CMDSTA` is `RF_CORE_CMDSTA_SCHEDULING_ERR` - this looks like transmit starts while other operation was acitve, possibly receiving, or any other.

to log such conflict, i have used logging commands to RF-core. like [here](https://github.com/alexrayne/contiki-ng/commit/d63db24a95a3e0b5383a3c264aa912faff5bcf2e)

atiselsts, 2020-10-05:
I'm shooting in the dark here, but perhaps the radio should be reset more aggressively after the first operation that fails.
For example, if `prop_div_radio_setup()` fails as in the example above, then perhaps the RF core should be turned off. The current code is 
```
  if(prop_div_radio_setup() != RF_CORE_CMD_OK) {
    PRINTF("on: prop_div_radio_setup() failed\n");
    return RF_CORE_CMD_ERROR;
  }
```
which could be changed by adding `rf_core_power_down();` call inside the if.

The question is of course why does it fail in the first place, which I believe has to do with the RF core (the Cortex-M0 microcontoller for the radio funcionality) hanging because some operations take too long in these low-rate modes. This is certainly not a problem with TSCH timing. The problem only shows up on this particular platform with specific settings; in a project with Rajeev we tried out the 5 kbps mode and also experienced some reboots.

alexrayne, 2020-10-05:
> if `prop_div_radio_setup()` fails as in the example above, then perhaps the RF core should be turned off.

i have meet some [situations ](https://github.com/alexrayne/contiki-ng/commit/5d1aead6b3d42b2002b8e58343f76948d8762d6d) when code forget power off() core after unfinished operation (possibly timedout sent aor receive) , and starts power on. There unfinished operetion is still active, and so go into receive mode fails with same error.

to handle this error i`ve used `CMD_ABORT` - it looks work well.

lili1218, 2020-10-06:
> I'm shooting in the dark here, but perhaps the radio should be reset more aggressively after the first operation that fails.
> For example, if `prop_div_radio_setup()` fails as in the example above, [...]

@atiselsts
if there is any other log/file I can provide to make this issue more clear, please do let me know.  thank you

lili1218, 2020-10-06:
> > if `prop_div_radio_setup()` fails as in the example above, then perhaps the RF core should be turned off.
> 
> i have meet some situations ([fix](https://github.com/alexrayne/contiki-ng/commit/5d1aead6b3d42b2002b8e58343f76948d8762d6d)) when code forget power off() core after unfinished operation (possibly timedout sent aor receive) , and starts power on(). There unfinished operation is still active, and so go into receive mode fails with same error.
> 
> to handle this error i've used `CMD_ABORT` - it looks work well.

@alexrayne thanks for your suggestion, I will try your fix here to see whether it can solve.

atiselsts, 2020-10-14:
@lili1218 did you test my suggestion? What do you think?

lili1218, 2020-10-15:
@atiselsts 
I tried with your suggestion. but the device still hang. according to the log, it seems before radio chip completely power down, there is another command to radio which cause hanging issue.  I am running one more test with other modification(until complete power down, no more radio activity allowed). will share detail log and changed file later on. 
also, if possible, can you share the file you change for 5kbps(smartrf-setting.c/ prop-mode.c or others) ? . if allowed, I want to try at my end. since right now, the device hang completely...

atiselsts, 2020-10-15:
It was for a proprietary project and we don't have the permission to share the code, as far as I know.

lili1218, 2020-10-19:
@atiselsts 
sure, thanks for reply, totally can understand

just run a test again. I just find out those hanging devices are all stop with similar pattern.

I will see the log, the device will do rf_core_power_down before it did PT_SPAWN(rx)(normally it should't happen)
once this happens, will cause device hangs (following is the log)

------------------------------------------------start of log ------------------------------------------------
2020-10-18 12:49:35.835869 non-active-slot {asn 00.0002bec1}
2020-10-18 12:49:36.830817 non-active-slot {asn 00.0002bec2}
2020-10-18 12:49:37.823664 prop-mode:off
2020-10-18 12:49:37.827215 rf_core_power_down: 37155
2020-10-18 12:49:37.834617 rf_core_power_down: rx-slot on {asn 00.0002bec3}
2020-10-18 12:49:37.835976 37155
2020-10-18 12:49:37.896995 prop-mode:off
2020-10-18 12:49:37.900376 rf_core_power_down: 37156
2020-10-18 12:49:37.906259 rf-core: fs_powerdown: CMDSTA=0x00000086
2020-10-18 12:49:37.915438 rf-core: rf_core_get_rat_rtc_offset: SYNC_STOP_RAT fail, ret 0 CMDSTA=0x00000086
2020-10-18 12:49:37.920085 rf_core_power_down: 37156
2020-10-18 12:49:37.922438 off rx no-data
2020-10-18 12:49:37.929061 rx-slot off {asn 00.0002bec3} link:2, curr_channel:11
device hangs here
------------------------------------------------end of log ------------------------------------------------

rx-slot on {asn xx.xxxxxxx}: this line is printed before doing PT_SPAWN(tx/rx) in tsch-slot-operation.c
and rf_core_power_down: is printed in the begin and end of rf_core_power_down()
,and the number after rf_core_power_down indicates how many time the rf_core_power_down function is being called

could you please help why it happens ? very appreciate for your help

[smartrf-settings.txt](https://github.com/contiki-ng/contiki-ng/files/5401066/smartrf-settings.txt)

[cc13xx-50kbps-tsch.txt](https://github.com/contiki-ng/contiki-ng/files/5401015/cc13xx-50kbps-tsch.txt)
[tsch-slot-operation.txt](https://github.com/contiki-ng/contiki-ng/files/5401031/tsch-slot-operation.txt)

alexrayne, 2020-10-19:
there is not enough debug - what happens after slot operation: it may hungs in `do - while` cycle
```
... while(!tsch_schedule_slot_operation(t, prev_slot_start, time_to_next_active_slot, "main"));
```
ar maybe your code hungs not in TSCH stack, but in some other code?

need marks, that rTimer enter/exit happens.

atiselsts, 2020-10-27:
I wanted to test this, but I cannot build your version. Some files are missing, for example rf_patch_cpe_sl_longrange.h, and you don't provide modifications done in others, for example rf-core.c.

Did you also try sending CMD_ABORT one more time when power down fails? Perhaps need to send abort again, wait for completion, and then repeat trying to power off.

Seeing that the TI has completely changed the approach to the low-level ops in the simplelink platform, maybe you can try it? It does some things differently, for example run CMD_STOP instead of CMD_ABORT before shutting down. I don't know about the energy efficiency this platform provides, though.

lili1218, 2020-11-05:
hi,
I did the test based on your advice that sending CMD_ABORT when power down fails, and monitoring networks for a week. seems it is a proper fix! thanks for you help!
As for a simplelink platform, I thought contiki-ng cannot support TSCH in sub-G Hz currently ? if it works, we would like to try out as well. please do correct me if I have misunderstanding. thanks~

atiselsts, 2020-11-05:
Simplelink TSCH support exists since a few months back. It does not work on all chips (CC1352P1 has problems) and there are some interoperability problems, but that's not relevant if you only need to use CC1310. On CC1310, Simplelink TSCH just needs more testing, there are no known problems at the moment.

closed 2020-12-07 by lili1218