contiki-ng/contiki-ng #1416: Personal Area Network (PAN) ID never assigned unless TSCH is selected as the channel access method.

Robert-Lasater, 2020-10-15:
Inspection of the source code shows `frame802154_set_pan_id` is only called from the file `os/net/mac/tsch/tsch.c`; it is never called from any programming in the corresponding folder for CSMA, `os/net/mac/csma`.

alexrayne, 2020-10-15:
there os/net/mac/framer/frame802154.h:IEEE802154_CONF_PANID - defines default PADID, current CSMA examples uses that, and in project-conf.h this may be overrided.

Robert-Lasater, 2020-10-15:
>there os/net/mac/framer/frame802154.h:IEEE802154_CONF_PANID- defines default PADID, current CSMA examples uses that, and in project-conf.h this may be overridden.

TSCH programming sets this PAN ID based on what it gets from its remote partner.  Look at 'tsch_associate` in `os/net/mac/tsch/tsch.c`.  Around line 714 is this call: `frame802154_set_pan_id(frame.src_pid);`

I am not an expert at all here.  Does this PAN ID only make sense with TSCH?  Because that's the only Channel Access Method that sets the PAN ID dynamically.

My understanding is the PAN ID helps prevent a 802.15.4 node from being hijacked by an unauthorized RPL Border Router.