contiki-ng/contiki-ng #1432: how cc13x0-cc26x0 platform decides end of receiving in prop mode

lili1218, 2020-11-05:
Hi team,
I am studying on TSCH in sub-G hz with cc13x0-cc26x0 platform. it seems that the prop-mode layer uses rssi_threshold to decide whether the receiving is done or not. Since I hear from my client that cc1310 chip has register to tell whether it receives complete data. just curious, if it has certain register why prop-mode use rssi_threshold to decide instead of reading register.
or my/my client's understanding is not correct ? thanks

atiselsts, 2020-11-05:
> it seems that the prop-mode layer uses rssi_threshold to decide whether the receiving is done or not. 

That is correct.

Maybe @g-oikonomou can comment why its done this way?

g-oikonomou, 2020-11-05:
To the best of my knowledge there is no such register (or if there is, please enlighten me).

There is an interrupt flag that is set when the chip sees the synch word (`MDMSOFT`). One could inspect that flag, but unfortunately the software logic required in order to clear the flag correctly is complex and we decided against using it. Among the problems is that the flag will stay set even if the radio rejects the frame. The flag will also stay set after frame reception. There are also some issues related with clearing the flag _after_ the synch word of the next incoming frame has started.

g-oikonomou, 2020-11-05:
Let me rephrase that cause i am not 100%sure I understand the question. We use RSSI to determine if the chip is currently receiving. After it has finished receiving we no longer need RSSI, we check the status of our RX buffers to see if there is a frame there ready to be read out.

lili1218, 2020-11-05:
> Let me rephrase that cause i am not 100%sure I understand the question. We use RSSI to determine if the chip is currently receiving. After it has finished receiving we no longer need RSSI, we check the status of our RX buffers to see if there is a frame there ready to be read out.

yes. that is my question, and the problem we are facing now is when environment noise is very strong, then the TSCH layer will wait until timeout while receiving. then there will be a dl-miss happen which will cause tx end missing an ack. that is my observation and assumption. I will check with my client to see what exactly he said about the register. if there exists such register, I will share with u.
thanks

atiselsts, 2020-11-05:
@lili1218 we are aware of this problem, it's absolutely an issue in sub-GHz TSCH networks. Sounded like you already had a fix in mind - with register approach, that's why I got very interested in this discussion.

If that approach does not work out, perhaps the TSCH core code can be modified. For instance, the lines 626 and 820 in tsch-slot-operation.c could be expanded to look at the `pending_packet()` status besides the current `receiving_packet()` check.

lili1218, 2020-11-06:
@atiselsts  
I will check with my client, if he can share some useful information regarding register I am very happy to share with u
and thanks for another approach you provide, I will test it locally. 
thanks

alexrayne, 2020-11-13:
Fot TSCH stack prop-mode driver from some time used another technic for receive detection - as [atiselsts](https://github.com/atiselsts) sayd. 
And [here](https://github.com/alexrayne/contiki-ng/commit/2be35400209023846e20115417535b18c34dc908)
i have improved it with monitoring current receive packet task status - it ends when radio reports receive status complete.

Maybe this aproach help you some?

atiselsts, 2020-11-13:
The whole approach looks sensible, how well tested it is? @alexrayne

alexrayne, 2020-11-13:
> how well tested it is?

i`ve worked with it couple of month this summer, right when develop PR #1384 - port for cc1352. And #1399 also has developed with it. 
examples, wich i used for tests - wokrs. But, unfortunaly, my develop have finished before and project looks in anabios, therefore it not tested in a fields. 
But, i'm happy with this.

lili1218, 2020-11-16:
> Fot TSCH stack prop-mode driver from some time used another technic for receive detection - as [atiselsts](https://github.com/atiselsts) sayd.
>
> And [...]

yes, I think this is very helpful. Since after checking with my customer, the way he suggests is very similar to your approach. I will test in office environment first, if it works properly here, I will try to test in a field as well. will report the result after finishing test for your information. it will takes few weeks just let you know in advance since we are pending with other issue. But we definitely will test it. thanks for your sharing

lili1218, 2020-12-21:
hello, @atiselsts @alexrayne 
I have tested alex's approach in both indoor and outdoor environment.
indoor with 40 nodes network and background noise is around -100, and total reliability is 97% with 5 mins sending 1 packet.

As for out door test, first it performs better then RSSI based approach.test environment at outdoor the background noise is around -70 to -80. at the same location, when using RSSI based approach(which threshold is -90 by default) I will see st 2 on data-sender side and dl-miss rxbeforeAck at data-receiver side  which is expected I would say. when I change both devices to alex's approach , i didnt see any issue described above. it can establish network without issue. and I run this outdoor test with total 17 node for 6 days until now, data reliability is 98%.

please let me know if there is further thing I can do for this. thanks

atiselsts, 2021-05-26:
Should be fixed now that #1515 is merged.

closed 2021-05-26 by atiselsts