contiki-ng/contiki-ng #1443: Extending mqtt-client to allow more than one active device and SensorTag environmental sensors

jsolderitsch, 2020-11-20:
Would there be community interest in modifications to the current mqtt-client example to support more than one active device and being able to distinguish them while they are running? Some sort of device ID is needed that can part of the publish and subscribe topic strings (and/or the mqtt message payload). Currently there is a client_id that is defined to support using IBM Quickstart as an mqtt broker. The hardware specific portion of this id could be used as the basis of this device ID.

Separately, SensorTags have environmental sensors like light, temperature and humidity. There is a notion of arch and platform specific extensions already in the example code but no examples of using these sensors.

I have working code that accomplishes both of these enhancements. More work remains to make sure that the older cc26x0-cc13x0 platform will work with the approach I have taken so far. Things work with the SimpleLink platform and even with the cc1352r1 based LPSTK SensorTag as a device on this platform. I can have two of these connected to a rpl-border-router and a local mosquitto broker can be used to show the devices are operating and reporting sensor values.

If there is interest, I can clean up my code and create a branch suitable for use as a Pull Request.

g-oikonomou, 2020-11-21:
> Would there be community interest in modifications to the current mqtt-client example to support more than one active device and being able to distinguish them while they are running? Some sort of device ID is needed that can be part of the publish and subscribe topic strings (and/or the mqtt message payload). Currently there is a client_id that is defined to support using IBM Quickstart as an mqtt broker. The hardware specific portion of this id could be used as the basis of this device ID.

I do not understand. Is this `client_id` not adequate to meet the requirement of distinguishing among devices?

> Separately, SensorTags have environmental sensors like light, temperature and humidity. There is a notion of arch and platform specific extensions already in the example code but no examples of using these sensors.

This is definitely a desirable addition. I've not made up my mind about a clean way to add such extensions to the build. I would definitely want to keep the sensortag `mqtt_client_extension_t` variables and respective code in their own file. Unfortunately there is only a single `MQTT_CLIENT_EXTENSIONS` to pull them all in, so I think there would have to be some pre-processor spaghetti in the `mqtt-client-extensions.c`

g-oikonomou, 2020-11-21:
@jsolderitsch James, I have created a branch [contrib/mqtt-client-stk-ext](https://github.com/g-oikonomou/contiki-ng/tree/contrib/mqtt-client-stk-ext) with a template of how I'd imagine the modifications to the build system and also how I'd imagine new sensors being hooked in. I have added also a stub for a sensortag-specific sensors code file.

g-oikonomou, 2020-11-21:
We can also consider moving `mqtt-client-extensions.c` to `arch/platform/cc26x0-cc13x0/sensortag/`. Which would mean some duplication, since we'd need one for launchpads, one for sensortag, one for srf. But that file simply exposes a symbol so that's not a big deal, and the respective files would be a lot cleaner.

The `sensortag-ext` files could be moved to a `common` dir under `arch/platform`, allowing both simplelink and cc26x0-cc13x0 to pull them in without changes. I bet we'd need some pre-processor work inside in order to get them to work, not unlike the work in the base demo PR #1290

g-oikonomou, 2020-11-21:
... which I have now done (and will now stop the spam).

There is now a `arch/platform/cc26x0-cc13x0/common/mqtt-client-extensions.c`. This only uses the two on-chip sensors and is a copy of the original file of the same name. All launchpads and SmarfRF boards pull this one in.

There is a separate, sensortag-specific `arch/platform/cc26x0-cc13x0/sensortag/mqtt-client-extensions.c`, only used when building for sensortags.

jsolderitsch, 2020-11-21:
Thanks for the encouragement to work on this. I will checkout your branch and fit my current changes into it and we can finalize the configuration together I hope. I do want Simplelink as a fully supported platform because that's the one that supports the newer tag based on the cc1352r1.

To explain my concern about the device ID, the client ID as currently computed is something like:
```
d:contiki-ng:mqtt-client:00124bbc3f5d
```
The last part is what I propose using as the client device id
```
00124bbc3f5d
```
Then the evt topic string would be transformed into something like:
```
iot-2/id/00124bbc3f5d/evt/status/fmt/json
```
and the cmd topic string could be:
```
iot-2/id/00124bbc3f5d/cmd/leds/fmt/json
```
The code I have working implements this approach.

g-oikonomou, 2020-11-22:
OK so you want a unique device identifier as part of the topic itself. I am not so sure about that. Do you want your subscribers to be able to receive publish messages from specific devices?

jsolderitsch, 2020-11-22:
Yes, I expect devices to publish, and have subscribers subscribe to, device specific events. I want  logic (say running in the cloud) to be able to direct commands to specific devices as well. I think the device ID being in the topic string makes more sense than having this in the mqtt payload.

I am curious as to why this is off-putting to you.

g-oikonomou, 2020-11-23:
> I am curious as to why this is off-putting to you.

From wikipedia (I know...):

> In software architecture, publish–subscribe is a messaging pattern where senders of messages, called publishers, do not program the messages to be sent directly to specific receivers, called subscribers, but instead categorize published messages into classes without knowledge of which subscribers, if any, there may be. Similarly, subscribers express interest in one or more classes and only receive messages that are of interest, without knowledge of which publishers, if any, there are. 

So the idea of pub/sub is, you are saying "something" but you don't know who you're saying it to. Similarly, you are interested in "a topic", but not interested in who's talking about it.

So what you are proposing could make sense in some use-cases (certainly in yours!), and it is certainly technically feasible, but I would not be too keen in making the default approach in our generic MQTT example :)

jsolderitsch, 2020-11-23:
I guess there is some room for disagreement here. Maybe it is more natural to rely on the identity of an emitter to not be in the topic after all but in the payload. That was the way I first started to implement it but then had a change of mind on it. 

Things have identity. The IBM Cloud and many of the examples I have seen put identity information in the topic string. If you want all of them to be communicated to you, the '+' wildcard allows you to subscribe to information from all of them. But you can focus on one if you wish.

If you want to exert control over specific devices, then the ID of the device needs to be communicated. It was an easy extension to mqtt-client's current code to have the cmd topic string to also have identity in the string. That way a device can subscribe ONLY to commands intended for it, rather than for commands for devices of its type. When I want to turn a tag's LED, I want to do that to a specific tag, not all tags currently connected on the wireless network.

Perhaps when we work out what contiki-ng accepts as an extension, I will have to reserve some functionality to a private version of mqtt-client.c.

mqtt in the IoT world is not quite a general implementation of pub/sub but a means for devices to emit and react to information, IMHO.

I did imagine my proposal would be quite so controversial!

g-oikonomou, 2020-11-23:
Not controversial at all! Not in the slightest, but I have to say that we will probably need to agree to disagree here :)

If I want to control a specific device, I will want to control the device "in the kitchen", not the device "with ID == ABCDEFG". So I would much rather my topic structure be along the lines of:

`home/kitchen/humidity`

instead of

`home/00124bbc3f5d/humidity`.

In doing so, I can manipulate the "sensortag in the kitchen" instead of the sensortag with ID 00124bbc3f5d. I can then move the 00124bbc3f5d sensortag to the bathroom, and replace it with a new sensortag in the kitchen. Publishing a "turn off the LED of the sensortag in the kitchen" achieves the same physical effect (it toggles a LED in the kitchen, not in the batrhoom). So despite the change of hardware location in the physical world, publishing to the same topic achieves the same physical outcome.

Having said all that, I hope you will agree with me that in a pub/sub-based design, the exact structure of the topic hierarchy is specific to the use-case being developed. Our expectation would be for every user of the mqtt client example to need to somehow edit the structure of the topic hierarchy, as opposed to using it as it is.

Support for sensortag and simplelink/LPSTK sensor very very much desirable though :)

Last but not least: Many thanks for your interest in the project and your desire to contribute back! 😄

jsolderitsch, 2020-12-02:
> ... which I have now done (and will now stop the spam).
>
> There is now a `arch/platform/cc26x0-cc13x0/common/mqtt-client-extensions.c`. This only uses the two on-chip sensors and is a copy of the [...]

I kind of understand what is being proposed at the platform level, but what about the arch/cpu/ level?

In the current code arrangement there are folders for cc26x0-cc13x0 and cc2538. Each of these then has a single pair of builtin-sensors.h and .c files where the real logic is contained. Does the subfolder hierarchy there get all complicated like we see on the platform side so there are builtin-sensortag-sensors files for the tag logic and builtin-launchpad-sensors files for the launchpad specific logic? Or are we trying to get device specific in the platform folder and we eliminate the cpu stuff for the most part. 

To me this arrangement looks very spaghetti-like and we have spaghetti file arrangements at both the platform level and the cpu level.

But maybe I am not seeing the point here.