contiki-ng/contiki-ng #1444: Build problems for openmote-cc2538 (inconsistent Python version)

guillaumenibert, 2020-11-21:
Hello,

I tried to build `hello-world.c` (located in `examples/hello-world`) for Openmote CC2538 Rev. A1.

Step to reproduce (assuming that the working directory is `examples/hello-world`) :
```bash
make TARGET=openmote BOARD=openmote-cc2538 BOARD_REVISION=REV_A1 savetarget
make hello-world.upload TARGET=openmote BOARD_REVISION=REV_A1
```

Output :

```python
  [...]
  CC        ../../arch/cpu/cc2538/./ieee-addr.c
  CC        ../../arch/cpu/cc2538/cc2538.lds
  CC        ../../arch/cpu/cc2538/./startup-gcc.c
  CC        hello-world.c
  LD        build/openmote/openmote-cc2538/hello-world.elf
  OBJCOPY   build/openmote/openmote-cc2538/hello-world.elf --> build/openmote/openmote-cc2538/hello-world.bin
python ../../tools/cc2538-bsl/cc2538-bsl.py --bootloader-invert-lines -e -w -v -b 450000 -p /dev/ttyUSB0 -a 0x00202000 build/openmote/openmote-cc2538/hello-world.bin
  File "../../tools/cc2538-bsl/cc2538-bsl.py", line 82
    print(message, end=attr, file=sys.stderr)
                      ^
SyntaxError: invalid syntax
../../arch/platform/openmote/Makefile.openmote:44: recipe for target 'hello-world.upload' failed
make: *** [hello-world.upload] Error 1
rm hello-world.o build/openmote/openmote-cc2538/obj/startup-gcc.o
```

I went to see the [JelterT cc2538-bsl GitHub repository](https://github.com/JelmerT/cc2538-bsl/commits/master) and I found that `tools/cc2538-bsl/cc2538-bsl.py` now only supports **Python 3** according to recent commits.
However it seems that `contiki-ng/arch/platform/openmote/Makefile.openmote` uses **Python 2** (line 31) :
```
PYTHON = python
```

I changed the above line to
```
PYTHON = python3
```

and installed pyserial with pip3 and now it works.

Is this the best way to alleviate the problem?

Thank you in advance.

Best regards,

Guillaume Nibert

jeppenodgaard, 2020-11-23:
> Is this the best way to alleviate the problem?

I think so.

I made a symlink to python3, but I think this is a proper solution.

g-oikonomou, 2020-11-23:
Yes I agree. If the script is python3 only, we should adapt our build system to that effect. I bet it is more than just openmote.

```
$ git grep PYTHON
arch/cpu/cc26x0-cc13x0/Makefile.cc26x0-cc13x0:PYTHON = python
arch/cpu/cc26x0-cc13x0/Makefile.cc26x0-cc13x0:  $(PYTHON) $(BSL) $(BSL_FLAGS) $<
arch/cpu/simplelink-cc13xx-cc26xx/Makefile.cc13xx-cc26xx:PYTHON = python
arch/cpu/simplelink-cc13xx-cc26xx/Makefile.cc13xx-cc26xx:       $(PYTHON) $(BSL) $(BSL_FLAGS) $<
arch/platform/cc2538dk/Makefile.cc2538dk:PYTHON = python
arch/platform/cc2538dk/Makefile.cc2538dk:       $(PYTHON) $(BSL) $(BSL_FLAGS) $(BSL_ADDRESS_ARG) $<
arch/platform/openmote/Makefile.openmote:PYTHON = python
arch/platform/openmote/Makefile.openmote:       $(PYTHON) $(BSL) $(BSL_FLAGS) $(BSL_ADDRESS_ARG) $<
arch/platform/zoul/Makefile.zoul:PYTHON = python
arch/platform/zoul/Makefile.zoul:       $(PYTHON) $(BSL) $(BSL_FLAGS) -b $(BSL_SPEED) -a $$$${BSL_ADDRESS} -p $(MOTE) $$<
examples/sensniff/Makefile:PYTHON ?= python
examples/sensniff/Makefile:     $(PYTHON) $(SENSNIFF) $(SENSNIFF_FLAGS)
```
We ought to deal with this before the upcoming release. I am marking this as a bug.

PR #1448 (2020-11-26): BSL: Change PYTHON value to python3
Should fix #1444
closed 2020-11-26 by atiselsts