contiki-ng/contiki-ng #1449: Execution problems with old compilers on nrf52840-dongle

warner83, 2020-11-24:
Hello, 
I had some issue in executing two examples (specifically, rpl-border-router and example-coap-server) on two nrf52840 dongles compiled on Ubuntu 16.04. The distro ships with an old compiler arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3.
In particular the code run, however, it seemed to have issues in writing into the NRF_RADIO structure. For instance the channel on the example-coap-server was not properly set. 

Upgrading to a newer compiler,  arm-none-eabi-gcc 5.3.1 20160307 in my case, solved the issue.

atiselsts, 2020-11-24:
We may need to update `Makefile.gcc` to warn the user when such an old version is used.

g-oikonomou, 2020-12-02:
#1462 for the warning.

Leaving this open and moving this to milestone 4.7 for improved handling if we can.

alexstanoev, 2020-12-15:
The root cause was fixed by #1478.

Do we want to revise the warning from #1462?

PR #1462 (2020-12-02): Warn when using old GCC versions with platform nRF52840
Older GCC versions generate broken executables for the nRF52840 platform. 
The resulting binaries appear functional but are unable to send or receive any packets.

#1449 has seen this happen with arm-none-eabi-gcc 4.9.3, and I can reproduce it with arm-none-eabi-gcc 5.2.1.
This PR warns anyone using a version of gcc older than 6.x with the nRF52840 platform and prompts them to update their toolchain.

Closes #1449.
PR #1503 (2021-02-10): Improve gcc version detection and revert obsolete make warning
The root cause for the compiler version warning added in #1462 was fixed by #1478 so this can be reverted safely.
This also closes #1449.

This PR also improves the gcc version detection for warnings as the current approach doesn't handle double digit versions correctly and misinterprets gcc 10 as 1.
closed 2021-02-10 by g-oikonomou