contiki-ng/contiki-ng #1453: Migration to Python 3

atiselsts, 2020-11-26:
Python 2 is end of life and stopped being officially supported in Jan 2020. pip support for Python 2 will be dropped in Jan 2021. This means we should migrate all Python codebase in Contiki-NG to Python3.

Here are potential Python2 dependencies in Contiki-NG at the moment.

1. Scripts that are called by the default Python interpreter at the moment, but probably can be called by Python3 without any code changes:
- [ ] <strike>BSL scripts for ARM boards</strike> done in #1448
- [ ] `motelist`
- [ ] `run-cooja.py` script in `result-visualization` example
- [ ] `parse.py` script in `rpl-req-resp` example
- [ ] `sensniff` (it looks like it may already support Python3 despite the comment in the source code!)

2. Dependencies:
- [ ] Dockerfile (includes pip module installation)
- [ ] `tools/vagrant/bootstrap.sh` (includes pip module installation)

3. Scripts that require actual porting effort (only legacy platform stuff here):
- [ ] BSL for Z1
- [ ] BSL for Sky
- [ ] `tools/sky/serial/` (serial dump scripts for Sky and Z1)
- [ ] `tools/jn516x/mote-list.py` (NXP board)

g-oikonomou, 2020-11-26:
> Scripts that require actual porting effort (only legacy platform stuff here):

What do we want to do with that lot? I am personally not particularly keen on spending effort on those.

g-oikonomou, 2020-12-01:
motelist: #1465

atiselsts, 2021-04-26:
Dockerfile and benchmark scripts: https://github.com/contiki-ng/contiki-ng/pull/1552

atiselsts, 2021-05-05:
Explicitly use Python 2 as discussed: #1558

closed 2022-06-29 by joakimeriksson