contiki-ng/contiki-ng #1455: TSCH_SCHEDULE_AND_YIELD() sometime wait not enough time

nbrunner, 2020-11-26:
In [TSCH_SCHEDULE_AND_YIELD](https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/mac/tsch/tsch-slot-operation.c#L330), when tsch_schedule_slot_operation() return 0 the busy wait until `offset - RTIMER_GUARD` instead of `offset`.
If `RTIMER_BUSYWAIT_UNTIL_ABS(0, ref_time, offset);` is put outside the if, i think it should resolve this problem.

atiselsts, 2020-11-26:
Thanks for the report.

It does look like TSCH_SCHEDULE_AND_YIELD has a problem, and your proposed approach sounds correct.

Additionally, I think the `check_timer_miss()` in the  `tsch_schedule_slot_operation()` can be optimized, as there is not need to subtracts the guard time from the `offset` twice.

@simonduq could you check this?

alexrayne, 2020-11-26:
IMHO, beter hide into `RTIMER_GUARD` into `tsch_schedule_slot_operation` :
1) guard cant be forgotten, so `tsch_schedule_slot_operation` should always use it to prevent rtimer falure. As i remember, on some platform rtimer not tolerant vs miss target time, it just hungs awaiting until RTC come override.
2) in such case `tsch_schedule_slot_operation` can be used directly, without `TSCH_SCHEDULE_AND_YIELD`

Here is old [solution](https://github.com/alexrayne/contiki/blob/cc946357d976bed99b4481cac9f52044416b108c/os/net/mac/tsch/tsch-slot-operation.c#L344), that i use

atiselsts, 2020-11-27:
@alexrayne I agree, the guard should be present in `tsch_schedule_slot_operation`. I was thinking more about removing it from the `TSCH_SCHEDULE_AND_YIELD` macro.

However, doing something like that at this point (after 5+ years of working code) is very risky despite the code being imperfect, as any changes may affect TSCH timing and cause additional slight time sync errors to appear.

Let's play it safe and just do what @nbrunner is proposing. See #1456

alexrayne, 2020-11-27:
> However, doing something like that at this point (after 5+ years of working code) is very risky despite the code being imperfect, as any changes may affect TSCH timing and cause additional slight time sync errors to appear.

Code, i proposed - is from my fork, that was merged with contiki-ng - other words, it is contiki-ng with legacy history. It is my current work version - it was tested this summer on cc1350/52 and cooja platforms. 
It is fresh code.