contiki-ng/contiki-ng #1457: cc1310 guard time setting

lili1218, 2020-11-27:
hello,
Since with our testing with cloned contiki-ng version 4.5, I just found out sometimes a pair of node cannot connect(gw failed to receive KA right after node receiving EB and sending out KA). I go through the code, I am a little bit confused about our guard time which is TSCH_CONF_RX_WAIT (2200us). 
the receiver side, it wake up to listen at tsch_timing[tsch_ts_rx_offset] - RADIO_DELAY_BEFORE_RX
which is equal to tsch_timing[tsch_ts_rx_offset]- 0.015ms
also is equal to tsch_timing[(CC13XX_TSCH_DEFAULT_TS_TX_OFFSET - (TSCH_CONF_RX_WAIT / 2)] - 0.015ms

our sender side start to send packet(from preable) at  tsch_timing[tsch_ts_tx_offset] - RADIO_DELAY_BEFORE_TX
which is equal to tsch_timing[tsch_ts_tx_offset] - 0.8ms

so my question is from the code, it looks like guard time 1.1 ms didnt consider time taken by preamble and sync word. therefore, the actual guard time is less then 1.1ms.

can someone help me on this. thanks

atiselsts, 2020-11-27:
TSCH does not take into account the preamble when computing the guard time. The standard does not say anything about that, and on some platforms the preamble duration is configurable, complicating the task.

See also issue #565.

However, on CC26x0-CC13x0 platform the situation is a bit different. The radio is turned on at the very start of the slot in this platform. If it was deep-sleeping, it takes around 1 ms for that turning-on process to be completed. The listening starts as soon as the radio is fully on. The calls to `tsch_radio_on(TSCH_RADIO_CMD_ON_WITHIN_TIMESLOT)` and `tsch_radio_off(TSCH_RADIO_CMD_OFF_WITHIN_TIMESLOT)` effectively does nothing on this platfrom.

lili1218, 2020-11-27:
understand, but can I said with default config setting since it mention `CC13XX_RADIO_PHY_HEADER_LEN 6`
can I said in the current code, it will shrink the guard time. instead of 1.1ms, it will be 1.1ms - timing on sending preamble and sync word.
would you suggest I to change setting to 
`#define CC13XX_TSCH_DEFAULT_TS_RX_OFFSET (CC13XX_TSCH_DEFAULT_TS_TX_OFFSET - RADIO_DELAY_BEFORE_TX- (TSCH_CONF_RX_WAIT / 2))` ? 

since I have seen node sometimes cannot communicate, it looks more like timing incorrect due to guard time is eaten by preamble and sync word.

thanks

atiselsts, 2020-11-27:
I suggest simply increasing `TSCH_CONF_RX_WAIT` value, and checking whether you also might need to move the Tx offset forward (increase `CC13XX_TSCH_DEFAULT_TS_TX_OFFSET`).

Since the Rx starts immediately after the start of the timeslot, as soon as the radio has turned on, reducing the Rx offset will not make any difference. At least in theory it should not.

Did you get any chance to check the things discussed in #1432 ?

lili1218, 2020-11-27:
Got it. Thanks for your suggestion.
Since we are stilling pending on other issue, after 15 Dec (end of current
project) I will be able to deploy to the field and try! Will definitelt
report my test result!

alexrayne, 2020-11-27:
Hallow, Lili!
i got stucked about long preabmble that eating guardtime, and try discuss it [here](https://github.com/contiki-ng/contiki-ng/issues/1330)

This was lead to changes,  most of them [here](https://github.com/alexrayne/contiki-ng/commit/c60ec712a203a31c2f7223dc48d666a344b78951) - it provides different strategies for preample waiting, when guard time enough, or not - for detect preamble.

I solved my problem there, you can test it, and i m interesting - is it good or bug?

lili1218, 2020-12-01:
hello Alexrayne,
sorry since I dont have fully understand of what you change currently and we only have short time to test, instead of trying your solution, I simply increase guard time to solve this issue right now. It seems helpful. I will try to understand your change later~ thanks for your sharing!

lili1218, 2020-12-01:
> TSCH does not take into account the preamble when computing the guard time. The standard does not say anything about that, and on some platforms the preamble duration is configurable, complicating [...]

@atiselsts 
since with cloned code, I just found one the radio is not turned-on at the start of slot from tsch-conf.h

> #ifdef TSCH_CONF_RADIO_ON_DURING_TIMESLOT
#define TSCH_RADIO_ON_DURING_TIMESLOT TSCH_CONF_RADIO_ON_DURING_TIMESLOT
#else
#define TSCH_RADIO_ON_DURING_TIMESLOT 0
#endif

as you mentioned previously, it takes around 1ms for radio to on.
But in the code, after enabling CCA, the tx slot, radio it on after cca_offset

> #if TSCH_CCA_ENABLED
        cca_status = 1;
        /* delay before CCA */
        TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, tsch_timing[tsch_ts_cca_offset], "cca");
        TSCH_DEBUG_TX_EVENT();
        tsch_radio_on(TSCH_RADIO_CMD_ON_WITHIN_TIMESLOT);

so can I said that in this setting, actually, not really all the node performs at most 128us listening?

same as rx slot, radio is on after rx_offset.

>     /* Wait before starting to listen */
    TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, tsch_timing[tsch_ts_rx_offset] - RADIO_DELAY_BEFORE_RX, "RxBeforeListen");
    TSCH_DEBUG_RX_EVENT();
    /* Start radio for at least guard time */
    tsch_radio_on(TSCH_RADIO_CMD_ON_WITHIN_TIMESLOT);
    packet_seen = NETSTACK_RADIO.receiving_packet() || NETSTACK_RADIO.pending_packet();
    if(!packet_seen) {
      /* Check if receiving within guard time */
      RTIMER_BUSYWAIT_UNTIL_ABS((packet_seen = NETSTACK_RADIO.receiving_packet()),
          current_slot_start, tsch_timing[tsch_ts_rx_offset] + tsch_timing[tsch_ts_rx_wait] + RADIO_DELAY_BEFORE_DETECT);
    }

therefore, in rx slot, the actually listen time might be smaller than tsch_ts_rx_wait since radio takes 1 ms to turn on sometimes.

can I said it will perform better if TSCH_RADIO_ON_DURING_TIMESLOT flag is enable? since it wont have radio-1-ms-on potential issue which might eat guard time as well.

atiselsts, 2020-12-01:
For CC1310, TSCH_RADIO_ON_DURING_TIMESLOT is enabled by default at:
https://github.com/contiki-ng/contiki-ng/blob/develop/arch/cpu/cc26x0-cc13x0/cc13xx-cc26xx-def.h#L91

Are you using CCA with TSCH? The CCA feature is not tested a lot, I personally do not know anyone who is using it.

lili1218, 2020-12-01:
@atiselsts 
sorry  my fault. didnt check it properly. thanks!
for the CCA, i enable it.However with default timing, it will have TxBeforeTx warning log since overlapping timing in CCA and transmitting preamble. therefore, I fine tuning timing to avoid it.

lili1218, 2020-12-01:
after changing timing as below

> #define CC13XX_TSCH_DEFAULT_TS_CCA_OFFSET         3300
#define CC13XX_TSCH_DEFAULT_TS_CCA                128
#define CC13XX_TSCH_DEFAULT_TS_TX_OFFSET          4500
#define CC13XX_TSCH_DEFAULT_TS_RX_OFFSET          (CC13XX_TSCH_DEFAULT_TS_TX_OFFSET - (TSCH_CONF_RX_WAIT / 2))
#define CC13XX_TSCH_DEFAULT_TS_RX_ACK_DELAY       2000
#define CC13XX_TSCH_DEFAULT_TS_TX_ACK_DELAY       3000
#define CC13XX_TSCH_DEFAULT_TS_RX_WAIT            TSCH_CONF_RX_WAIT
#define CC13XX_TSCH_DEFAULT_TS_ACK_WAIT           3000
#define CC13XX_TSCH_DEFAULT_TS_RX_TX              192
#define CC13XX_TSCH_DEFAULT_TS_MAX_ACK            10000
#define CC13XX_TSCH_DEFAULT_TS_MAX_TX             21600

/* Timeslot length: 40000 usec */
#define CC13XX_TSCH_DEFAULT_TS_TIMESLOT_LENGTH  45000

and put TSCH_CONF_RX_WAIT

I found out it will has error

> [INFO: TSCH-LOG  ] {asn 00.00000013 link  1  17   0  2 157 ch 15} !dl-miss RxBeforeListen 94 96
[INFO: TSCH-LOG  ] {asn 00.0000001f link  2  31   0  0  1 ch 15} !dl-miss RxBeforeListen 96 96
[INFO: TSCH-LOG  ] {asn 00.00000024 link  1  17   0  2 157 ch 25} !dl-miss RxBeforeListen 94 96
[INFO: TSCH-LOG  ] {asn 00.00000035 link  1  17   0  2 157 ch 26} !dl-miss RxBeforeListen 94 96
[INFO: TSCH-LOG  ] {asn 00.0000003e link  2  31   0  0  1 ch 20} !dl-miss RxBeforeListen 96 96
[INFO: TSCH-LOG  ] {asn 00.00000046 link  1  17   0  2 157 ch 20} !dl-miss RxBeforeListen 94 96

even I put slot length larger and disable CCA, it still wont help.
I cannot explain it. can someone point out what is the error with this re-write timing design and give me some suggestions? thanks

atiselsts, 2020-12-01:
The time between slot start and Rx start is too short (by around 2 ticks or 30 microseconds), the radio does not have enough time to turn on. Your TSCH_CONF_RX_WAIT guard time value is huge, try decreasing it by around 50 usec (30 usec + a safety margin), then it should be fine.

I doubt that you need that large guard time. You should do some experiments and look at the packet reception rates when different guard time values are used.

lili1218, 2020-12-01:
> The time between slot start and Rx start is too short (by around 2 ticks or 30 microseconds), the radio does not have enough time to turn on. Your TSCH_CONF_RX_WAIT guard time value is huge, try [...]

since we are suffering poor quality for our external lock( we use our device to test instead of launch pad, therefore I need to increase guard time with such huge value)
thanks for your feedback, but could you please explain a little bit detail? I cannot understand why it makes slot start and rx start very close..

atiselsts, 2020-12-01:
Start of a TSCH slot is like this:

```
                   | <-- guard time -->|
          | ------ | xxxxxxxx|xxxxxxxx | --------
          ^        ^         ^         ^
   slot start      ^         ^         rx end (= rx start + guard time)
               rx start      ^
                          tx offset
```
The more you increase guard time, the closer to the slot start rx start takes place.

alexrayne, 2020-12-01:
> I don't see it will change the interval between slot start and rx slot start. thanks for your help

in my logicanalisator scopings - it takes about 1-1.5ms CPU time to prepare for frame start. So looks that you is very close to this limit.
imho, this startup time takes about 700-800us to startrup RFcore, and other time - to inspect TSCH queue for packets

lili1218, 2020-12-02:
thanks for both of  your sharing

closed 2020-12-07 by lili1218