contiki-ng/contiki-ng #1458: Unused variable in `tests/07-simulation-base/code-ipv6/sender/udp-sender.c` with the latest Docker image cause a build failure

guillaumenibert, 2020-11-29:
Hello,

I tried to build `udp-sender.c` (located in `tests/07-simulation-base/code-ipv6/sender/udp-sender.c`) for Openmote CC2538 Rev. A1.
I use the latest version of Contiki-NG with Docker.

Step to reproduce (assuming that the working directory is `tests/07-simulation-base/code-ipv6/sender`) :
```bash
make TARGET=openmote BOARD=openmote-cc2538 BOARD_REVISION=REV_A1 savetarget
make udp-sender.upload TARGET=openmote BOARD_REVISION=REV_A1
```

Output :

```c
  [...]
  CC        ../../../../arch/cpu/cc2538/./startup-gcc.c
  CC        udp-sender.c
udp-sender.c: In function 'process_thread_udp_process':
udp-sender.c:37:24: error: unused variable 'send_timer' [-Werror=unused-variable]
   static struct etimer send_timer;
                        ^
udp-sender.c: At top level:
cc1: error: unrecognized command line option '-Wno-unused-constant-variable' [-Werror]
cc1: all warnings being treated as errors
../../../../Makefile.include:384: recipe for target 'udp-sender.o' failed
make: *** [udp-sender.o] Error 1
```

It seems that removing the variable `static struct etimer send_timer;` (line 37 from `udp-sender.c`) solves the problem.

Best regards,

Guillaume Nibert

guillaumenibert, 2020-11-29:
Here is the diff :
```diff
diff --git a/tests/07-simulation-base/code-ipv6/sender/udp-sender.c b/tests/07-simulation-base/code-ipv6/sender/udp-sender.c
index b2d4dde6a..10fbea28d 100644
--- a/tests/07-simulation-base/code-ipv6/sender/udp-sender.c
+++ b/tests/07-simulation-base/code-ipv6/sender/udp-sender.c
@@ -34,7 +34,6 @@ receiver(struct simple_udp_connection *c,
 PROCESS_THREAD(udp_process, ev, data)
 {
   static struct etimer periodic_timer;
-  static struct etimer send_timer;
   uip_ipaddr_t addr;
 
   PROCESS_BEGIN();
```

atiselsts, 2020-12-04:
@guillaumenibert would you mind preparing a PR with your proposed change? We could merge it straight away.

guillaumenibert, 2020-12-04:
I also removed this variable from `sender/unicast-sender.c` and  `receiver/udp-receiver.c` because of the same build failure.

alexstanoev, 2020-12-09:
Can this be closed now?

closed 2020-12-09 by guillaumenibert