contiki-ng/contiki-ng #1484: Cann't open cooja of X11 from a container made by contiki-ng

onnonn, 2021-01-01:
I use the paraments 
`docker run --privileged --sysctl net.ipv6.conf.all.disable_ipv6=0 --mount type=bind,source=/Users/onn/contikidocker/contiki-ng,destination=/home/user/contiki-ng -e DISPLAY=host.docker.internal:0 --net=host -v /tmp/.X11-unix:/tmp/.X11-unix -v /dev/bus/usb:/dev/bus/usb -ti contiker/contiki-ng`
and install xquartz and socat which are written in the wiki .
but when I reboot the $DISPLAY of my osx 10.15.7 is empty.
[image]

and then when I run `ant run`
the container shows that 
[image]

the socat shell show that
[image]

could u tell me which step is wrong in my shell code?I will appreciate it if u can give me a tip. Looking forward to your reply.

yatch, 2021-01-04:
Could you try this?
1. Start XQuartz
2. Open "Terminal" from "Applications" on the menu bar (`xterm` will be opened)
3. Run `socat` with the options in the terminal
4. Run `docker run` with the options in a different terminal

yatch, 2021-01-06:
@onnonn I'm going to close this issue. Feel free to reopen if the issue persists.

onnonn, 2021-01-06:
> Could you try this?
> 
> 1. Start XQuartz
> 2. Open "Terminal" from "Applications" on the menu bar (`xterm` will be opened)
> 3. Run `socat` with the options in the terminal
> 4. Run `docker run` with the options in a different terminal

when I open XQuartz firstly, then open socat has something wrong like this (maybe the port was used by x11)

[image]

besides that,I have four questions.
1.why the environment variable of macos $DISPLAY is empty after all settings have been done after installing XQuartz?
2.If the $DISPLAY of macos should be set manually, what should be set? The ip of mac or that of the docker container?
 if use the ip of the container , should the docker command value <docker run --net=hosts > be added? Is it possible that docker macos desktop 's UNIX socket is embedded in the middle of the network layer by the virtual sockets file, which leads to this problem?

if you can provide an official setup video tutorial , we will appreciate it greatly！

onnonn, 2021-01-06:
there is only one success when I try this
 https://superuser.com/questions/1535440/socat-e-connect5-len-4-af-1-0-4-no-such-file-or-directory-when-trying
but when I reboot my mac ，it can never been repeated

yatch, 2021-01-06:
$DISPLAY should be automatically set. I have no idea why this is not the case on your environment... Better to reinstall XQuartz??

I think, you enable "Allow connections from network clients" in "Security" tab of X11 Preference, which opens by clicking "Preferences..." menu of XQuartz. That's why port 6000 is occupied by X11.bin. Try the steps described above after disabling the setting.

Good luck.

yatch, 2021-01-13:
@onnonn  any update?

yatch, 2021-02-02:
If you're using Big Sur, https://www.xquartz.org/releases/XQuartz-2.8.0_beta3.html may be worth a try. Close this for now since it's been inactive.

closed 2021-02-02 by yatch