contiki-ng/contiki-ng #1485: Telnet Support

fkerem, 2021-01-05:
Hi,

In Contiki (legacy), there was support for telnet.
Does contiki-ng support telnet?
I couldn't find an example for it.

Thanks.

yatch, 2021-01-06:
> My nodes are running arm-based embedded linux.

Although I'm not sure what you're trying to do, Contiki-NG doesn't have the telnet implementation, which has been removed by https://github.com/contiki-ng/contiki-ng/commit/6220aea14acbf2e21e9dbbd63da3a46f6025349a .

closed 2021-01-06 by yatch

fkerem, 2021-01-06:
I'm trying to connect to an embedded linux node from an outer network through telnet.
I set up the border router to reach the node and am able to ping the node but I cannot connect it through telnet even though it listens.
Is this expected? Can't we connect to a telnet using the IPv6 address which was provided by the border router?

yatch, 2021-01-06:
> I'm trying to connect to an embedded linux node from an outer network through telnet.

Is Contiki-NG is running on the embedded linux node directly? If Contiki-NG is running a radio board which connects to the embedded linux via serial or something, you need to route packets between them somehow.

You should be able to establish a TCP connection even with an IPv6 address assigned through RPL if everything is properly configured. Please describe the settings in details.

fkerem, 2021-01-06:
Yes, this can be the case. I'm using ARM Cortex-A8 (https://www.iot-lab.info/docs/boards/iot-lab-a8-m3/).
It has a co-microcontroller on the same board (M3 is the co-microcontroller).
It seems that I'm flashing the firmware on M3 co-microcontroller but Telnet is running on A8 board.

If the packets would not be routed properly, I wouldn't be able to ping the node I believe.
I'm not very familiar with the way this board works. Do you have any suggestion? @yatch 
Thank you so much for your support!

yatch, 2021-01-06:
> It seems that I'm flashing the firmware on M3 co-microcontroller but Telnet is running on A8 board.

These two are different hosts from the viewpoint of the network. 

> If the packets were not routed properly, I wouldn't be able to ping the node I believe.

No. You ping to the M3 board, not to the A8 board that doesn't have the IPv6 address you specifies with the ping command.

> I'm not very familiar with the way this board works. Do you have any suggestions? 

Technically, what you're trying is possible, but it's not easy. There is no off-the-shelf solution. I'd suggest revisiting what you're really want to do with the boards, and what would be the best way to realize.

fkerem, 2021-01-06:
Thank you so much, I really appreciate your help @yatch .
Let's see, what I can do... :)

fkerem, 2021-01-12:
Dear @yatch, these two microcontrollers have a serial link. 
I would like to forward the packets from one microcontroller to another for port 23 (Telnet) using this serial link. 
Can I do this by configuring or updating tunslip6?

yatch, 2021-01-13:
SLIP may help. By the way, I'm afraid I cannot help you anymore on this since I don't think it is an issue of this project. Good luck.