contiki-ng/contiki-ng #1488: Event queue overflow, due to very fast tsch scan_timer event

MaripeD, 2021-01-07:
I tried rpl-udp example on cc1310, with tsch enable. when the node(udp-client) scans for network formation (but udp-server is not running, so no EB), scan_timer event firing very fast, not letting the processor to clear the other process post. Even single serial command (eg: help) increase the event count and it is not cleared after (variable: nevents). 

Sample logs:
-------------- initial debug -----------------------------
11:49:31.802 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 1
11:49:31.802 -> process: calling process 'send EB process' with event 136
11:49:31.802 -> process: calling process 'Event timer' with event 130
11:49:31.802 -> process: calling process 'main process' with event 136
11:49:31.835 -> process: calling process 'Event timer' with event 130
11:49:31.835 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 0
11:49:31.835 -> process: calling process 'main process' with event 136
11:49:31.835 -> process: calling process 'Event timer' with event 130
11:49:31.835 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 0
11:49:31.835 -> process_post: Process 'Event timer' posts event 136 to process 'send EB process', nevents 1
11:49:31.879 -> process: calling process 'main process' with event 136
11:49:31.879 -> process: calling process 'Event timer' with event 130
11:49:31.879 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 1
11:49:31.879 -> process: calling process 'send EB process' with event 136
11:49:31.879 -> process: calling process 'Event timer' with event 130
11:49:31.879 -> process: calling process 'main process' with event 136
11:49:31.879 -> process: calling process 'Event timer' with event 130
11:49:31.879 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 0
11:49:31.901 -> process: calling process 'main process' with event 136
11:49:31.901 -> process: calling process 'Event timer' with event 130
11:49:31.901 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 0

-------------- after 1 help command --------------------------
11:50:09.545 -> process: calling process 'Event timer' with event 130
11:50:09.545 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 2
11:50:09.545 -> process: calling process 'main process' with event 136
11:50:09.545 -> process: calling process 'Event timer' with event 130
11:50:09.578 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 2
11:50:09.578 -> process: calling process 'send EB process' with event 136
11:50:09.578 -> process: calling process 'Event timer' with event 130
11:50:09.578 -> process: calling process 'main process' with event 136
11:50:09.578 -> process: calling process 'Event timer' with event 130
11:50:09.578 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 1
11:50:09.611 -> process: calling process 'main process' with event 136
11:50:09.611 -> process: calling process 'Event timer' with event 130
11:50:09.611 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 1
11:50:09.611 -> process: calling process 'main process' with event 136
11:50:09.611 -> process: calling process 'Event timer' with event 130
11:50:09.611 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 1
11:50:09.645 -> process_post: Process 'Event timer' posts event 136 to process 'send EB process', nevents 2
11:50:09.645 -> process: calling process 'main process' with event 136
11:50:09.645 -> process: calling process 'Event timer' with event 130
11:50:09.645 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 2

--------------- after 8 help commands ------------------------
11:50:56.975 -> process: calling process 'Event timer' with event 130
11:50:56.975 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 11
11:50:57.008 -> process: calling process 'send EB process' with event 136
11:50:57.008 -> process: calling process 'Event timer' with event 130
11:50:57.008 -> process: calling process 'main process' with event 136
11:50:57.008 -> process: calling process 'Event timer' with event 130
11:50:57.008 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 10
11:50:57.008 -> process: calling process 'main process' with event 136
11:50:57.041 -> process: calling process 'Event timer' with event 130
11:50:57.041 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 10
11:50:57.041 -> process: calling process 'main process' with event 136
11:50:57.041 -> process: calling process 'Event timer' with event 130
11:50:57.041 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 10
11:50:57.075 -> process_post: Process 'Event timer' posts event 136 to process 'send EB process', nevents 11
11:50:57.075 -> process: calling process 'main process' with event 136
11:50:57.075 -> process: calling process 'Event timer' with event 130
11:50:57.075 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 11
11:50:57.075 -> process: calling process 'send EB process' with event 136
11:50:57.075 -> process: calling process 'Event timer' with event 130
11:50:57.114 -> process: calling process 'main process' with event 136
11:50:57.114 -> process: calling process 'Event timer' with event 130
[... 15 lines cut ...]
11:52:52.754 -> process: calling process 'main process' with event 136
11:52:52.754 -> process: calling process 'Event timer' with event 130
11:52:52.754 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 31
11:52:52.754 -> process_post: Process 'Event timer' posts event 136 to process 'Stack check', nevents 32
11:52:52.754 -> soft panic: event queue is full when event 136 was posted to Stack check from Event timer
11:52:52.787 -> process: calling process 'main process' with event 136
11:52:52.787 -> process: calling process 'Event timer' with event 130
11:52:52.787 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 31
11:52:52.787 -> process_post: Process 'Event timer' posts event 136 to process 'Stack check', nevents 32
11:52:52.820 -> soft panic: event queue is full when event 136 was posted to Stack check from Event timer
11:52:52.820 -> process: calling process 'main process' with event 136
11:52:52.820 -> process: calling process 'Event timer' with event 130
11:52:52.820 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 31
11:52:52.820 -> process_post: Process 'Event timer' posts event 136 to process 'Stack check', nevents 32
11:52:52.853 -> soft panic: event queue is full when event 136 was posted to Stack check from Event timer
11:52:52.853 -> process: calling process 'main process' with event 136
11:52:52.853 -> process: calling process 'Event timer' with event 130
11:52:52.853 -> process_post: Process 'Event timer' posts event 136 to process 'main process', nevents 31
11:52:52.853 -> process_post: Process 'Event timer' posts event 136 to process 'Stack check', nevents 32
11:52:52.886 -> soft panic: event queue is full when event 136 was posted to Stack check from Event timer
11:52:52.886 -> process: calling process 'send EB process' with event 136
11:52:52.886 -> process: calling process 'Event timer' with event 130
11:52:52.886 -> process_post: Process 'Event timer' posts event 136 to process 'send EB process', nevents 31
11:52:52.886 -> process_post: Process 'Event timer' posts event 136 to process 'Stack check', nevents 32
11:52:52.920 -> soft panic: event queue is full when event 136 was posted to Stack check from Event timer

alexrayne, 2021-01-11:
looks that you have TSCH_ASSOCIATION_POLL_FREQUENCY more then CLOCK_SECOND ?

MaripeD, 2021-01-14:
the default value TSCH_ASSOCIATION_POLL_FREQUENCY is 100.
and
etimer_set(&scan_timer, CLOCK_SECOND / TSCH_ASSOCIATION_POLL_FREQUENCY);
so, scan_timer expiring for every 10ms, which may very fast than event queue clearance. this could potentially cause event queue overflow, when it is trying to form a network.
correct me if it is wrong.
Thanks

alexrayne, 2021-01-14:
this should checks, some platforms have TSCH_ASSOCIATION_POLL_FREQUENCY == 1000

atiselsts, 2021-05-26:
@MaripeD interesed in testing #1515 ?