contiki-ng/contiki-ng #1490: Link error when building zoul tests

mizapf, 2021-01-13:
I've got a bundle of Zoul Re-Motes RevB, so I checked out the contiki-ng project and tried to build the test programs in examples/platform-specific/zoul.

However, I end up with this (the same happens with contiki-os):

$ make TARGET=zoul 
which: no rlwrap in (/home/michael/bin:/usr/local/bin:/usr/bin:/bin)
  MKDIR     build/zoul/remote-revb/obj
  CC        ../../../arch/platform/zoul/dev/tsl256x.c
  CC        ../../../arch/platform/zoul/dev/sht25.c
  CC        ../../../arch/platform/zoul/dev/bmpx8x.c
  CC        ../../../arch/platform/zoul/dev/motion-sensor.c
  CC        ../../../arch/platform/zoul/dev/adc-sensors.c
  CC        ../../../arch/platform/zoul/dev/weather-meter.c
[ ... more lines ... ]
  CC        ../../../os/net/routing/rpl-lite/rpl-of0.c
  CC        ../../../os/net/routing/rpl-lite/rpl-timers.c
  CC        ../../../arch/cpu/cc2538/cc2538.lds
  CC        ../../../arch/cpu/cc2538/./startup-gcc.c
  CC        test-tsl256x.c
  LD        build/zoul/remote-revb/test-tsl256x.elf
/usr/lib64/gcc/arm-none-eabi/10/ld: build/zoul/remote-revb/obj/bme280-sensor.o:(.bss.bme280_mea+0x0): multiple definition of `bme280_mea'; build/zoul/remote-revb/obj/bme280.o:(.bss.bme280_mea+0x0): first defined here
collect2: error: ld returned 1 exit status
make: *** [../../../arch/cpu/arm/cortex-m/Makefile.cortex-m:26: build/zoul/remote-revb/test-tsl256x.elf] Error 1
rm test-tsl256x.o build/zoul/remote-revb/obj/startup-gcc.o

Any ideas what I may be doing wrong here?

g-oikonomou, 2021-01-13:
I just tried locally and it works, so I am suspecting a toolchain version thing - you are not necessarily doing anything wrong. Perhaps you have a newer toolchain version that complains about something that older ones don't?

I tried with the version below on macos

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

mizapf, 2021-01-13:
I installed the cross compiling toolchain on my Tumbleweed Linux (openSUSE).

$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (SUSE Linux) 10.2.1 20201202 [revision e563687cf9d3d1278f45aaebd03e0f66531076c9]
Copyright (C) 2020 Free Software Foundation, Inc.
...

Still weird that such a thing as duplicate symbols could change over toolchain versions.

closed 2021-05-09 by g-oikonomou